awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #475: Issue with the build conflicting with AWS python libraries like boto3

js1972, 2020-06-03:
I'm finding that when I build this sdk (wtih kvsink) that I can no longer use boto3 as it complains about not being able to find openssl.
If I dont build kvssink then boto3 will work just fine with the existing openssl on the system (nvidia jetson platform).

So I'm trying to build kvssink without dependencies so that I can use the openssl already on the system and just apt-get install the others:
curl, liblog4cplus-dev, libjsoncpp-dev.

I'm running cmake like this:
```
cmake .. -GNinja -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF
```
Which is all good except for JSMN. It complains that jsmn is not available. How do I resolve this as jsmn is not an installable library it is simple a header file ? I've tried placing it in the open-source folder but it does not help.

Any idea how I can build without installing all the deps from source but get around this JSMN issue?

chehefen, 2020-06-08:
we are working on make the BUILD_DEPENDENCIES=OFF path. Will keep you posted.

chehefen, 2020-06-19:
the change in done in f620d3a7e8f08883660bab8c7a39d4b90246a714. You can follow instruction here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#configure to do BUILD_DEPENDENCIES=OFF

MushMal, 2020-06-26:
Please reopen if still having issues

closed 2020-06-26 by MushMal