awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #477: How can I use kvssink with an AWS Educate Account / Temporary AWS Access

TheSilentHorizon, 2020-06-04:
Basically whats in the headline. Kvssink only has only a param for id and secret , but no param for token.

I already tried using credential-Path but theres no example of how to set up a credential File for kvssink. The one in ".aws\credentials" dont seem to be recognized.

MushMal, 2020-06-04:
@TheSilentHorizon there is no param for the session token. This is done specifically as kvs sink credential integration is not "interactive" per-say. What I mean by this is that the credential provider will be created at the initialization time. Presence of the session token implies that there is an expiration associated with it (up-to 1 hour). Once the streaming starts, we can't supply fresh credentials like in a fully-integrated application. This is the reason that for temporary credentials currently you have two options - one is IoT and the other one is the file credentials.

I agree that we should have a better example of the file credential provider and will be adding some soon.

Meanwhile, if you need to use session token, specify the credential file and have a single line in the file containing items separated by space:

"CREDENTIALS access_key expiration secret_key session_token"

where expiration is in a form of "YYYY-MM-DDTHH:mm:SSZ"

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/src/source/Common/FileCredentialProvider.c#L164

TheSilentHorizon, 2020-06-04:
@MushMal Thanks for replying so fast! Is there a value inside the aws educate platform where i can read this exact expiration value or do i have to calculate it myself?

MushMal, 2020-06-04:
To be honest I am not familiar with it. The expiration is an absolute time when the creds will expire. Internally, KVS SDK uses UINT64 to represent the UNIX time.

MushMal, 2020-06-04:
From what I understand, there is nothing special when it comes to the account id/access key/secret key/session token. You just need to retrieve these somehow.

TheSilentHorizon, 2020-06-04:
I found it, its in the Account Details in Vocareum.

And I tested it, it works. I have a improvement id like to ask, but I do that in a separate issue.

Thanks a lot!

closed 2020-06-04 by TheSilentHorizon

MushMal, 2020-06-04:
Sounds good! We are always looking forward for folks to contribute to the design and the codebase. If you have something concrete, please don't hesitate to do a PR

asyschikov, 2022-05-04:
If you have your temp credentials available in the terminal, you can use the following script to put them in KVS supported format:

```
import boto3
session = boto3.Session()
cred = session.get_credentials()
print(
    'CREDENTIALS',
    cred.access_key,
    cred._expiry_time.isoformat(),
    cred.secret_key,
    cred.token
)
```

and then run `python3 export-kvs-creds.py >credentials.txt`