awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #478: Clear indication of misconfigured credentials

TheSilentHorizon, 2020-06-04:
**Is your feature request related to a problem? Please describe.**
When you have a malformed credential file in the kvssink plugin, you get a return code "0x9" which says "STATUS_OPEN_FILE_FAILED".  There should be a error code which indicates that the credential isnt readable, but i think this is too vague.

**Describe the solution you'd like**
A seperate error code for a malformed configuration. Makes this more straight forward.

Related question: #477

MushMal, 2020-06-04:
@TheSilentHorizon thanks for the request. We can certainly improve on this.

MushMal, 2020-06-05:
@TheSilentHorizon I just checked the code and we only return STATUS_OPEN_FILE_FAILED whenever the underlying read file operation fails. Unfortunately, due to the platform independent nature of the codebase, we have to rely on fopen c API that doesn't give a lot of details.

TheSilentHorizon, 2020-06-05:
@MushMal But you know in this case that its the configuration file, dont you? Since its the file mentioned in "-credential-path". So at least that could give a hint there. I assumed earlier, that the error i recieved was due to my pipelines source, not because of a malformed configuration file.

MushMal, 2020-06-05:
Ok, I understand the issue. Generally speaking, the entire codebase is layered in this manner. The low-level operations will most likely have no concept of the higher layers. In this case, we do have some context so the status code for failed file open can be translated to more specific status.

MushMal, 2020-06-05:
The main difficulty is the backwards compat in this case if any applications do expect the more generic open file failed status code. Highly doubt that there is any but this will warrant a version bump.

MushMal, 2020-06-05:
Resolving this issue - just pushed the change in C producer. We will rebase the CPP producer on latest version of C producer as soon as possible

closed 2020-06-05 by MushMal