awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #479: [QUESTION]

imizus, 2020-06-05:
Hi! I am using Producer SDK in the application, that takes frames from a simple UDP-based internal protocol to a KVS Stream.

Sometimes I'm facing a packet loss in the UDP. In this case I can't provide the SDK with an ongoing frame in time. And it happened that if there is high amount of lost frames, the SDK fails to keep up with the connection and some issues occur that lead to a necessity to restart the streaming.

I wonder if there is a problem with my stream configuration:

```
	config.m_streamRetention = hours(2);
	config.m_streamMaxLatency = seconds(15);
	config.m_fps = 30;
	config.m_bitrate = 120000000;

	auto stream_definition = make_unique<StreamDefinition>(config.m_streamName.c_str(),
		config.m_streamRetention,
		nullptr,//&tags,
		"",
		STREAMING_TYPE_REALTIME,
		"video/h264",
		config.m_streamMaxLatency,
		seconds(1), // fragment duration
		milliseconds(1),
		true, // key frame fragmentation
		true, // frame_timecodes
		true, // absolute timecodes
		true, // application fragment acks required
		true, // recover after an error
		true, // recalculate metrics at runtime
		NAL_ADAPTATION_FLAG_NONE,
		config.m_fps,
		config.m_bitrate);
```
If there is a way to improve this config?

MushMal, 2020-06-05:
@imizus the older version of the SDK does not automatically support the "intermittent" producer case.

Please upgrade to the latest commits. In this case, it will support automatic intermittent producer scenario. When the application stops producing frames into the stream object, the connection will timeout (from the backend) and will notify the underlying stream object. If the buffer is empty, the SDK will not re-try to create a new session. As soon as a new frame comes in, it will automatically issue PutMedia call. 

The only thing to note here is that as in your case with the persistence, due to the fact that we won't be getting the last fragment persisted ACK, we will re-send the last fragment first to get the ACK. In order to handle this, the application will have to explicitly specify EoFR frame at the end of the session so the backend will know the fragment is "closed". Most use cases this won't matter at all.

MushMal, 2020-06-05:
BTW, the config looks good.

imizus, 2020-06-07:
Thank you for the rapid answer.

I have updated to SDK 3.0.0 and now it seems to work much better, but another new issue occured.
During the data uploading a single message pops out hundreds times:

```
07:13:18.450 ERROR getKinesisVideoStreamData(): operation returned status code: 0x5200002f
```

In the doc it's written that this code means `STATUS_NO_MORE_DATA_AVAILABLE`, but the incoming and outgoing FPS are measured, both equal to 30.

Why is it happening? Is there any impact of this on the working of the app and how to take rid of it?

MushMal, 2020-06-07:
@imizus this is a known issue that was fixed later in PIC. This is not an error but a status code indicating the buffer is empty (good) but it gets misinterpreted as a warning.

Generally, the networking thread pulls the bits from the lower-layer buffer. The buffer can return the bits, no more data, eos status codes to signal certain things to the network layer (C Producer layer) and these are not errors.

The streaming shouldn't be affected by this erroneous printout

imizus, 2020-06-07:
Thank you again!

And another question.

On the side of consumer we have a Java application that downloads the stream and decodes it. And sometimes we see 2 errors (on different consumer instances):

* java.lang.RuntimeException: Add new ParserException type
* java.lang.IllegalArgumentException: Trying to decode an EBML ID and it wants 4 more bytes, but IDs max out at 4 bytes

What do they mean?

MushMal, 2020-06-07:
It’s hard to tell without more details. However, I am guessing that this is likely caused by the producer side dropping a partial frame perhaps. The real-time path will evaluate to but in and bit out to minimize the latency so by the time the backend detects a broken fragment the bits have been already sent down to the consumer

MushMal, 2020-06-26:
Resolving due to staleness

closed 2020-06-26 by MushMal