awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #480: [BUG] not compiled on OSX 10.13

ifuschini, 2020-06-05:
**Describe the bug**
Not compile on macbook pro 2019 macOS 10.15.5
**To Reproduce**
[ 16%] Linking C shared library libcurl.dylib
Undefined symbols for architecture x86_64:
  "_CONF_modules_free", referenced from:
      _Curl_ossl_cleanup in openssl.c.o
  "_ENGINE_cleanup", referenced from:
      _Curl_ossl_cleanup in openssl.c.o
  "_ERR_free_strings", referenced from:
      _Curl_ossl_cleanup in openssl.c.o
  "_EVP_MD_CTX_create", referenced from:
      _Curl_ossl_md5sum in openssl.c.o
      _Curl_ossl_sha256sum in openssl.c.o
  "_EVP_MD_CTX_destroy", referenced from:
      _Curl_ossl_md5sum in openssl.c.o
      _Curl_ossl_sha256sum in openssl.c.o
  "_EVP_cleanup", referenced from:
      _Curl_ossl_cleanup in openssl.c.o
  "_OPENSSL_add_all_algorithms_noconf", referenced from:
      _Curl_ossl_init in openssl.c.o
  "_SSL_COMP_free_compression_methods", referenced from:
      _Curl_ossl_cleanup in openssl.c.o
  "_SSL_get_ex_new_index", referenced from:
      _ossl_get_ssl_conn_index in openssl.c.o
      _ossl_get_ssl_sockindex_index in openssl.c.o
  "_SSL_library_init", referenced from:
      _Curl_ossl_init in openssl.c.o
  "_SSL_load_error_strings", referenced from:
      _Curl_ossl_init in openssl.c.o
  "_SSLeay", referenced from:
      _Curl_ossl_version in openssl.c.o
  "_SSLv23_client_method", referenced from:
      _ossl_connect_step1 in openssl.c.o
  "_SSLv3_client_method", referenced from:
      _ossl_connect_step1 in openssl.c.o
  "_sk_num", referenced from:
      _cert_stuff in openssl.c.o
      _get_cert_chain in openssl.c.o
      _verifyhost in openssl.c.o
      _X509V3_ext in openssl.c.o
  "_sk_pop", referenced from:
      _cert_stuff in openssl.c.o
  "_sk_pop_free", referenced from:
      _cert_stuff in openssl.c.o
  "_sk_value", referenced from:
      _get_cert_chain in openssl.c.o
      _verifyhost in openssl.c.o
      _X509V3_ext in openssl.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/libcurl.dylib] Error 1
make[4]: *** [lib/CMakeFiles/libcurl.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [build/src/project_libcurl-stamp/project_libcurl-build] Error 2
make[1]: *** [CMakeFiles/project_libcurl.dir/all] Error 2
make: *** [all] Error 2
CMake Error at open-source/amazon-kinesis-video-streams-producer-c/CMake/Utilities.cmake:59 (message):
  CMake step for libcurl failed: 2
Call Stack (most recent call first):
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:96 (build_dependency)

MushMal, 2020-06-05:
Seems you don't have SSL installed. Are you using a clean build? Can you attach the entire build logs?

You could also try -DBUILD_DEPENDENCIES to compile the dependencies

disa6302, 2020-06-15:
@ifuschini , 

Any updates on this?

chehefen, 2020-06-19:
you can follow the instruction here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#configure to build with BUILD_DEPENDENCIES=OFF.

MushMal, 2020-06-26:
Resolving due to staleness

closed 2020-06-26 by MushMal