awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #481: [BUG]: RaspberryPi , when I try to run make after the Cmake as per instructions in installation instruction i see the following error

outkasst, 2020-06-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**SDK version number**
Include the SDK version you are running. If it is a specific commit on master branch, include that

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.
[ 98%] Building CXX object CMakeFiles/kvs_gstreamer_audio_video_sample.dir/samples/kvs_gstreamer_audio_video_sample.cpp.o
/home/pi/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/samples/kvs_gstreamer_audio_video_sample.cpp: In function ‘GstFlowReturn on_new_sample(GstElement*, CustomData*)’:
/home/pi/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/samples/kvs_gstreamer_audio_video_sample.cpp:445:100: error: ‘gst_segment_to_running_time_full’ was not declared in this scope
         dts_sign = gst_segment_to_running_time_full(segment, GST_FORMAT_TIME, buffer->dts, &buf_dts);
                                                                                                    ^
CMakeFiles/kvs_gstreamer_audio_video_sample.dir/build.make:62: recipe for target 'CMakeFiles/kvs_gstreamer_audio_video_sample.dir/samples/kvs_gstreamer_audio_video_sample.cpp.o' failed
make[2]: *** [CMakeFiles/kvs_gstreamer_audio_video_sample.dir/samples/kvs_gstreamer_audio_video_sample.cpp.o] Error 1
CMakeFiles/Makefile2:281: recipe for target 'CMakeFiles/kvs_gstreamer_audio_video_sample.dir/all' failed
make[1]: *** [CMakeFiles/kvs_gstreamer_audio_video_sample.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

**Platform (please complete the following information):**
 - OS: [e.g. Linux, Windows]
 - Processor: [e.g. ARMv4]
 - Version [e.g. 18.04]

**Additional context**
Add any other context about the problem here.

chehefen, 2020-06-08:
which gstreamer version are you using? You can run `gst-launch-1.0 --version` to find that out.

outkasst, 2020-06-08:
Hi,

This is what i get:
gst-launch-1.0 version 1.4.4
GStreamer 1.4.4
http://packages.qa.debian.org/gstreamer1.0

thanks

chehefen, 2020-06-09:
your gstreamer version is too old. it's missing the gst_segment_to_running_time_full api.

outkasst, 2020-06-09:
hi,
Is it possible to install or update gstreamer with pip? Or how do i install it?
thx

disa6302, 2020-06-09:
You can run the following instruction on your Pi:

`sudo apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools`

MushMal, 2020-06-09:
This should be somewhere in the readmes I believe.

@outkasst could you please update your gstreamer and retry. Open a separate issue if anything. Please Close this one if you are able to compile

disa6302, 2020-06-15:
@outkasst , 

Any updates?

outkasst, 2020-06-16:
still cant install it.. for now please go ahead and put it as not a bug..

closed 2020-06-16 by MushMal