awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #482: unable to stream audio from gstreamer

maheshwari-siddharth, 2020-06-07:
gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I4204_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink stream-name=MyKinesisVideoStream storage-size=512 osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink

I am getting below error 

erroneous pipeline: could not parse caps "video/x-raw,format=I4204_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512"

erroneous pipeline: no element "avenc_aac"

I tried most of the command from doc availabe for macOS nothing is working at all

Thanks
Siddharth Maheshwari

MushMal, 2020-06-07:
From what I understand this is from gstlibav plugin. You can install it and give it a try. GStreamer related questions are best answered in GStreamer forums

chehefen, 2020-06-07:
you need to do brew install gst-libav to get the plugin.

disa6302, 2020-06-09:
@maheshwari-siddharth , 

Any updates? Does installing gst-libav fix your problem?

maheshwari-siddharth, 2020-06-10:
let me share output of all commands

gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I420,width=1280,height=720 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=MyKinesisVideoStream storage-size=512  osxaudiosrc ! audioconvert ! avenc_aac ! queue

this command runs and shows video but no audio

gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I420,width=1280,height=720 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=MyKinesisVideoStream storage-size=512 osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

this command has an error 
WARNING: erroneous pipeline: No sink-element named "(null)" - omitting link

gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I4204_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink stream-name=MyKinesisVideoStream storage-size=512 osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

this command again has an erorr
could not parse caps "video/x-raw,format=I4204_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512"

MushMal, 2020-06-10:
We have a few samples of command lines defined here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md

`gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,width=640,height=480,framerate=30/1,format=I420 ! omxh264enc periodicty-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name="my-stream-name" access-key="YourAccessKey" secret-key="YourSecretKey" alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.`

maheshwari-siddharth, 2020-06-10:
@MushMal

maheshwari-siddharth, 2020-06-10:
@MushMal 
this command give me this error
WARNING: erroneous pipeline: no element "v4l2src"

disa6302, 2020-06-10:
@maheshwari-siddharth ,

Can you try this pipeline? 

gst-launch-1.0 -v avfvideosrc ! videoconvert ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey" osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

This uses the Mac OS audio/video camera source

This is directly referenced from: 
`https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/macos.md#running-the-gst-launch-10-command-to-start-streaming-both-audio-and-raw-video-in-mac-os`

MushMal, 2020-06-10:
@maheshwari-siddharth all your questions are related to GStreamer and as such GStreamer forums might have better answers. I didn't know what platform or capabilities you are using and gave a sample command line. Please check our docs for mac - there are some samples in there.

Please resolve this issue if your question has been answered

maheshwari-siddharth, 2020-06-10:
gst-launch-1.0 -v avfvideosrc ! videoconvert ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey" osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

ERROR: from element /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0: Internal data stream error.

gst-launch-1.0 -v avfvideosrc device-index=1 ! h264parse ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey" osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

ERROR: from element /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0: Internal data stream error.

MushMal, 2020-06-10:
here is an example of audio/video on mac osx

gst-launch-1.0 -v avfvideosrc device-index=1 ! h264parse ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey" osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

from https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/macos.md

In genreal, you can work with GStreamer forums to get your pipeline working and then only add the kvssink

maheshwari-siddharth, 2020-06-10:
> @maheshwari-siddharth all your questions are related to GStreamer and as such GStreamer forums might have better answers. I didn't know what platform or capabilities you are using and gave a sample [...]

we have a mac book air 2017 model with catalina installed , we have checked all commands used in the githhub

maheshwari-siddharth, 2020-06-10:
> here is an example of audio/video on mac osx
>
> gst-launch-1.0 -v avfvideosrc device-index=1 ! h264parse ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" [...]

ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0: Invalid video capture device index

disa6302, 2020-06-10:
@maheshwari-siddharth , 

As per offline discussions, you are encountering Internal data stream error for the pipeline I suggested. This could mean the settings your Mac uses are different and hence the pipeline does not work as expected. As suggested by @MushMal , please work on fixing the pipeline and if the pipeline issue is fixed and you still face issues, we would be happy to help you out in setting up. Thanks!

maheshwari-siddharth, 2020-06-10:
you advised the using a lib from Gstreamer i am usinhg it if Gstreamer lib is not working i thing you must raise this issue to Gstreamer , or if you have any AWS lib for this i am happy to use it but as you are working with Gtreamer closely i think they will take quicker action on your tickets or forum .
I just wanted a basic life cycle of kineses which aws is providing either from aws lib or Gstreamer Lib .

MushMal, 2020-06-10:
@maheshwari-siddharth the issue is that the setup that we have tried must be somewhat different than yours. The command lines from the doc do work on Mac OSx on macbook pros for example - pre-Catalina. It is impossible for us to have a comprehensive list of GStreamer/platform/OS/Software/Versions/Hardware combinations to test internally. Hence, we can just try to help with some advise but won't be able to actually debug the GStreamer pipeline on your device. Perhaps it's an issue with new Catalina - not sure.

Let's do first things first, it seems that your video is working - correct? 

* You are able to run video ingestion via gst-launch with kvs plugin?
* You are able to run the video sample?
* Are you able to run audio/video sample application?

If the answer to the questions above is positive then the internal error is likely caused by a resolution/bitrate/fps combination. Try to change those to suit for your hardware profile.

Again, we are sorry that we are not as effective to help you with GStreamer issues which are not something under our control.

maheshwari-siddharth, 2020-06-10:
You are able to run video ingestion via gst-launch with kvs plugin?
Yes - i am able to use video
You are able to run the video sample?
yes - I am run video sample too.
Are you able to run audio/video sample application?
No- i am not able to run the audio/video solution

if kineses is depends on Gstremaer and Gstreamer is not working then how you expect client to use the Kineses. you solution is incomplete without Gstreamer and you want client to talk to the gstreamer  your behalf, doesn't it strange

Even I  am not sure it is catalina.

MushMal, 2020-06-10:
@maheshwari-siddharth KVS is orthogonal to GStreamer. We can integrate with any media pipeline. We just have samples that integrate with GStreamer. 

In general, GStreamer is a platform and not a single tool. 

I would recommend you to troubleshoot audio/video without kvs plugin first. You can use simple fakesink. 

It's hard for us to use the endless permutation of different OS/hardware/software/configs/etc,,

maheshwari-siddharth, 2020-06-10:
you have many ways to pass the ball on client side but end of the day my kineses is not working.
you are asking me check  every single  plugin before working on AWS kineses, here your recommendation has not value use gstreamer.

MushMal, 2020-06-10:
@maheshwari-siddharth we are sorry you feel that we are passing the ball to you when we can't debug/troubleshoot your hardware for you for an entirely unrelated project to AWS/Kinesis.

There might be people in the community who have MacBook Air with Catalina that could give it a try to see what's wrong.

maheshwari-siddharth, 2020-06-10:
i dont think its hardware we have checked it amazon chime is perfectly working on my machine if you want i can get you checked. now it look like scapegoat by blaming the hardware.

MushMal, 2020-06-15:
Understood. Perhaps it's something that the RTSP plugin requires that's not supported on Catalina. I am resolving this issue as we can't make further progress on GStreamer RTSP plugin

maheshwari-siddharth, 2020-06-30:
Hi Mushmal

could you be able to fix that

Thanks
Siddharth Maheshwari

MushMal, 2020-06-30:
Apologies, MacOS Catalina with the hardware profile you are referring to is not something we have tested yet. We will update once we have a chance to test. Meanwhile, were you able to use fakesink instead of kvs sink to eliminate the kvs dependency?

maheshwari-siddharth, 2020-07-01:
can you please help me by giving a fakesink exact command, i will try that

MushMal, 2020-07-01:
You could try something like this

`gst-launch-1.0 autovideosrc device=/dev/video0 ! video/x-raw,width=1280,height=720 ! videoconvert ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=5000 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! decodebin ! autovideosink`

maheshwari-siddharth, 2020-07-04:
the command you mentioned is not even making "mykinesisvideostream"

below is a command we are able to make a video stream but audio is missing please help me here

gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I420,width=1280,height=720 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=MyKinesisVideoStream storage-size=512 access-key="" secret-key="" aws-region=""

MushMal, 2020-07-04:
It’s not supposed to. I was just trying to help you to troubleshoot your rtsp connection first by eliminating the KVS all together

maheshwari-siddharth, 2020-07-04:
what should i do now the get the audio please help hee too

maheshwari-siddharth, 2020-07-06:
what should i do how the get the audio please help here too

MushMal, 2020-08-03:
I am going to resolve this as I am hesitant to give advise on 3rd party software that I can't test/try on a given platform with a given configuration. GStreamer forums are the best place to ask for help for GStreamer plugin questions.

closed 2020-08-03 by MushMal

lehuy2012, 2021-04-22:
Currently, I using this command for a push from RTSP to the kinesis video stream.
-> I can see video on the KVS console but I can not hear any audio.
-> I also test using RTSP with VLC software, both video + audio are OK.
So, anybody can support me about this.
Thank you!

my command:

gst-launch-1.0 -v rtspsrc location="rtsp://35.xxx.xxx.xx" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name="mystream_name" storage-size=128 access-key="xxxxxxx" secret-key="xxxxxxxxx" aws-region="ap-northeast-1"