awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #484: OpenSSL Version: AWS Kinesis: Download and Build the Kinesis Video Streams C++ Producer SDK

phillipshaong, 2020-06-08:
Raspberry Pi 4
Raspbian Lite

After installing all the required dependencies, I get the following error message when trying to use the command `version OPENSSL_1_1_1 not found (required by /usr/lib/arm-linux-gnueabihf/libcurl.so.4)`

```
*** Installing runtime programs
install apps/openssl -> /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin/openssl
install ./tools/c_rehash -> /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin/c_rehash
/usr/bin/cmake: /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libssl.so.1.1: version `OPENSSL_1_1_1' not found (required by /usr/lib/arm-linux-gnueabihf/libcurl.so.4)
make[2]: *** [CMakeFiles/project_libopenssl.dir/build.make:76: build/src/project_libopenssl-stamp/project_libopenssl-install] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/project_libopenssl.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at open-source/amazon-kinesis-video-streams-producer-c/CMake/Utilities.cmake:59 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:78 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
```

Do I need to 

Thanks!

disa6302, 2020-06-08:
@phillipshaong ,

Could be a linking error. Can you try setting LD_LIBRARY_PATH to point to the openssl path and see if it works? 

`export LD_LIBRARY_PATH=/usr/local/lib`, assuming this is where openssl gets installed by default.
Also, check which openssl version is being referenced by running `which openssl`

The SDK installs openssl version 1.1.0l by default. and the one you installed install 1.1.1d. That might be the reason you are facing this problem.

MushMal, 2020-06-08:
What I am thinking is that the library or the path is not "mapped" properly as openssl executable itself is failing to find the library.

disa6302, 2020-06-13:
@phillipshaong , 

Any updates?

phillipshaong, 2020-06-17:
Thanks for following up! I'm not sure why, but after one of the reboots I just did, it stopped giving the error. I don't know why, since previous reboots have failed to fix the problem.

closed 2020-06-17 by phillipshaong