awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #485: [BUG]

imizus, 2020-06-08:
**Describe the bug**
I'm running an application based on the Producer SDK and at some point facing a crash. 

Here is the stacktrace:
```
********************************
*      SIGNAL CAUGHT           *
********************************
Process ID: 105134
Thread ID: 112749
Signal: 11
Exception: Segmentation fault
Reason: address not mapped to object
Fault Address: 0x8
************************
*      BACKTRACE       *
************************
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/KinesisVideoUploadApp(_ZN4HPCI13SignalHandler14PrintBacktraceEiP9siginfo_t+0x32)[0x4c25e2]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/KinesisVideoUploadApp(_ZN4HPCI13SignalHandler8OnSignalEiP9siginfo_tPv+0x8)[0x4c2638]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7f91a068c390]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x332cf)[0x7f91a21062cf]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x33c51)[0x7f91a2106c51]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x354de)[0x7f91a21084de]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x26340)[0x7f91a20f9340]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x2b9b0)[0x7f91a20fe9b0]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x360c2)[0x7f91a21090c2]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(curl_multi_perform+0x10d)[0x7f91a2109f5d]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(curl_easy_perform+0x13b)[0x7f91a2100a0b]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/libcproducer.so(curlCompleteSync+0x124)[0x7f91a2b245c4]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/libcproducer.so(curlCallApi+0x28)[0x7f91a2b1e9a8]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/libcproducer.so(putStreamCurlHandler+0x11f)[0x7f91a2b1fb4f]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f91a06826ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f919f72841d]
```

**SDK version number**
I am using Producer SDK 3.0.0.

**Platform (please complete the following information):**
 - OS: Linux
 - Processor: Intel Xeon Platinum 8176
 - Version: Ubuntu 16.04.4

MushMal, 2020-06-08:
@imizus the crash seems to originate in libcurl but there is no debug symbol info to understand what could cause this as the curl crash could indeed be caused by the producer library as well.

Could you please build the debug version of the SDK to capture the crash stack (running under the debugger and dumping backtraces for threads are the best, or at least get the code dumps). What's the scenario that's causing the crash - please include some of the debug logs leading to the crash so we can get more insight.

imizus, 2020-06-08:
There is nothing significant in the log before the crash. It was uploading flawlessly and then the crash occured.

The only useful info we have about it is that it occured on 2 instances of the application with a span of 5 minutes.

Concerning the debug version of the SDK, maybe I'll give it a try... The problem is that it seems that it can't keep 30FPS on our system....

MushMal, 2020-06-08:
@imizus the SDK itself is super lightweight and should power through with debug logs and debug codebase with little impact. I understand that this might not be the best but could provide useful info.

An alternative is to create the symbols, configure the exception handler to dump the cores and match against the symbols. 

The best core dump can be generated by running the debug code as aside from the symbols it also removes some of the optimizations to map directly to source lines. 

The crash dump simply points to a NULL deref 

`Fault Address: 0x8`

though, it's not clear what curl library is attempting to do to cause it in pthread library call.

MushMal, 2020-06-08:
one more thing. Try to bisect the issue. Try to run with a single stream and see whether this causes an issue that way you won't have the latency or perf problems.

If this works OK then it points to some major pressures on the system. What's your memory consumption/situation? Most of the OS-es overcommit the memory and will crash when the actual memory is claimed. What's your storage configuration? I doubt that memory overcommit is the issue here

imizus, 2020-06-08:
We'll try to make a Debug version and reproduce the issue. The problem is that it's relatively rare.

MushMal, 2020-06-08:
Understood. Please let us know how we can help.

We had had some issues that are rare which required us to put together long-haul 'canary' applications that run with full debug symbols to capture the root cause.

imizus, 2020-06-10:
I have built a Debug version of all libraries including libcproducer.so. But it still lacks the exact line numbers in the stack trace. To do so, I set the CMAKE_BUILD_TYPE=Debug. I believed 

What am I doing wrong?

imizus, 2020-06-10:
Update: I have deciphered my recent crash stacktrace manually.

Here is the new stacktrace from the Debug version:
```
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/KinesisVideoUploadApp(_ZN4HPCI13SignalHandler14PrintBacktraceEiP9siginfo_t+0x45)[0x671bd9]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/KinesisVideoUploadApp(_ZN4HPCI13SignalHandler8OnSignalEiP9siginfo_tPv+0x24)[0x672098]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7fd082575390]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x332cf)[0x7fd083fef2cf]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x33c51)[0x7fd083fefc51]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x354de)[0x7fd083ff14de]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x26340)[0x7fd083fe2340]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x2b9b0)[0x7fd083fe79b0]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(+0x360c2)[0x7fd083ff20c2]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(curl_multi_perform+0x10d)[0x7fd083ff2f5d]
/usr/lib/x86_64-linux-gnu/libcurl.so.4(curl_easy_perform+0x13b)[0x7fd083fe9a0b]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/libcproducer.so(curlCompleteSync+0x1af)[0x7fd084a4aff6]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/libcproducer.so(curlCallApi+0x6e)[0x7fd084a41929]
/home/produser/FreedCode/Base/Apps/KinesisVideoUploadApp/libcproducer.so(putStreamCurlHandler+0x1a8)[0x7fd084a470f7]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7fd08256b6ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7fd08161141d]
```

curlCompleteSync+0x1af corresponds to:
```
<SDK root>/open-source/amazon-kinesis-video-streams-producer-c/src/source/Response.c:423
```

curlCallApi+0x6e corresponds to:
```
<SDK root>/open-source/amazon-kinesis-video-streams-producer-c/src/source/CurlApiCallbacks.c:660
```
putStreamCurlHandler+0x1a8 corresponds to:
```
<SDK root>/open-source/amazon-kinesis-video-streams-producer-c/src/source/CurlApiCallbacks.c:2191
```

MushMal, 2020-06-12:
@imizus any further update on this issue? 

As I was thinking about this issue, I also wanted to ask about kernel overcommit or threading issues. As I can't tell what's wrong with the libpthread dereferencing NULL, I am thinking that this could happen if there is a failure to allocate memory for a thread for example. This points to OOM conditions. There is also possibility that the process has more threads than the max allowing it. 

Could you please check whats the memory allocation is in the process (top command perhaps - check the RSS). Also, check how many threads are there for the process. Check how many FDs are opened.

bigfatbrowncat, 2020-06-14:
I spent some hours running many instances of the application from gdb on the server. No luck to catch even one crash of this kind. That looks suspicious...

Just an idea. Which exact version of libcurl this library has been proven to work correctly with?

I am investigating 2 cases:
1. Building against a wrong libcurl
2. Building against a correct libcurl, but running (on a different machine) against a wrong one.

disa6302, 2020-06-15:
@bigfatbrowncat , 

We have been working with curl-7_68_0 (You can check it here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/CMake/Dependencies/libcurl-CMakeLists.txt) 

We were working with the debug version previously and now we build the RELEASE version after this commit: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/487

MushMal, 2020-06-15:
@bigfatbrowncat it's hard to know what exactly could have caused the issue in the first place. Perhaps linking against an older version of the curl library that's on the system instead of the one that the library was built against - similar to the scenario you've highlighted. I doubt that this is libpthread related issue - it's a manifestation of the crash rather than the root cause.

Could you perhaps also try running the release version in parallel to see whether the clean full build keeps exhibiting this?

MushMal, 2020-06-22:
It's been over a week for the last update. Do we know any more details? If not, this issue is stale and I will be resolving it soon

MushMal, 2020-06-23:
Resolving as stale. Please feel free to reopen or cut a new one.

closed 2020-06-23 by MushMal