awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #486: [BUG] Loading kvssink Gstreamer Plugin in a Native CUDA App

jlong29, 2020-06-09:
**Expected behavior**
I first validated that the gstreamer plugin kvssink worked via the gst-launch cli. I also tested the samples build with the SDK 3.0.0 and these works. I then expect the kvssink element to load successfully with the command ```gst_element_factory_make``` in a C/C++ program. **I also expect clear documentation for AWS regarding how to use the kvssink in a Native C/C++ App.** This is straightforward with other gstreamer plugins. My previous pipeline in this App used the udpsink plugin with no problem. Swapping in kvssink is the problem.

**Describe the bug**
gst_element_factory_make is failing for kvssink plugin in a native CUDA C++ App. The Gstreamer code is contained within an class declared/defined in .h/.cpp, which uses g++ to create an .o file, which is linked into the main App with nvcc (all this works when kvssink is swapped for udpsink).

**To Reproduce**
Makefile:
NOTE: AWS_KVS is the root directory of the 3.0.0 SDK built with the BUILD GSTREAMER PLUGIN cmake flag
```
#Gstreamer KVS Plug-in
KVS_PIC    = $(AWS_KVS)/open-source/amazon-kinesis-video-streams-pic
KVS_OPENS  = $(AWS_KVS)/open-source/local/include
KVS_PRO    = $(AWS_KVS)
KVS_CPRO   = $(AWS_KVS)/open-source/amazon-kinesis-video-streams-producer-c
KVS_GST    = $(AWS_KVS)/kinesis-video-gstreamer-plugin

INCLUDES  += -I$(KVS_PIC)/src/client/include
INCLUDES  += -I$(KVS_PIC)/src/common/include
INCLUDES  += -I$(KVS_PIC)/src/heap/include
INCLUDES  += -I$(KVS_PIC)/src/mkvgen/include
INCLUDES  += -I$(KVS_PIC)/src/state/include
INCLUDES  += -I$(KVS_PIC)/src/trace/include
INCLUDES  += -I$(KVS_PIC)/src/utils/include
INCLUDES  += -I$(KVS_PIC)/src/view/include

INCLUDES  += -I$(KVS_OPENS)
INCLUDES  += -I$(KVS_OPENS)/curl
INCLUDES  += -I$(KVS_OPENS)/log4cplus
INCLUDES  += -I$(KVS_OPENS)/openssl

INCLUDES  += -I$(KVS_PRO)/src
INCLUDES  += -I$(KVS_PRO)/src/credential-providers
INCLUDES  += -I$(KVS_PRO)/src/common
INCLUDES  += -I$(KVS_PRO)/src/JNI/include
INCLUDES  += -I$(KVS_CPRO)/src/include

LIBRARIES += -L$(KVS_PRO)/build -lKinesisVideoProducer
LIBRARIES += -L$(KVS_PRO)/build -lgstkvssink
LIBRARIES += -L$(KVS_PRO)/open-source/local/lib -lcrypto
LIBRARIES += -L$(KVS_PRO)/open-source/local/lib -lcurl
LIBRARIES += -L$(KVS_PRO)/open-source/local/lib -ljsmn
LIBRARIES += -L$(KVS_PRO)/open-source/local/lib -llog4cplus
LIBRARIES += -L$(KVS_PRO)/open-source/local/lib -lssl
```
Q: What exactly are the include directories and libraries required? This is not clear from the docs

I then build an *.o for the code that includes KinesisVideoProducer.h, <gst/gst.h>, and <gst/app/gstappsrc.h>

The only way I can get the App to compile with kvssink is to include KinesisVideoProducer.h in the source file, which g++ converts to an .o file. When I include it in the header, I get name conflicts related to atomic functions in a CommonDefs.h file with the repo, because nvcc then includes the KinesisVideoProducer.h, which brings in too many names and conflicts

```
# my Gstreamer Class
myGstreamer.o: myGstreamer.cpp myGstreamer.h
        $(EXEC) $(HOST_COMPILER) $(CCFLAGS) $(INCLUDES) -c myGstreamer.cpp

NOTE: CCFLAGS     = -O2 -Wall -std=c++11, HOST_COMPILER = g++ 5.4

myKVS: mainKVS.o other.o myGstreamer.o
        $(EXEC) $(NVCC) $(ALL_LDFLAGS) $(GENCODE_FLAGS) -o $@ $+ $(LIBRARIES)

NOTE: ALL_LDFLAGS = -m64 -std=c++11 -Xcompiler -O2 -Wall
```

**SDK version number**
3.0.0

**Platform (please complete the following information):**
 - OS: Linux, Windows
-Compiler: gcc/g++ 5.4, nvcc v9.0.252
 - Processor: aarch64 (Jetson TX1)
 - Version Ubuntu16.04

MushMal, 2020-06-09:
@jamesiri Thanks for the report. Just to reiterate on what I am reading:

* Need concise documentation on how to include kvs plugin in the application - Are you able to get details from https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/0a972632590b8baa11878b3cfc325fee0ad37b14/samples/kvs_gstreamer_audio_video_sample.cpp aside from the need to document the params

* Naming conflicts: Could you please provide more details as to what's actually conflicting?

jlong29, 2020-06-10:
Thanks for following up. I managed to solve the problem. It turns out I did not exactly understand how Gstreamer loads libraries at runtime within a native C/C++ App. I was trying to link to the gstkvssink library at compile time using make and Gstreamer only loads libraries at runtime. It uses it's search path, which includes GST_PLUGIN_PATH and LD_LIBRARY_PATH.

My mistake was that I am using a USB camera that I had not made user accessible by writing a udev rule. So I had to run my App is super user, which made my user .bashrc not visible, which made my GST_PLUGIN_PATH and LD_LIBRARY_PATH variables, with the paths to libgstkvssink.so, not visible, which created my runtime problem. By writing a udev rule to make the USB camera user accessible, I could run my App in user space, and then the plugin loaded properly, because it saw the necessary variables in .bashrc.

RE: naming collisions, I do believe there is still a problem with the KinesisVideoProducer.h includes and names brought in by CUDA 9 runtime via nvcc. I've managed to get around it as described above, but I'm pretty sure you can re-create it if you use nvcc to compile one of the gstreamer examples.

Mostly a false alarm, but I hope I helped expose a compatibility issue with nvcc and the KVS Producer SDK.

MushMal, 2020-06-10:
@jlong29 unfortunately, I don't have a CUDA compiler handy here. 

Regarding the naming conventions. When compiling C++ one has a benefit of the namespaces which makes it impossible to collide (we use a deep AWS KVS specific namespace). In C layers, though, there is a possibility of a collision. We have tried to avoid such collisions by having prefixed and specific names at the higher layers. CommonDef.h is one that can have collisions. Please try to be more specific and we will see if we can suggest some help there.

Meanwhile, please go ahead and close this issue as the original problem seems to be solved.