awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #491: [QUESTION] Workaround for NALUs ending with 0x00

farleylai, 2020-06-16:
NALUs ending with `0x00` would be rejected by KVS with error `0x3200000d` as discussed in #357 and the [comment](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/357#issuecomment-591616138).

However, as we are expanding support for major NVR vendors which expose their video stream access through RTSP, there is a vendor Dahua whose H.264 NALUs always end with `0x00`.
In short, their sequences of NALU types are listed as follows:
- key frame: [AUD, SPS, PPS, SEI, IDR]
- non-frame: [AUD, SEI, non-IDR]

Previous workaround to replace 0x00 with 0x01 does not work this time. To be specific, `0x3200000d` goes away and the fragments can be persisted after replacement but the KVS console reports no media found (there is a very small chance the stream playing works but only in a short time). We also tried to remove AUD/SEI to ensure SPS/PPS being the first NALUs of a key frame but the issue remains.

Therefore, we are in need of a guideline and an appropriate workaround in this case for KVS since supporting that vendor is necessary. Hopefully, it's just something trivial missing.

PS: the first 30 frames in H.264 bitstream (annex. b) are attached FYI. We made sure those are decodable by ffmpeg/VLC. The CPD from RTSP is the same as in the bitstream. The `parsing.log` shows the start code and the ending byte as described.

[dahua.tar.gz](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4784536/dahua.tar.gz)

MushMal, 2020-06-16:
Thanks @farleylai . We will investigate this and come up with something soon

MushMal, 2020-06-16:
One more thing... Is your application supplying the CPD or you are relying on auto-CPD extraction built in the SDK?

MushMal, 2020-06-16:
Kind of weird NALUs anyway

0000 0001 6764 001e ac1b 1a80 b03d ffff
0028 0021 6e0c 0c0c 8000 01f4 0000 7530
7430 07d0 0001 312d 5de5 c686 00fa 0000
2625 abbc b850 0000 0000 0168 ee38 3000

Looking at the frames seems EVERY non-frame NALu here returns an extra 0x00.

MushMal, 2020-06-16:
For example - the I-frame:

000 0001 0910 **00**00 0000 0167 6400 1eac
1b1a 80b0 3dff ff00 2800 216e 0c0c 0c80
0001 f400 0075 3074 3007 d000 0131 2d5d
e5c6 8600 fa00 0026 25ab bcb8 50**00** 0000
0001 68ee 3830 0000 0000 0106 000d bcff
8749 b516 3cff 8749 b516 4001 0400 7808
1006 01c4 80**00** 0000 0001 65b8...

MushMal, 2020-06-16:
Wondering if a special pre-processing needs for this particular camera vendor that simply gets rid of all the zeroes over 3 in a row and then feeds it into the rest of the processing... Need some experimentation to see whether that will work

farleylai, 2020-06-16:
The CPD(SPS/PPS) are extracted from RTSP/SDP by ffmpeg and also available in every key frame. The comparison indicates both are the same (including the ending zero byte).

We actually managed to remove the ending zero bytes (seems like ffmpeg parsing skips it) and made sure still decodable by ffmpeg. Nonetheless, if uploaded to KVS, the console still shows no media found. Hence, it is unclear what is still missing.

Also, this is the 2nd time we encounter NALUs ending with zero bytes from a different vendor. There seems like quite a chance this can happen even with major vendors.

MushMal, 2020-06-16:
By looking at it, we have two issues here. 

1) The extra 0s at the end of some of the NALus but not all of them.
2) AUD/SEI NALus included in the CPD which might be breaking both the producer side and/or the backend playback/HLS/MPEG-DASH.

Can you please answer some of my questions

* How are you using CPD: Are you directly specifying it or you rely on the SDK to extract it from the I-frame?
* When the Console playback shows "no fragments found" it simply means that it can't find the fragments - perhaps this is something different. Can you check the producer side logs in that case whether the video has been successfully uploaded up?

I would like to understand this issue better in order to handle both the broken encoders producing extra 0s and process the non-SPS/PPS NALus

farleylai, 2020-06-16:
> Can you please answer some of my questions

Sure,
 
> * How are you using CPD: Are you directly specifying it or you rely on the SDK to extract it from the I-frame?

The latter. Since the ending zero causes problems, the SPS/PPS in key frames are filtered for KVS to accept.

> * When the Console playback shows "no fragments found" it simply means that it can't find the fragments - perhaps this is something different. Can you check the producer side logs in that case whether the video has been successfully uploaded up?

My bad. Looking into the messages, there is a stream error for max fragment duration reached, implying the key frame flag was not set appropriately. This is indeed the case that the keyframe flag was lost as the NALUs are filtered for recomposition. Therefore in summary, just filtering out the ending zero byte and AUD/SEI seem to work at least in those cases we encountered so far. Nonetheless, it would be appreciated and handy if KVS can handle those not necessarily uncommon exceptions.

MushMal, 2020-06-18:
@farleylai could you please take a look at https://github.com/awslabs/amazon-kinesis-video-streams-pic/pull/68

It does handle the bad encoder output as well as it handles the AUD and SEI NALus.

This said, for some reason the decoding fails during the playback in the console (not yet sure whether this is actually your frames that are bad).

farleylai, 2020-06-18:
@MushMal Thanks for the effort. Might we expect the next release to apply this fix?

The playback issue was confirmed due to the loss of keyframe flag in the process of filtering out the trailing zero byte. After removing the ending zero and restoring the keyframe flag, everything seems to work fine. I appreciate that KVS improves its interoperability with popular brands in the market.

MushMal, 2020-06-18:
Absolutely. We just need to fix this property. Do you mind trying this patch/branch and see how this works - I couldn’t get this to work with the frames you’ve attached. I modified the video only sample to play these frames in a loop with a single key frame and the rest are p frames. The default is setting the flags to adapt the annexb to avcc for the console playback but I am getting a decoding error which is surprising. Perhaps, aside from adapting (which gets rid of the extra zero) we should also remove the AUD and SEI? If so, should we do that for the frames and the cpd? These need to be answered first before we feel comfortable with the changes and cutting a release

farleylai, 2020-06-18:
@MushMal The streaming upload to KVS and the console playback work fine on my side as long as SPS/PPS from CPD are rearranged to be the very first NALUs and the trailing zero bytes are filtered out from non-VCL NALUs given auto CPD extraction (the previous playback issue is due to the loss of keyframe flags).

As far as observed, almost all the non-VCL NALUs including the SPS/PPS in CPD contain the trailing zero byte from the Dahua NVR. Maybe the NALU parsing should follow the a similar way that ffmpeg/VLC deal with the trailing zero case for maximizing the interoperability?

I would like to give it a shot of the patch, is it sufficient to first check out the SDK@v3.0.0 and then the branch `bad_encoder_fix` of the submodule at `open-source/amazon-kinesis-video-streams-pic`?

MushMal, 2020-06-18:
You can simply point your c producer commit to the commit of the branch in the PR. Remove the dependencies directory and run cmake. BTW, you might need to enable the fixup for the bad encoder by specifying the compile time define. I know this is very involved. I will try it tomorrow our time and remove the NALus from the CPD and check the playback

MushMal, 2020-06-18:
Sorry, should have given more info - to enable 4 zeroes and one fixup you need to specify FIXUP_ANNEX_B_TRAILING_NALU_ZERO compiler define. I am hard-coding this for now in the local copy in c producer to get this compile flag through (for some reason I am not able to get this in via the ExternalProject_Add in libkvspic_CMakeLists.txt.

farleylai, 2020-06-18:
Got it. Will try to build soon.

MushMal, 2020-06-18:
Hmm, odd, got it working but I needed to actually remove the SEI and AUD NALus from the frames in order to get it working. The CPD works fine. Will need to implement some logic for that.

farleylai, 2020-06-18:
The CPD (SPS/PPS) is provided in the RTSP/SDP (not included in the attachment but the same as the SPS/PPS in the frame) and prepended before AUD in the frame to make the streaming upload and console playback work. I think the issue is that the current KVS parsing assumes the very first NALUs must be SPS/PPS for auto CPD extraction? If that is the case, perhaps, skipping until the SPS/PPS are found in the first frame may work better?

MushMal, 2020-06-18:
The fix I made handles the CPD just fine. The problem now is that after adapting to AvCC, we also need to remove the AUD and SEI from the frames. Once I did that, the playback works (which is the MSE in the browser)

MushMal, 2020-06-18:
And, I am thinking that perhaps the right approach to the frame adaptation is to remove any NALus which have the nal_ref_idc = 0 (for H264)

farleylai, 2020-06-18:
Agreed, that would make sense.

I also built the `bad_encoder_fix` branch within the master branch of `amazon-kinesis-video-streams-producer-c` with `-DFIXUP_ANNEX_B_TRAILING_NALU_ZERO=ON` option on for cmake.
However, if the trailing zeros are not filtered out manually, the console still shows `Fragment decoding error` even with AUD/SEI removed. Am I missing something from your build/changes?

PS: from the generated CMakefiles as follows, this compiler define should be on FYI:

```
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicClient.dir/DependInfo.cmake:  "FIXUP_ANNEX_B_TRAILING_NALU_ZERO"
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicClient.dir/flags.make:C_DEFINES = -DFIXUP_ANNEX_B_TRAILING_NALU_ZERO
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/DependInfo.cmake:  "FIXUP_ANNEX_B_TRAILING_NALU_ZERO"
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/flags.make:C_DEFINES = -DFIXUP_ANNEX_B_TRAILING_NALU_ZERO
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/DependInfo.cmake:  "FIXUP_ANNEX_B_TRAILING_NALU_ZERO"
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/flags.make:C_DEFINES = -DFIXUP_ANNEX_B_TRAILING_NALU_ZERO
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicState.dir/DependInfo.cmake:  "FIXUP_ANNEX_B_TRAILING_NALU_ZERO"
build/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicState.dir/flags.make:C_DEFINES = -DFIXUP_ANNEX_B_TRAILING_NALU_ZERO
build/CMakeCache.txt:FIXUP_ANNEX_B_TRAILING_NALU_ZERO:BOOL=ON
dependency/libkvspic/kvspic-src/src/mkvgen/src/NalAdapter.h:#ifdef FIXUP_ANNEX_B_TRAILING_NALU_ZERO
dependency/libkvspic/kvspic-src/CMakeLists.txt:option(FIXUP_ANNEX_B_TRAILING_NALU_ZERO "Fix-up some bad encoder behavior leaving a trailing zero at the end of NALu" OFF)
dependency/libkvspic/kvspic-src/CMakeLists.txt:if(FIXUP_ANNEX_B_TRAILING_NALU_ZERO)
dependency/libkvspic/kvspic-src/CMakeLists.txt:  add_definitions(-DFIXUP_ANNEX_B_TRAILING_NALU_ZERO)
dependency/libkvspic/kvspic-src/CMakeLists.txt:  target_compile_definitions(kvspic_test PRIVATE ALIGNED_MEMORY_MODEL=TRUE FIXUP_ANNEX_B_TRAILING_NALU_ZERO=TRUE)
```

MushMal, 2020-06-18:
If you are able to stream the bits in then you are properly using the fix-up. The main question then is why the MSE is failing to decode. Here, I think the problem is that the encoded frames should not contain SEI after the IDR slice. I will be providing a fix shortly.

farleylai, 2020-06-18:
OK, so you mean the console is supposed to decode successfully without manually removing the trailing zero? Got it and I remember initially it failed to upload with error `0x3200000d` immediately. Hence, with this fix, the uploading works but the console still finds something wrong for some reason.

BTW, what is the MSE you were referring to, the Media Source Extensions?

Also, where did you see the case that SEI follows IDR?
The frames in the attachment should only contain the following sequence: 
- key frame: [AUD, SPS, PPS, SEI, IDR]
- non-key frame: [AUD, SEI, non-IDR]
Or is there something I didn't follow?

MushMal, 2020-06-18:
Correct. As for the console playback, the backend does the re-packaging of the fragments into MPEG fragments which get loaded in your browser and submitted to the MSE (Media Source Extension) for the actual depackaging, decoding and playback.

I am just speculating about the MSE playback... Saying that perhaps SEI shouldn't be in the stream in between IDR and P-frames. In either case, when I remove the AUD and SEI NALus the playback works fine. I need to spend a little time on this

farleylai, 2020-06-19:
Got it but it seems like our code change may still have subtle inconsistence because mine also removes the AUD/SEI but the fragment decoding error remains.

MushMal, 2020-06-19:
I get normal playback with your frames in C producer sample (I had modified it) with this commit: https://github.com/awslabs/amazon-kinesis-video-streams-pic/commit/cdb9160cbb20b598b6f6516d552e5dcd0b82e075

Can I ask you to code review it?

farleylai, 2020-06-19:
Sure, it would be my pleasure and I will review it soon.

farleylai, 2020-06-20:
I rebuilt the code base with the latest commit and the behavior is consistent with yours. After going through the code changes, I sort of figured out why those NALUs before SPS/PPS may need to be filtered out. 

First, to my understanding,
- AVCC and AnnexB. frames should be convertible to each other regardless of the order of NALUs inside
- the only special concern is to compose the AVCC extradata or CPD out of one or more SPS/PPS extracted from NALUs before the IDR

Then the first assumption is that the console is able to handle or skip all the possible NALU types coming in an AVCC frame. If that is the case (likely according to @chehefen, the AVCC frames generated by GStreamer can be played), it would be preferred not to remove non-VCL NALUs as is.

Now, what should be the way to deal with the trailing zero byte since that would change the NALU size in AVCC frames? Based on a small experiment that converts the 1st key frame in AnnexB. to MKV using ffmpeg, the generated AVCC in MKV with NALUs containing the trailing zero byte which is counted in the NALU size. Therefore, semantically the trailing zero byte should be viewed as the last byte of the previous NALU instead of the start code with four zero bytes of the next NALU.

Next, from the code change, it seems like the extracted CPD includes everything before IDR.
If AUD/SEI are present, both would be included (without non-VCL removal), right? Consequently, `mkvgenSetCodecPrivateData(...)` -> `mkvgenAdaptCodecPrivateData(...)` -> `getVideoWidthAndHeightFromH264Sps(...)` may assume the first NALU is SPS? If that is the case, removing non-VCL NALUs would a must to get the width/height right, wouldn't it?

I am hoping the assumptions of the SPS/PPS in AnnexB. bitstreams could be relaxed while preserving the NALUs as much as possible for maximal interoperability as long as the console can support. Is it possible to simply parse and collect individual SPS/PPS NALUs in separate lists for making the CPD and get the video width/height later by treating the trailing zero byte as the last byte of the previous NALU? Then the test cases would be to compare the adapted AVCC CPD/frames with the counterparts converted by ffmpeg or GStreamer for consistence.
What do you think?

Last but not least after reading the code, the fragmentation is actually done on the client side.
Currently, the 1st fragment from `GetMedia` is usually at least 4s late. I am wondering if the live streaming latency from KVS can be further reduced to 2s or less if the network bandwidth allows since most surveillance applications would expect the timeliness and the traditional RTP streaming is capable of delivery at a lower latency. Maybe moving the mkv fragmentation to the server side could help a bit?

MushMal, 2020-06-22:
@farleylai there are two commits in the PR I submitted. The first one handles SPS/PPS extraction from the CPD (which might have other NALus) and handling of 4 zeroes and 1 as an Annex-B format. As we spoke, this is an invalid Annex-B format produced by a bad encoder, hence, I did put this behind a compile-time flag instead of a dynamic option. Understandably, it's the extra zero that's interfering with the following Annex-B start code and the code simply removes that zero by treating the Annex-B start code as 4 byte long - semantically it's the same as removing the last zero of the NALu. This is only done when a special compile flag is set and ONLY when we do Annex-B -> AvCC adaptation. The second change I did was for the Console playback fix. As is, the frames you had provided still failed to decode (at least on Chrome) MSE as the Console playback is nothing but re-packaging the MKV fragment to MPEG fragment and playing it back on the client web browsers MSE. I couldn't get more information on what exactly is failing during the decoding. However, I noticed that when I removed the non-VCL NALus the console playback worked fine. From what I understand, there are other contents attempted by @chehefen which work fine with SEI NALus but your frames failed. 

Few things to note. KVS SDK is designed to be independent on the Console playback. Some customers DO NOT adapt the Annex-B to AvCC as they don't need to have a Console playback in their solution. KVS playback solutions (Console playback, HLS, MPEG-DASH, clip generation API) operate on AvCC format with the most straight-forward CPD - one SPS and one PPS packaged in AvCC format (which SDK extracts and packages give the adaptation flags). Some hardware decoders (which are being used by browsers) are very sensitive to the format of the encoding whereas the software based encoders can be more forgiving.

Now, getting back to the changes, as you saw, these are ONLY enabled via compile time options. We need your help stabilizing these changes as we have no means of testing or reproducing these locally. The changes, once pushed in and compiled without the "special" compile flags will evaluate to only handling SPS and PPS in Annex-B -> AvCC adaptation when multiple other NALus are present which won't interfere with the existing clients. You might need to do some experimentation with the flags and see what will work for you and what your consumer story is.

Recommend: Try to run the latest and give some information whether it works on your hardware and in your consumer application (not sure what it is).

P.S. Regarding the latency. I wasn't sure I understood your question. GetMedia API is the real-time API and should have minimal latency - primarily the network bit latency + minor backend processing latency - sub-second ranges. The console playback (as HLS/MPEG-DASH) are not real-time playbacks - they are live playbacks, meaning they operate on the fragment granularity (as opposed to GetMedia which is bit-granularity based). So, for the Console playback, the overall latency will have the network bit latency (up and down), the small persistence and indexing latency on the backend, the fragment duration, the MSE playback latencies. The fragment duration here is likely the biggest contributor. At any rate, for more clarification on the latency, etc, please open a separate issue so it can be tracked separately.

farleylai, 2020-06-23:
@MushMal Thanks for making effort to improve the interoperability of KVS. Perhaps, I didn't make it clear earlier on. As commented previously, there was an inconsistency between your first commit (`5e9f8c`) and removing the trailing zero byte on my side that WORKS with the master branch WITHOUT filtering out AUD/SEI. However, you said those non-VCL NALUs need to be removed too for the console playback to work and then the 2nd commit (`cdb91`) was proposed for that. As a result, there is likely some other changes in the 1st commit (`5e9f8c`) preventing the console to play properly. After reviewing the commits further, the function `mkvgenExtractCpdFromAnnexBFrame(...)` changed in the 1st commit (`5e9f8c`) seems like the cause. In comparison,

- master (`1ac6146`): AUD is skipped first if any. Then the next two NALUs are assumed to be SPS/PPS to extract as CPD. What is left is simply ignored. This can work with the sequence of the NALUs in the key frame as AUD->SPS->PPS->SEI->IDR and produce CPD with SPS/PPS only. This behavior is consistent with ffmpeg where AUD/SEI are excluded from the extradata in the resulting MKV but only SPS/PPS.

-  1st commit (`5e9f8c`): there is a while loop iterating until the IDR is encountered and everything before the IDR is viewed as the CPD to set that potentially includes the additional AUD/SEI before and after SPS/PPS, producing different CPD from the master code.

As a consequence, this could justify why the 2nd commit (`cdb91`) works by removing non-VCL NALUs beforehand since the changed `mkvgenExtractCpdFromAnnexBFrame(..)` would produce the same CPD containing SPS/PPS only without AUD/SEI as the master branch. Nonetheless, it ends up more like a fix of the side effect from the 1st commit in an unexpected way. Otherwise, it makes sense to me if `mkvgenExtractCpdFromAnnexBFrame(...)` can be rewritten to explicitly check the NALU types for SPS/PPS instead of assuming the order of NALUs in the key frame.

Lastly, I understand the concern and am OK with you making the changes as compile time options though runtime warnings of non-compliant bitstreams could suffice with KVS more interoperable by default. This actually feels more like a marketing decision in that sense.

As for the latency, I agree it is better to discuss in a separate issue. So far, we observed the start stream time is within 2s from the time to call `GET_MEDIA` with `StartSelectorType=NOW` but the actual time to read the first frame out could take up to 3s, summing up to 4s latency on average before the first frame can be decoded and played. The 2nd and later frames can be read in a burst across the 1s fragment boundaries. Thus, we are wondering there may be additional server side buffering/aggregation of the fragments to send.

BTW, our consumer application is to analyze the video streams pulled from the cameras/NVR deployed on the customer side and stored on KVS. As more customers are involved, the higher chance we are exposed to bad encoders. Nonetheless, we don't usually have the opportunity to reject those potentially wild bitstreams.

MushMal, 2020-06-26:
The fixes are in PIC. We will be updating the C producer and CPP producer commits soon

MushMal, 2020-06-29:
Changes are in

closed 2020-06-29 by MushMal