awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #492: [QUESTION] - Encountered could not open file ./kvs_log_configuration error

yoyomans, 2020-06-18:
Hi all,
currently a bit lost with how to resolve this error I am getting...
```
log4cplus:ERROR could not open file ./kvs_log_configuration
ERROR - Unable to create Rotating Credential provider. Error status: 0x9
ERROR - Unable to create Rotating Credential provider. Error status: 0x9
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: Could not initialise supporting library.
Additional debug info:
/amazon-kinesis-video-streams-producer-sdk-cpp/src/gstreamer/gstkvssink.cpp(1374): gst_kvs_sink_change_state (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:
Failed to init kvs producer. Error: Unable to create Rotating Credential provider. Error status: 0x9
2020-06-18 23:38:21 [3069394960]
```

I have followed the instructions in the readme but when trying to start my data stream I get the above error, previously people who encountered this error had resolved it by setting the LD_LIBRARY_PATH env variable which I have done as instructed by the readme instructions and it is currently set to: `/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib`

In the `open-source/local/lib` directory there was no `kvs_log_configuration` file so I made one but that didn't fix my issue and I got the same error, tried resetting the access_key/secret_key credentials as well to no avail 

Any advice/help would be appreciated.

environment details on the system I am trying to run this are:
Device: Raspberry pi 4 b
OS: raspbian 
Folder plugin run from: `/amazon-kinesis-video-streams-producer-sdk-cpp`
Command line to start the data stream used: 
```
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=5120000 periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name=“{data-stream-name}" frame-timestamp=dts-only access-key=“{access-key}" secret-key=“{secret-key}" aws-region=“{aws-region}"
```

disa6302, 2020-06-18:
@yoyomans , 

The config file exists in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/samples

Can you try adding this this to your pipeline and see if it works? 

`log-config=samples/kvs_log_configuration`

MushMal, 2020-06-18:
There are two issues. 

1) The higher-level assets such as CPP SDK or GStreamer plugin use log4cplus for logging which requires the configuration file. Make sure you point it to https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_log_configuration though, this will primarily affect the logging only and shouldn't cause fatal errors.
2) You seem to want to use the file-based credential provider. This is a way to supply fresh credentials with the session token during the playback as the kvs gstreamer plugin doesn't have interactions with the application at runtime. In this way you should specify the file path to the credentials file which should contain the credentials in a single line separated by space. For more info on the format check out: https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/src/source/Common/FileCredentialProvider.c#L164

yoyomans, 2020-06-19:
Hi thanks for your replies, my problem was I hadn't set the AWS_ACCESS_KEY_ID and the other env variables needed and had just pasted the credentials directly into the command, when replacing the the strings in the command with the newly set env variables I got this new error
```
2020-06-19 17:35:57 [2895115360] INFO - describeStreamResultEvent(): Describe stream result event.
2020-06-19 17:35:57 [2895115360] INFO - writeHeaderCallback(): RequestId: 4f4e4163-a88d-429d-b1be-5c814e05f6c7
curlCompleteSync(): HTTP Error 403 : Response: {"message":"The security token included in the request is invalid."}
Request URL: https://kinesisvideo.us-east-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential={AWS_ACCESS_KEY_ID}/20200619/us-east-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=f5e4a4ab76881a2cc591826c84fdcf619a27f69fbe793479a4e4cf0e385c8d8d
    content-length: 31
    content-type: application/json
    host: kinesi
2020-06-19 17:35:58 [2904573024] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"message":"The security token included in the request is invalid."}
2020-06-19 17:35:58 [2904573024] INFO - describeStreamResultEvent(): Describe stream result event.
```

can you tell me the correct format I am meant to use to define these env variables when launching `gts-launch` I've seen some examples with `{}` either side of env variables and some examples with speech marks and some without.

here is what I am currently using to launch the data stream: 
```
gst-launch-1.0 -v v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc periodicty-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=teststream access-key={AWS_ACCESS_KEY_ID} secret-key={AWS_SECRET_ACCESS_KEY} aws-region={AWS_DEFAULT_REGION} 
```

disa6302, 2020-06-19:
@yoyomans ,

Please set the environment variables this way: 
`access-key=$AWS_ACCESS_KEY_ID`

yoyomans, 2020-06-19:
@disa6302 thank you so much for all your help, it appears to be working now :D

closed 2020-06-19 by yoyomans