awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #493: [BUG]

indoorrobot, 2020-06-19:
In the last few days we are encountering major time stamping issues:
The timestamp does not correlate the time the video was uploaded at - not when using producer timestamp nor with server timestamp.

To make it even weirder, the video in producer timestamp leaped to future time.... happens with ntp service disabled and after system time was verified.

**SDK version number**
2.1.0

 - OS: Linux
 - Processor: ATOM
- Streaming using gst with kvssink.

see example here: https://drive.google.com/file/d/1eNr5blC7Dhp8It1HPHi3Wk0fE-QzSoKL/view?usp=sharing

MushMal, 2020-06-19:
Both of the videos attached have no visual artifacts which is good. Regarding the producer timestamps... That will be whatever the producer device timestamps reports. If I remember it correctly, you guys use the KVS sink. It uses the default GETTIME() implementation which evaluates to defaultGetTime() API call in PIC utils/src/Time.c You can overwrite this API to suit your needs. I am puzzled though about the server-side timestamps. Those should be the true ingestion time timestamps. You might have some drift from the world time which should be equal to fragment duration + small processing time on the backed (usually a few ms) + the buffer duration on the producer device.

indoorrobot, 2020-06-21:
Hi,

Just a few clarifications.

* Up until the last few days everything worked fine. Both producer and the Server timestamps were correct when we pulled it using hls with PRODUCER_TIMESTAMP.
* The datetime on the device appears to be correct.
* As can be seen in the provided video, when using server timestamp the time is stuck even though the video is smooth.

chehefen, 2020-06-22:
@indoorrobot can you follow the steps here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#debug to get a mkv dump? Can you check whether the timestamp is incorrect inside the mkv?

MixMasterMitch, 2020-06-22:
It would also be beneficial to get a better sense of what is happening on the playback side too. Any chance you could open the network tab of the browser and send us the manifest that was loaded from KVS (should be URL path ending in `getDASHManifest.mpd`)?

MushMal, 2020-06-23:
Any further update on this?

indoorrobot, 2020-06-24:
Hi Guys,

I have updated the SDK to the latest release (3.0.0) and have set the buffer-duration.
For now, it seems I am unable to reproduce the issue.

I will keep you posted if something pops.

closed 2020-06-24 by indoorrobot