awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #494: [BUG] CMake error with "target_compile_definitions" and environment variables

Subburaj47, 2020-06-22:
**Describe the bug**
I believe the minimum_required_version in the available CMakes are 2.8. I face a problem with the cmake while cmake runs in 2.8 versions. 

1. The cmake function "target_compile_definitions" seems to be an unknown command.
2. "CMAKE_LANGUAGES_COMPILER_ENV_VAR" and "CMAKE_LANGUAGES_COMPILER" modules are not found.

**To Reproduce**
Have a 2.8 version of cmake available in the build system. Checkout the 3.0.0 SDK and trigger the cmake. 

**Expected behavior**
CMake would have got executed peacefully and subsequently the build.

**SDK version number**
SDK version used is 3.0.0

**Platform (please complete the following information):**
running in linux.

**Additional context**
Any option other than the cmake version upgrades would help
[amazon-kinesis-video-streams-producer-sdk-cpp.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4813699/amazon-kinesis-video-streams-producer-sdk-cpp.log)
.

Sean-Der, 2020-06-22:
@Subburaj47 Thanks for the report!

What Linux distro + version are you using? Will reproduce this today, thanks!

Subburaj47, 2020-06-23:
@Sean-Der : Running on a thirdparty system. Here are the details.
/mnt/ramdisk/root # cat /proc/version
Linux version 3.10.104+ (charlie@isbu-virtual-machine) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-2) ) #1 PREEMPT Mon May 18 16:01:21 CST 2020

MushMal, 2020-06-26:
Seems that CMake 2.8.11 contains target_compile_definitions

https://cmake.org/cmake/help/v2.8.11/cmake.html#command:target_compile_definitions

and this is the earliest version that started to have the directive.

We should update the make file to reflect this

MushMal, 2020-08-03:
This is fixed

closed 2020-08-03 by MushMal