awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #499: Windows 7 and KvsSink

lpawlaczyk, 2020-06-29:
I have been trying to run KvsSink on Windows 7 with latest GStreamer. KvsSink fails with a null pointer exception from GLib. Are there any reasons why KvsSink would not work on Windows 7 if it was compiled using latest VS 2019 on Windows 10?

MushMal, 2020-06-29:
@lpawlaczyk could you please include a little more information about your usage:

* Are you using kvs sink from gst-launch-1.0 command line or embedded within your application?
* Can you reproduce the issue with stock gst-launch-1.0 and the kvs sink?
* Can you reproduce the issue when replacing kvs sink with fake sink?

Could you please capture debug KVS logs and GST logs? You can use GST_DEBUG=4 environment variable.

lpawlaczyk, 2020-06-30:
1. I am using KvsSink from command line with the following setup:

C:\gstreamer\1.0\x86_64\bin>gst-launch-1.0 -v filesrc location=file.mp4 ! qtdemux ! h264parse ! avdec_h264 ! videorate ! video/x-raw,framerate=1/1 ! videoscale!  video/x-raw,width=416,height=416 ! x264enc bframes=0 key-int-max=1 bitrate=512 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvs
sink stream-name="TestLP" storage-size=512 access-key="<secret>" secret-key="<secret>" aws-region="us-east-2"

2. The output is added to this report.

3. The error is not present when using fakesink. I am using glimagesink for presentation and it works fine.

[bugreport.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4850044/bugreport.txt)
[kvsoutput.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4850045/kvsoutput.txt)

closed 2020-06-30 by lpawlaczyk

MushMal, 2020-06-30:
@lpawlaczyk it seems that you've resolved the issue without any explanation.

I am looking at the bug report and I can't see anything useful - I was hoping for some core dump or at least stack trace. From what I can decipher in the bug report is that there is a null dereference somewhere.

The logs indicate a timeout calling PutMedia backend API. As I don't see the verbose logs I can only assume that if you were successful calling the control plane APIs (ex: DescribeStream) and failed for data plane APIs (PutMedia) then you have a permission problem possibly? Though, that would not result in a timeout so I am puzzled.

Can you shed more light into this?