awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #500: Fragment duration is varied even though I have already changed in kvssink element (gstreamer)

supavit28, 2020-06-30:
**Issue**
I can't set fragment duration while using kvssink from gstreamer

**Steps to reproduce the behavior:**
1. Run example code with some more argument to match my requirement. (https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html)
```bash
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! clockoverlay halignment=left valignment=top time-format="%Y-%m-%d %H:%M:%S" font-desc="Sans, 6" ! videoconvert ! video/x-raw,format=I420,width=1280,height=720,framerate=20/1 ! omxh264enc control-rate=1 target-bitrate=4096000 periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720,framerate=20/1,profile=baseline ! kvssink stream-name="my-stream-name" access-key="my-access-key" secret-key="my-secret-key" aws-region="my-region-name" fragment-duration="3000"
```
or
```bash
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! clockoverlay halignment=left valignment=top time-format="%Y-%m-%d %H:%M:%S" font-desc="Sans, 6" ! videoconvert ! video/x-raw,format=I420,width=1280,height=720,framerate=20/1 ! omxh264enc control-rate=1 target-bitrate=4096000 periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720,framerate=20/1,profile=baseline ! kvssink stream-name="my-stream-name" access-key="my-access-key" secret-key="my-secret-key" aws-region="my-region-name" fragment-duration=3000
```
2. It will get KVS setting in output terminal as follow
```bash
2020-06-30 15:35:47 [3069394960] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Stream name: raspberry
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Streaming type: STREAMING_TYPE_REALTIME
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Content type: video/h264
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Max latency (100ns): 600000000
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Fragment duration (100ns): 30000000
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Key frame fragmentation: Yes
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Use frame timecode: Yes
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Absolute frame timecode: Yes
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Nal adaptation flags: 0
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Average bandwith (bps): 4194304
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Framerate: 25
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Buffer duration (100ns): 1200000000
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Replay duration (100ns): 400000000
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Connection Staleness duration (100ns): 600000000
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Store Pressure Policy: 1
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       View Overflow Policy: 1
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Segment UUID: NULL
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Frame ordering mode: 0
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo(): Track list
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Track id: 1
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Track name: kinesis_video
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Codec id: V_MPEG4/ISO/AVC
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Track type: TRACK_INFO_TYPE_VIDEO
2020-06-30 15:35:47 [3069394960] DEBUG - logStreamInfo():       Track cpd: NULL
```
and KVS metrics output as follow
```bash
2020-06-30 15:35:54 [2807034976] DEBUG - Kinesis Video client and stream metrics
        >> Overall storage byte size: 134217728
        >> Available storage byte size: 133592977
        >> Allocated storage byte size: 624751
        >> Total view allocation byte size: 144080
        >> Total streams frame rate (fps): 20
        >> Total streams transfer rate (bps): 24343864 (23773 Kbps)
        >> Current view duration (ms): 300
        >> Overall view duration (ms): 2249
        >> Current view byte size: 133475
        >> Overall view byte size: 622911
        >> Current frame rate (fps): 20.6096
        >> Current transfer rate (bps): 24343864 (23773 Kbps)
```
After I check with KVS console, the overall view duration is fragment duration shown in console (and also from SDK). 
3. It is obvious that I can't set fragment duration from gstreamer example.
4. Furthermore, if I increase resolution (with other setting is similar), I get different fragment duration.
For example, I set resolution = 1920x1080, I get fragment duration equals to 8 seconds (Sorry I have capture the screen on this issue)

**SDK version number**
I pull from this repository on 16 June 2020 (GMT +7.00): https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git

**Platform (please complete the following information):**
 - OS: Raspbian
 - Processor:  ARM v8 (Raspberry Pi 4 Model B 2 GB) 
 - Version:
> PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

**Additional**
- Do I use wrong command to set fragment duration ??
- There is any limitation on fragment duration (such as maximum fragment duration). I can found only default value in documentation (2000 ms)
- If not, then how to set fragment duration ?

MushMal, 2020-06-30:
@supavit28 it seems that you are using the hardware accelerated encoder so the RPi should be able to handle 1080p resolution without major CPU impact. I can't quite tell what's causing the fragment duration changes when you increase the resolution - perhaps you need to change a few encoder params including the key-frame interval param and the bitrate. Even with hardware encoder, the CPU will have a non-linear impact when selecting larger resolutions.

Now, to get to the fragmet_duration parameter. When using kvs SDK (plugin) with video like h264, the SDK is configured to have key_frame_fragmentation = TRUE. This means that the encoder generates IDR frames (key-frames) which will drive the fragmentation. So, in your case, you will need to set the key-frame intervals in your encoder. Perhaps this can help? 

https://gist.github.com/mitsuhito/ad3ae3e341726687457c#file-gistfile1-sh-L108

supavit28, 2020-07-01:
@MushMal Thank you for your suggestion. Now I try to tune some parameters on both encoder and kvssink. It seems that there is some issue to set fragment duration. I will report on the issue and the solution once I found it because I hope it will help someone who face this problem too.
If I still struggle, I will comment on this issue again. Can you please open this issue for a couple of days?

MushMal, 2020-07-01:
Sure, please feel free to provide updates. Normally, with RPi and Raspbian OS-es we haven't had much problems using the hardware encoder. Please take a look at this page with some RPi sample pipelines: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md

supavit28, 2020-07-02:
Now I am be able to set fragment duration (with limitation) by setting interval-intraframes to match with framerate (as show in below code)
```bash
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! clockoverlay halignment=left valignment=top time-format="%Y-%m-%d %H:%M:%S" font-desc="Sans, 6" \
! videoconvert ! video/x-raw,format=I420,width=1280,height=720,framerate=10/1 ! omxh264enc control-rate=1 target-bitrate=1024000 interval-intraframes=30 inline-header=FALSE \
! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720,framerate=10/1,profile=baseline \
! kvssink stream-name="my-stream-name" access-key="my-access-name" secret-key="my-secret-name" aws-region="my-region-name"
```
The result from above code is 
```bash
2020-07-02 15:21:48 [2807034976] DEBUG - Kinesis Video client and stream metrics
        >> Overall storage byte size: 134217728
        >> Available storage byte size: 133790915
        >> Allocated storage byte size: 426813
        >> Total view allocation byte size: 144080
        >> Total streams frame rate (fps): 21
        >> Total streams transfer rate (bps): 27639808 (26992 Kbps)
        >> Current view duration (ms): 0
        >> Overall view duration (ms): 3300
        >> Current view byte size: 60887
        >> Overall view byte size: 425533
        >> Current frame rate (fps): 21.395
        >> Current transfer rate (bps): 27639808 (26992 Kbps)
```
But here are limitations which I found
1. If I change interval-intraframes=10 (expected fragment duration = 1 s), it gets both 1s and 2s fragment on terminal log.
```bash
2020-07-02 15:26:02 [2817184864] DEBUG - Kinesis Video client and stream metrics
        >> Overall storage byte size: 134217728
        >> Available storage byte size: 134044787
        >> Allocated storage byte size: 172941
        >> Total view allocation byte size: 144080
        >> Total streams frame rate (fps): 41
        >> Total streams transfer rate (bps): 33554432 (32768 Kbps)
        >> Current view duration (ms): 1300
        >> Overall view duration (ms): 1300
        >> Current view byte size: 172461
        >> Overall view byte size: 172461
        >> Current frame rate (fps): 41.0646
        >> Current transfer rate (bps): 33554432 (32768 Kbps)
2020-07-02 15:26:03 [2817184864] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-02 15:26:04 [2807034976] INFO - writeHeaderCallback(): RequestId: f78d3efc-7f31-e21a-a0fa-a2a075bab2a3
2020-07-02 15:26:04 [2807034976] DEBUG - Kinesis Video client and stream metrics
        >> Overall storage byte size: 134217728
        >> Available storage byte size: 133888192
        >> Allocated storage byte size: 329536
        >> Total view allocation byte size: 144080
        >> Total streams frame rate (fps): 27
        >> Total streams transfer rate (bps): 31876704 (31129 Kbps)
        >> Current view duration (ms): 1900
        >> Overall view duration (ms): 2400
        >> Current view byte size: 261721
        >> Overall view byte size: 328605
        >> Current frame rate (fps): 27.6661
        >> Current transfer rate (bps): 31876704 (31129 Kbps)
```
This problem will not happen when setting interval-intraframes to get expected fragment duration more than 1 second.

2. If I increase resolution to 1366x768 and lower bitrate to 500kpbs while leave other parameters same as the code shown above. The fragment duration is increase to 6.6s.

3. Somehow if I increase resolution to 1600x900, it gets fragment duration as 4s. This leads me confusing.

__Roughly current status__
- So as of now in my case, I can't set fragment duration via kvssink.
- However, I can set using interval-intraframes that match with framerate.
- The working code has maximum stream resolution as 1280x720.

lherman-cs, 2020-07-02:
@supavit28 from your comment, it sounds like you're referring Current view duration as fragment duration. To make sure that we're on the same page, current view duration is not related to fragment duration.

Current view duration is simply a sum of frame durations that our internal buffer has at the moment the metrics got taken. So, for 2 and 3, these can definitely affect the current view duration, because those parameters can affect how fast we process your source and the number can be really varied everytime since it also depends on your network speed.

MushMal, 2020-07-03:
I am confused too. Per @lherman-cs suggestion, how do you determine the fragment duration? One way to do it is to look at the console playback which will tell you the fragment duration. If we really are talking about fragment durations (as opposed to current view duration which is about the duration of accumulated frames in the producer buffer which have not yet been streamed) then different encoders have different ways of specifying it. For example, in case of some hardware encoders, you specify the desired frame rate and specify the GoP side by specifying key-frame intervals in the number of frames. Say, you have 25fps frame rate configuration and 1:50 key-frame intervals. This will generate 2 second fragments (one key-frame every 50 frames) provided that the encoder can keep up with 25 fps. Some encoders can and do slow down with the increase in resolution so every 50th frame might not necessarily evaluate to 2 seconds any longer as the encoder can't keep up the frame rate.

For these question, the best source of information is your encoder manufacturer as KVS is not related to the encoders themselves.

supavit28, 2020-07-03:
I understand that the problem of fragment duration now is come from encoder not KVS.
I will try to test on better internet connection hope it fix my problem. Thank you both of you. @MushMal @lherman-cs 

About the **overall view duration** show on kvs log, I can confirm that it is the same value same as **fragment duration** which show on console.
For example, I running below code.
```bash
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! clockoverlay halignment=left valignment=top time-format="%Y-%m-%d %H:%M:%S" font-desc="Sans, 6" \
! videoconvert ! video/x-raw,format=I420,width=1280,height=720,framerate=10/1 \
! omxh264enc control-rate=1 target-bitrate=1024000 interval-intraframes=20 inline-header=FALSE \
! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720,framerate=10/1,profile=baseline \
! kvssink stream-name="my-steam-name" access-key="my-access-key" secret-key="my-secret-key" aws-region="my-region-name" fragment-duration=3000
```
The encoder will override the fragment duration setting from kvssink, so the fragment should be 2 seconds duration in my case. 

**Log screenshot**

[image]

**Console screenshot**

[image]

**Result export from my code using boto3**

[image]

MushMal, 2020-07-03:
Just to note that the KVS plugin configures the sdk (for video stream content type) to have key-frame-fragmentation = true which means that the fragment-duration parameter is not used as the fragmentation is driven by the encoder in this case

MushMal, 2020-07-06:
What's the latest on this issue? Can we proceed and close this issue as resolved?

supavit28, 2020-07-07:
Sorry for inactive a couple of days. I am on holiday. I think it can be closed now because it get clearer on how to set fragment duration.

closed 2020-07-07 by supavit28