awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #501: How to pass user id and user password of RTSP Stream?

thasin96, 2020-07-02:
I am trying amazon-kinesis-video-streams-producer-sdk-cpp example in docker environment. How can I pass user-id and user-pw argument of RTSP stream? If I directly pass stream credentials, I am getting 404 error. If I didn't use credentials, I am unable to upload stream. Any idea regarding this issue would be helpful. Feel free to ask further details, If you require.

lherman-cs, 2020-07-02:
@thasin96 Do you mind to share which exact example are you using? If you're using `kvs_gstreamer_sample.cpp` example, unfortunately, you need to modify the example here, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L880-L883. You need to pass `user-id` and `user-pw` to `rtspsrc` plugin where you can find more details here, https://gstreamer.freedesktop.org/documentation/rtsp/rtspsrc.html.

thasin96, 2020-07-03:
@lherman-cs - I tried `kinesis_video_gstreamer_sample_multistream_app` for that syntax is `AWS_ACCESS_KEY_ID=SAMPLEKEY AWS_SECRET_ACCESS_KEY=SAMPLESECRET ./kinesis_video_gstreamer_sample_multistream_app base-stream-name rtsp-url-file-name`

In this above exact example, how to pass `user-id` and `user-pw`? I tried passing authentication along wirh URP that also didn't help.

lherman-cs, 2020-07-03:
You need to modify the example source code and recompile it (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L880-L883):

Before:
```
    g_object_set(G_OBJECT (source),
                 "location", data->rtsp_url.c_str(),
                 "short-header", true, // Necessary for target camera
                 NULL);
```

After:
```
    g_object_set(G_OBJECT (source),
                 "location", data->rtsp_url.c_str(),
                 "short-header", true, // Necessary for target camera
                 "user-id", "<put your username here>",
                 "user-pw", "<put your password here>",
                 NULL);
```

disa6302, 2020-07-13:
@thasin96 , 

Can you close the issue if the question is answered? Thanks!

closed 2020-07-13 by thasin96