awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #502: [BUG] Can't find install-script after make

Anshulkaaya, 2020-07-02:
mkdir -p amazon-kinesis-video-streams-producer-c/build; 
cd amazon-kinesis-video-streams-producer-c/build; 
cmake ..
make

After this step, can't find install-script after make. Please check snapshot
[image]

lherman-cs, 2020-07-02:
@Anshulkaaya I'm sorry for the inconvenience. Recently, we updated our build system to remove custom scripts. The script that you're referring, `install-script`, is no longer necessary since calling `cmake ..` will do the same thing.

lherman-cs, 2020-07-02:
Since there's no actual bug, I'll mark this as resolve. The documentation fix is being tracked here, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/503.

Please feel free to reopen this issue if you need further assistance.

closed 2020-07-02 by lherman-cs