awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #504: [QUESTION]

jeromewe, 2020-07-03:
make[2]: *** [CMakeFiles/project_libopenssl.dir/build.make:91：build/src/project_libopenssl-stamp/project_libopenssl-download] 错误 1
make[1]: *** [CMakeFiles/Makefile2:73：CMakeFiles/project_libopenssl.dir/all] 错误 2
make: *** [Makefile:84：all] 错误 2
CMake Error at open-source/amazon-kinesis-video-streams-producer-c/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:64 (build_dependency)

-- Configuring incomplete, errors occurred!

lherman-cs, 2020-07-04:
@jeromewe Sorry for the inconvenience. Could you try to rerun the cmake command again? We've been seeing this strange issue where you need to configure it twice to make it work. Of course this is not ideal, but it should at least solve your issue for now. We will keep this ticket open to keep track this issue to deliver a long term solution.

lherman-cs, 2020-07-27:
Since the issue has gone stalled, I'll mark this as resolved. Please feel free to reopen it for further assistance!

closed 2020-07-27 by lherman-cs

amitrjaiswal1, 2020-07-27:
I am facing this build error.....

CMake Error at C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/tmp/project_libopenssl-gitclone.cmake:52 (message):  Failed to update submodules in:  'C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl'ninja: build stopped: subcommand failed.