awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #505: Can't compile and install the GStreamer sample in the kinesis-video-native-build

uyennguyen16900, 2020-07-08:
I ran the command `brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus` to compile and install the GStreamer sample in the kinesis-video-native-build directory.
When I ran `./min-install-script`, I got this error:
```
ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd, file was built for unsupported file format ( 0x2D 0x2D 0x2D 0x20 0x21 0x74 0x61 0x70 0x69 0x2D 0x74 0x62 0x64 0x2D 0x76 0x33 ) which is not the architecture being linked (x86_64): /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd
    ld: dynamic main executables must link with libSystem.dylib for architecture x86_64
    clang-4.0: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_d3138] Error 1
    make: *** [cmTC_d3138/fast] Error 2
```
If you have seen similar issues or have an idea on a solution, please help. Any help would be much appreciated!

MushMal, 2020-07-08:
You might want to re-install Apple Commend Line Tools for xCode. Looks like a tools issue

uyennguyen16900, 2020-07-08:
I ran `sudo rm -rf /Library/Developer/CommandLineTools` and then `xcode-select --install` to reinstall Apple Commend Line Tools, but it didn't solve the problem. 
```
Run Build Command(s):/usr/bin/make cmTC_841fb/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_841fb.dir
/build.make CMakeFiles/cmTC_841fb.dir/build
    Building C object CMakeFiles/cmTC_841fb.dir/testCCompiler.c.o
    /Users/uyennguyen/opt/anaconda3/bin/x86_64-apple-darwin13.4.0-clang   -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fP
IE -fstack-protector-strong -O2 -pipe  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk   -o CMakeFiles/cmTC_841fb.dir
/testCCompiler.c.o   -c /Users/uyennguyen/dev/microsoft-summer-challenge/tinydoor/kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/
kinesis-video-native-build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_841fb
    /usr/local/Cellar/cmake/3.17.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_841fb.dir/link.txt --verbose=1
    /Users/uyennguyen/opt/anaconda3/bin/x86_64-apple-darwin13.4.0-clang -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fPIE -fstack-protector-strong -O2 -pipe  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-hea
derpad_max_install_names -Wl,-pie -Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs  CMakeFiles/cmTC_841fb.dir/testCCompiler.c.o
-o cmTC_841fb
    ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd, file was built for unsupp
orted file format ( 0x2D 0x2D 0x2D 0x20 0x21 0x74 0x61 0x70 0x69 0x2D 0x74 0x62 0x64 0x2D 0x76 0x33 ) which is not the architecture bein
g linked (x86_64): /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd
    ld: dynamic main executables must link with libSystem.dylib for architecture x86_64
    clang-4.0: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_841fb] Error 1
    make: *** [cmTC_841fb/fast] Error 2

```

MushMal, 2020-07-08:
Are you using the latest SDK? I see kinesis-video-native-build in the path so I am wondering whether you have older version of the build intermixed with the new structure. Though, this wouldn't cause the linkage issue with the system library.

Can you please clone the latest repository and follow the instructions in the readme to compile? We don't have scripts for compiling any longer - the latest build uses CMake capabilities for building open-source components.

uyennguyen16900, 2020-07-08:
I deleted my project and cloned the latest repo. I ran `mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake ..` and got this error:
```
Run Build Command(s):/usr/bin/make cmTC_ee2ba/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_ee2ba.dir
/build.make CMakeFiles/cmTC_ee2ba.dir/build
    Building C object CMakeFiles/cmTC_ee2ba.dir/testCCompiler.c.o
    /Users/uyennguyen/opt/anaconda3/bin/x86_64-apple-darwin13.4.0-clang   -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fP
IE -fstack-protector-strong -O2 -pipe  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk   -o CMakeFiles/cmTC_ee2ba.dir
/testCCompiler.c.o   -c /Users/uyennguyen/dev/microsoft-summer-challenge/tinydoor/kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/
build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_ee2ba
    /usr/local/Cellar/cmake/3.17.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_ee2ba.dir/link.txt --verbose=1
    /Users/uyennguyen/opt/anaconda3/bin/x86_64-apple-darwin13.4.0-clang -march=core2 -mtune=haswell -mssse3 -ftree-vectorize -fPIC -fPIE

 -fstack-protector-strong -O2 -pipe  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-hea
derpad_max_install_names -Wl,-pie -Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs  CMakeFiles/cmTC_ee2ba.dir/testCCompiler.c.o
-o cmTC_ee2ba
    ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd, file was built for unsupp
orted file format ( 0x2D 0x2D 0x2D 0x20 0x21 0x74 0x61 0x70 0x69 0x2D 0x74 0x62 0x64 0x2D 0x76 0x33 ) which is not the architecture bein
g linked (x86_64): /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd
    ld: dynamic main executables must link with libSystem.dylib for architecture x86_64
    clang-4.0: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_ee2ba] Error 1
    make: *** [cmTC_ee2ba/fast] Error 2

```

chehefen, 2020-07-08:
you need to do `cmake .. -DBUILD_DEPENDENCIES=OFF` if you have used brew to install everything. Also remember to set PKG_CONFIG_PATH  to point to your openssl's pc file before running cmake. You can get the path by running `brew info openssl`

uyennguyen16900, 2020-07-08:
It works! Thank you for your help. But now I run into another error. 
[image]

Any help would be greatly appreciated!

closed 2020-07-09 by uyennguyen16900