awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #506: No such element or plugin 'kvssink'

uyennguyen16900, 2020-07-09:
After running `make` in the build directory I executed CMake, I followed the instructions to set the environment variables. 
I got `No such element or plugin 'kvssink'` when I ran `gst-inspect-1.0 kvssink`

[image]

I don't know if my build failed or GST_PLUGIN_PATH is not set properly. Any help is much appreciated!

disa6302, 2020-07-09:
@uyennguyen16900 ,

Did you run cmake with `-DBUILD_GSTREAMER_PLUGIN=TRUE` ? 
Can you check if gstkvssink library exists in the build directory?

uyennguyen16900, 2020-07-09:
I did run `cmake ..  -DBUILD_GSTREAMER_PLUGIN=TRUE `.
I don't see gstkvssink library exists in the build directory.

[image]

chehefen, 2020-07-09:
do you have gstreamer in your system?

uyennguyen16900, 2020-07-09:
I did download gstreamer

closed 2020-07-11 by uyennguyen16900

brahama, 2020-08-09:
I am having same issue here. Brew install gstreamer and still can not find.

franktseng0718, 2020-08-17:
@uyennguyen16900  I met the same problem as you. How did you solve this problem finally?

AtiqZaman, 2020-10-06:
I am facing the same issue

> After running `make` in the build directory I executed CMake, I followed the instructions to set the environment variables.
> I got `No such element or plugin 'kvssink'` when I ran `gst-inspect-1.0 [...]

I am facing the same issue, can anybody help?

MushMal, 2020-10-06:
> I am facing the same issue
>
>
> I am facing the same issue, can anybody help?

You might want to do some debugging yourself. There are plenty of other issues closed in the repository with the same issue. Ensure you specify to build gstreamer in the cmake command, make sure it finishes the build correctly, check to see if you have the ginary, set the plugin path and the ld lib path properly and it should work. If you still have issues, please cut a separate issue with detailed report on the above points.

AtiqZaman, 2020-10-15:
Thanks @MushMal for your reply and giving me a push to to debug by my own, after debugging, i have solved the issue, now my video from web camera is streaming to kinesis video stream.
Problem was that my GST_PLUGIN_PATH was not set properly, 
Althoght i have set the path through below commands :
export GST_PLUGIN_PATH='pwd'/build
export LD_LIBRARY_PATH='pwd'/open-source/local/lib
But that was not actually set because, after setting the above path when i run the below command 
echo  $GST_PLUGIN_PATH
It shows me another path which was not pointing to the kvssing 
So 
What i did is that i loged in as a super user by running below command 
$ sudo su

and then 

export GST_PLUGIN_PATH='pwd'/build
export LD_LIBRARY_PATH='pwd'/open-source/local/lib

After that when i run this command.   $ gst-inspect-1.0 kvssink
it showed me exactly what it has to..
and finally i run below command and my video streaming was started :

$ gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=atiqVideoStream storage-size=512 access-key='your access key' secret-key='your secret access key'
+1:

Void-Mohit, 2021-04-16:
**In my case plugin is builded  in amazon-kinesis-video-streams-producer-sdk-cpp folder so I removed /build in export path and it works

```export GST_PLUGIN_PATH='pwd'
export LD_LIBRARY_PATH='pwd'/open-source/local/lib```**

Tseema, 2022-02-15:
> Thanks @MushMal for your reply and giving me a push to to debug by my own, after debugging, i have solved the issue, now my video from web camera is streaming to kinesis video stream. Problem was [...]

this works like a charm. thank you