awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #507: [QUESTION] Backwards on KVS Pipeline using docker (2)

FlorianRuen, 2020-07-09:
Hi there,

I'm opening new case on a problem that is present from a long time ago (original post: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/466#issuecomment-655964920)
(Maybe if possible, can you reopen same case to centralize discussions ?)

@MushMal @lherman-cs  I retry to found the problem on this case. After many investigations, I try to pull your docker image again, and to keep exactly the same pipeline that work when camera and internet are the same network, there is still backward when camera and internet aren't the same networks.

I run this inside a docker container directly (without using my app, to try if the problem came from this application), but the problem still the same.

So I can't be sure the problem is from the pipeline or from the docker image that you provide. You haven't any way to try this configuration on your side ?

Thanks again

FlorianRuen, 2020-07-09:
To complete my first message:

- I try to watch the HLS stream when the pipeline is running (it means, when video came on KVS service using pipeline and by running command directly in docker container), the backwards/loop (call it what you want) happening.

- But when I watch the HLS stream when the pipeline isn't running (I watch the buffer, or stream saved), the backwards/loop not happening !! (It's exactly the same video feed, but no more video come on KVS, I stop my docker container on this case)

It's like there is an edit to remove backwards between two use cases (or the frames are put in the right order between)

I record a video on the two use cases, you can find here:

- With loop (case 1) = https://eyesr-files.s3.eu-central-1.amazonaws.com/stream_hls_with_pipeline_running.mkv
- Without loop (case 2) = https://eyesr-files.s3.eu-central-1.amazonaws.com/stream_hls_without_pipeline_running.mkv

Hope we can find a solution

MushMal, 2020-07-09:
@FlorianRuen having different number of BUFFERING ACKs and PERSISTED ACKs indicates an error and re-transmission. The SDK is configured to re-play form the buffer (from last ACK) on error which can result in duplicated fragments and it's by design.

Can you try changing the fragment selector type to Producer https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_HLSFragmentSelector.html#KinesisVideo-Type-reader_HLSFragmentSelector-FragmentSelectorType

and modify the continuity mode to see what your result is.

Also, try to capture the producer side error that might help explain what the issue is with your network

FlorianRuen, 2020-07-09:
@MushMal I need to put this parameter to the pipeline directly ? Just by adding FragmentSelectorType=PRODUCER_TIMESTAMP ? Or I need to change this on docker image or something else ?

Thanks for your help, hope we will find a solution to this issue.

chehefen, 2020-07-09:
@FlorianRuen can you send us the mkv file captured using KVS_DEBUG_DUMP_DATA_FILE_DIR=/path/to/dir. I've asked this in the previous channel but it seemed that you havent give us the file. Producer just sends the content without any modifications. So we wanna make sure of the integrity of the upstream first.

MushMal, 2020-07-09:
> @MushMal I need to put this parameter to the pipeline directly ? Just by adding FragmentSelectorType=PRODUCER_TIMESTAMP ? Or I need to change this on docker image or something else ?
> 
> Thanks for your help, hope we will find a solution to this issue.

This is on the Get HLS session call side

FlorianRuen, 2020-07-10:
@MushMal By default, when I use the KVS Dashboard, the default value is PRODUCER, so I try to change it to SERVER and back to PRODUCER, I record a video file to show you that it's exactly the same problem. [https://eyesr-files.s3.eu-central-1.amazonaws.com/stream_changement_producer_vers_server_et_inversement.mkv](https://eyesr-files.s3.eu-central-1.amazonaws.com/stream_changement_producer_vers_server_et_inversement.mkv)

But I got a question: It's normal that when I make a:

```
docker start CONTAINER (container already created and started one time with the pipeline)
docker exec CONTAINER /bin/bash
```
The pipeline is running by default ? Because on this morning, I just plug the camera and launch this, and I can see live video on KVS Dashboard (without any another command on an already existing container)

@chehefen Of course, here is the mkv file you asked : https://eyesr-files.s3.eu-central-1.amazonaws.com/video_debug_kvs_loop_brut.mkv / generated using the following pipeline, but there is no loop on the short mkv clip, and when I record this one, there is loops on the KVS Dashboard ... the loops happend during the transmit process ? because when this short clip is recorded, there is the same logs problem (with ACK different)

`KVS_DEBUG_DUMP_DATA_FILE_DIR=test/ gst-launch-1.0 rtspsrc location={RTSP_URL} short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name={STREAM_NAME} storage-size=256 access-key={KEY} secret-key={SECRET} aws-region={REGION}
`

Thanks for your help,

chehefen, 2020-07-13:
you can build the producer easily from scratch on raspberry pi. The steps are here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#configure. Can you try build the latest producer and see if the issue persists? You need to apt-get the dependencies first by doing `$ sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
`
Then build by doing `cmake .. -DBUILD_DEPENDENCIES=OFF`

FlorianRuen, 2020-07-14:
I try it today. EVerything is fine (no error), but when I try to run the command gst-inspect,the message is "No element kvssink", the documentaiton say that came from the export which failed, but I try many times, with good path and seems not working.

The compilation is successfully finish with message:
```

[ 94%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/src/KinesisVideoStream.cpp.o
[ 95%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/src/StreamDefinition.cpp.o
In file included from /usr/include/c++/8/vector:69,
                 from /home/eyesr_root/amazon-kinesis-video-streams-producer-sdk-cpp/src/StreamDefinition.h:10,
                 from /home/eyesr_root/amazon-kinesis-video-streams-producer-sdk-cpp/src/StreamDefinition.cpp:1:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::emplace_back(_Args&& ...) [with _Args = {com::amazonaws::kinesis::video::StreamTrackInfo__}; _Tp = com::amazonaws::kinesis::video::StreamTrackInfo__; _Alloc = std::allocator<com::amazonaws::kinesis::video::StreamTrackInfo__>]’:
/usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<com::amazonaws::kinesis::video::StreamTrackInfo__*, std::vector<com::amazonaws::kinesis::video::StreamTrackInfo__> >’ changed in GCC 7.1
    _M_realloc_insert(end(), std::forward<_Args>(__args)...);
    ^~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {com::amazonaws::kinesis::video::StreamTrackInfo__}; _Tp = com::amazonaws::kinesis::video::StreamTrackInfo__; _Alloc = std::allocator<com::amazonaws::kinesis::video::StreamTrackInfo__>]’:
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::vector<com::amazonaws::kinesis::video::StreamTrackInfo__>::iterator’ {aka ‘__gnu_cxx::__normal_iterator<com::amazonaws::kinesis::video::StreamTrackInfo__*, std::vector<com::amazonaws::kinesis::video::StreamTrackInfo__> >’} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
[ 96%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/src/StreamTags.cpp.o
[ 96%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/src/common/PutFrameHelper.cpp.o
[ 97%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/src/credential-providers/IotCertCredentialProvider.cpp.o
[ 98%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/src/credential-providers/RotatingCredentialProvider.cpp.o
[100%] Linking CXX shared library libKinesisVideoProducer.so
[100%] Built target KinesisVideoProducer
```

So no errors appear during the compilation process.

FlorianRuen, 2020-07-14:
I try to rebuild everything from scratch according to your previous message, and it seems there is no backwards and no loops on the KVS dashboard.

By the way, when I stop the pipeline by CTRL+C on raspberry pi, the stream stop immediatly on KVS dashboard (which is not the case using the previous docker image)

D I need to create a docker image from the rebuilt sources to use it with my application ? or do you advise me to stay with version C ?

chehefen, 2020-07-14:
you can move the process into background: https://stackoverflow.com/questions/46283647/how-to-move-a-running-process-to-background-unix

FlorianRuen, 2020-07-14:
I know this, but it's not my question.

My question is: today I use a Python script to get data from a database, building pipeline with those informations and running docker image with command line to send stream to KVS. Do I need to adapt my code to work with the C library directly (instead of docker image) or I can create a new docker image from the sources I built to fix the issue ?

If i use C library and run the command with Python script, I haven't any ways to know if the pipeline is running and so one (or I need to do a lot of works to adapt the code to know of all stats)

Because the docker image provided by KVS Team isn't working, if the problem happend with your docker image and not when I build from scratch, it's means the issue came from the docker image ?

chehefen, 2020-07-15:
The raspberry pi docker image doesnt do anything other than providing a built library. The end result should be the same as building the library from scratch. Not sure what you mean by doing a lot of works to adapt.

FlorianRuen, 2020-07-16:
AS I said, today I'm using my own python script to get pipeline detail (remote id, aws credentials and so one) from a webapp (and a database), to build the pipeline and to run it (and apply some other function when the pipeline is running to auto-stop it and so one)

But all this process is done to work with a docker image (docker connect, docker create, docker start ...), if I can't do that because YOUR image doesn't work correctly (new time, it's a problem on your side, you said your docker image is just an image from the build library, but how do you explain when I build from scratch, everything works, and it's not the case when I use your docker image ? Not the latest version ?)

This is what I mean by "a lot of works to adapt", because I need to switch all my code from docker to launch the pipeline using C librarie directly

Is it clear for you ?

chehefen, 2020-07-16:
sorry for the inconvenience. Our docker images need to be updated. For now you can work around by building your own image using the steps I mentioned before to build the library from scratch. We will try to update the docker images as soon as possible and will update in this thread once thats done.

disa6302, 2020-07-17:
@FlorianRuen , 

The docker images are updated. Feel free to use it if you would like. Sorry for the inconvenience again.

closed 2020-07-17 by disa6302

FlorianRuen, 2020-07-17:
@disa6302 Thanks, I've try and it seems to be ok after the update. Can you give more details on what happend . So if it's happen again I can know the reason and it will be faster to resolve something like this ?

Thanks

disa6302, 2020-07-17:
@FlorianRuen ,

Our cpp repo was using submodules up until a few days back. Now we do not use submodules and docker was not updated to reflect the change.

FlorianRuen, 2020-07-19:
@disa6302 Did you update the docker image for Rspberry pi also ? Because when I pull the docker image, I see the latest created/update date in 12 months ago ?

I try using this image and it seems there is backwards and loops ...

disa6302, 2020-07-19:
@FlorianRuen , 

Are you referring to the docker in https://github.com/aws-samples/amazon-kinesis-video-streams-demos/blob/master/producer-cpp/docker-raspberry-pi/Dockerfile? 
It says updated 2 days ago. I have updated all the dockerfiles.

FlorianRuen, 2020-07-19:
I refer to this docker image on your ECR repo

`546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi:latest`

This is the url from documentation

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker-download

disa6302, 2020-07-19:
@FlorianRuen , 

Can you check now? To see if you have got the updated docker image in the ECR repo?

FlorianRuen, 2020-07-19:
@disa6302 It seems good, when I run the docker pull, there is some downloaded files

I'm trying to run the pipeline using this new image and I will be back on this case to update

FlorianRuen, 2020-07-19:
@disa6302 Very strange, the docker image succesfully updated but still backward on docker image using my app (my app just get the docker image, create a container, connect a second network and exec the pipeline into the container, but there is backward)

I try and there is no backward when I run the pipeline directly in the container.

Maybe this is cause by the python docker API or something I ignore

I will adapt my code to work directly with C producer