awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #508: [QUESTION] How is the time zone set for a stream's server/producer time?

jupyter-note, 2020-07-09:
I've written a Python consumer that ingests the latest fragment from the stream (using the GetMedia API with NOW selector) and does some processing. I'd now like to monitor how much time has elapsed since each fragment was produced. 

I can successfully read the producer time from the fragment's metadata and convert it to a Python datetime object. It's then straightforward to determine the time elapsed. However, I'd like to deploy this system in varied settings where the producer may be in a different time zone than the consumer, so I'd love to learn how the producer/server time zone is initially set and if there's a way to manually set them when producing video.

chehefen, 2020-07-10:
in producer timestamp mode, getMedia receives the same timestamp as the frame that was put into producer. So the consumer should have context about what time zone the stream was produced. By default, producer uses absolute timestamp. So you can easily convert each frame timestamp to UTC time also.

jupyter-note, 2020-07-11:
Great, thank you!

closed 2020-07-11 by jupyter-note