awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #509: [QUERY] Failed to receive ack for uploads beyond 60 seconds

Subburaj47, 2020-07-10:
**Describe the bug**
We have scenario where we don't get the acknowledgement for the uploads for 60 seconds. This occurs intermittently. Just before this, I could see the "storage overflow" reported from the stream callback. The storage size we have configured is 3MB.
I have seen logs where even after reporting the storage overflow handler, the stream gets recovered and the uploads go successful. 

Kindly suggest if I have to modify some settings or look for any particular logs. Also suggest in which scenario would the acknowledgments to the uploads would fail. We now do a stream recreation whenever we land up in the ack failure for 60 seconds.

**To Reproduce**
Try running with encoder config which could produce 1080p high definition frames. After quite some time, I see this happening. 

**Expected behavior**
The uploads should be acknowledged at swiftly.

**SDK version number**
3.0.0

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
Running on a thirdparty system. Here are the details.
/mnt/ramdisk/root # cat /proc/version
Linux version 3.10.104+ (charlie@isbu-virtual-machine) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-2) ) #1 PREEMPT Mon May 18 16:01:21 CST 2020
**Additional context**
Add any other context about the problem here.

MushMal, 2020-07-10:
@Subburaj47 There are few issues that you are touching and I would like to take an in-depth look at this.

The SDK has a storage pressure callback (StorageOverflowPressureFunc) which gets called on less than 5% of storage availability. The default handler is only printing debug log but doesn't/can't handle it. If the pressure increases the storage will run out of space. Previously, the PutFrame API call would fail with storage out of memory error but in the latest commits (including rel 3.0.0) the storage OOM will result in the particular streams tail frames being evicted from the content view, potentially also resulting in dropped frame callback if those frames haven't been sent to the backend yet. There is a certain handler logic built in for the dropped frame. The end behavior is that the stream will recover and continue but frames could be dropped from the tail (lost).

Regarding the ACKs. One of the major reasons why KVS streaming has application level ACKs and doesn't rely solely on the TCP/networking ACKs is due to staleness. What this means is that network hops (switches, gateways) and sometimes the load balancer on AWS side can happily send network ACKs but drop the bits without delivery. It could also happen on the actual ingestion host networking stack due to an error or kernel issues. The application ACKs are designed to handle these situations where the Buffering ACK indicates that the backend host process has started receiving the bits and starts parsing the fragment, the Received ACK indicates that the parsing of a fragment has completed and the Persisted ACK indicates that the fragment bits have been indexed and durably persisted so the client can purge the workset (on which, the SDK gets rid of the frames constituting the fragment from the tail all together). 

In your case, it's not yet clear why the network stops receiving ACKs - could be networking (some network topologies struggle with duplex mode of transmission for HTTP1.1 we utilize). For this, you could listen to stream stale callback (StreamConnectionStaleFunc) which is configurable in StreamInfo on how long to wait for an ACK before invoking this. The default handler will reset the connection. https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/e14548d518666d551d637cf03300698962b7a227/src/source/ConnectionStaleStateMachine.c#L48

chehefen, 2020-07-13:
@Subburaj47, can you show us the log where you dont get ack for 60s? Thanks

Subburaj47, 2020-07-14:
> @Subburaj47, can you show us the log where you dont get ack for 60s? Thanks

@MushMal : Thanks for the detailed answer.
@chehefen : As such we have only the app logs complaining about the putFrame failure (based on the return value). We will enable the kvs debug logs and repro the problem. Will share you the logs once we encounter it. 
Thanks.

MushMal, 2020-07-14:
Our current theory is that we are actually getting ACKs but the core logic of mapping the ACK timestamp to the timeline in the content view is erroring with 

0x52000047 | STATUS_ACK_TIMESTAMP_NOT_IN_VIEW_WINDOW

which means that the frames have been pushed out of the view by the PutFrames and ContentStore pressure. The result is that the client callback is not being called which then triggers your app logic to detect no ACKs.

MushMal, 2020-07-14:
I apologize, actually, I misspoke, the fragmentAckReceivedFn callback will still be called even if the timestamp is out of the temporal window: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/src/StreamEvent.c#L928

MushMal, 2020-08-03:
Resolving this as stale.

closed 2020-08-03 by MushMal