awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #510: [QUESTION]Should the streamClosedHandler be invoked for both stopSync() and stop()

Subburaj47, 2020-07-10:
During the uninit, we use the stopSync and noticed that **streamClosedHandler invoked**. I believe the stopSync( ) is a blocking call. Is this a correct behavior to get an callback for this? If so, does the ack convey the stopSync completed or just the API return is fairly enough?
And the same callback is seen when the stop( ) is used.
[image]

** Any design considerations/constraints **
We are using 3.0.0 SDK.
[image]

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

MushMal, 2020-07-10:
stopSync() is a blocking API indeed. However, that has no bearing on whether the callbacks will be called or not. The core of the SDK is live-less (aka - no threads of it's own). It uses any of the liveness (threads) that call public APIs for control or events to drive the state machinery that can result in callbacks being issued.

In this case, once the stream is properly closed, the stream closed callback will be called. The async stop() will also trigger the callback, after all, stop sync is calling stop and they awaits for this callback itself to unblock. If the stream has no data in the buffer and the last ACK has already been received, the stop() async API will promptly close the stream as there is no outstanding work to be done.

disa6302, 2020-07-17:
@Subburaj47 , 

Closing this assuming your question is answered. Feel free to open an issue if you have face any issues. Thanks!

closed 2020-07-17 by disa6302