awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #511: The security token included in the request is invalid

uyennguyen16900, 2020-07-11:
I ran this command to stream my macbook’s webcam to Kinesis Video Streams
```
gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I420, width=1280,height=720 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=TinydoorStream storage-size=512 access-key=AWS_
ACCESS_KEY_ID secret-key=AWS_SECRET_ACCESS_KEY aws-region="us-west-2"

```
 and I got the following error

```
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2020-07-10 21:04:31 [0x1052d95c0] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and f
lags 0x00000001
2020-07-10 21:04:31 [0x1052d95c0] INFO - heapInitialize(): Creating AIV heap.
2020-07-10 21:04:31 [0x1052d95c0] INFO - heapInitialize(): Heap is initialized OK

2020-07-10 21:04:31 [0x1052d95c0] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000
000000000002
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - getSecurityTokenHandler invoked
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1594440271811860 Expiration: 0
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000
000000000010
2020-07-10 21:04:31 [0x1052d95c0] INFO - createDeviceResultEvent(): Create device result event.
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000
000000000040
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - clientReadyHandler invoked
2020-07-10 21:04:31 [0x1052d95c0] INFO - try creating stream
2020-07-10 21:04:31 [0x1052d95c0] INFO - Creating Kinesis Video Stream TinydoorStream
2020-07-10 21:04:31 [0x1052d95c0] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Stream name: TinydoorStream
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Content type: video/h264
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Max latency (100ns): 600000000
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Fragment duration (100ns): 20000000
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Key frame fragmentation: Yes
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Use frame timecode: Yes
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Absolute frame timecode: Yes
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Nal adaptation flags: 0
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Average bandwith (bps): 4194304
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Framerate: 25
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Buffer duration (100ns): 1200000000
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Replay duration (100ns): 400000000
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Connection Staleness duration (100ns): 600000000
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Store Pressure Policy: 1
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      View Overflow Policy: 1
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Segment UUID: NULL
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Frame ordering mode: 0
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo(): Track list
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Track id: 1
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Track name: kinesis_video
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - logStreamInfo():      Track cpd: NULL
2020-07-10 21:04:31 [0x1052d95c0] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000
000000000002
2020-07-10 21:04:31 [0x1052d95c0] INFO - writeHeaderCallback(): RequestId: 5ce52889-b38a-4d0e-8732-8afd031a85b9
2020-07-10 21:04:32 [0x7000012f6000] WARN - curlCompleteSync(): HTTP Error 403 : Response: {"message":"The security token included in the request is invalid."}
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AWS_ACCESS_KEY_ID/20200711/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=d75d7a0e3fb796db56f3cc1ee775f30e9f61cfc81430265b6c8a0bd741b3c026
    content-length: 35
    content-type: application/json
    host: kinesisv
2020-07-10 21:04:32 [0x7000012f6000] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"message":"The security tok
en included in the request is invalid."}
```

I did define AWS_ACCESS_KEY_ID and AWS_SECRET_ACCESS_KEY environment variables with the AWS access key id and secret key.

chehefen, 2020-07-13:
if you already defined the in the environment variable, plugin can pick it up directly. you dont need to defined them again in access-key and secret-key

disa6302, 2020-07-17:
@uyennguyen16900 , 

Are you still facing issues?

lherman-cs, 2020-07-27:
Since the issue has gone stalled, I'll mark this as resolved. Please feel free to reopen it for further assistance!

closed 2020-07-27 by lherman-cs