awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #512: [QUESTION]Query on the max latency, buffer duration and the content store values.

Subburaj47, 2020-07-14:
There are few scenarios where we encountered the streamLatencyPressure callback happening continuously.

** Any design considerations/constraints **
We configure the stream with the following stream definition values for the max latency, buffer duration and the content store.
max latency - **60 seconds**
buffer duration - **120 seconds**

From the documentation, we understood that we hit the streamLatencyPressure callback whenever the buffer duration value is higher than the max latency. Based on this, we planned to alter the max latency to **120 seconds** and the buffer duration to **60 seconds**. The reason we ended up with 60 is that, in higher resolutions our content store would hold a max of 1 minute frame data. On that basis we reduced the buffer duration as 60.

Kindly comment if this stream definition would be a suitable choice or if the understanding has a gap. Also, for lower resolutions (where we have the luxury to store frame data more than a minute), by altering the buffer duration to **60 seconds**, would it limit the frames stored to 60 irrespective of the content store configured?

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

MushMal, 2020-07-14:
Some background:

bufferDuration (https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L926) this value controls the temporal view duration of the ContentView. This simply means how long the buffer should be kept in memory max, given, the ContentStore can accommodate the actual frames sizes. Note: the ContentStore is shared amongs the streams and is the physical store for the frames while buffering while the ContentView is per-stream and has the logical view of the timeline.
maxLatency (https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L894) should be a value smaller than the bufferDuration which controls, as you've mentioned, when the stream latency pressure callback is called. Selecting a larger value or the sentinel value of 0 will mean that the no callback will be issued at all. This will then circumvent the continuous recovery logic in the C Producer layer that will attempt to toggle the connection when this callback is called in hopes of getting a better connection to stream out the accumulated buffer.

There is another parameter which controls the streaming behavior and that's the content store size in DeviceInfo. This is the physical size of the storage to allocate. The latest SDK commits have the default logic of not failing the PutFrame call but rather pushing the tail frames out, thus, potentially causing dropped frames if the frames have not been submitted.

Now, I would like to give some more info on the handling of the workset. 
When the frames are produced into the SDK, those are being added to the ContentStore and the information is added to the ContentView. The networking thread will then get continuous bits from the already packaged frames based on the temporal view from ContentView. The actual bits are stored in the ContentStore. This allows flexibility in buffering and handling variety of pressure scenarios. So, for example, if the frames are produced at a higher rate than the networking thread can send them over, the buffer will start getting saturated and once the maxLatency duration is reached, a stream latency callback will be issued. If this trend continues, the frames will get dropped from the tail once the buffer duration is reached.
On the other hand, if the stream is configured with non-zero persistence, the backend will be sending Persisted ACK as soon as the fragment (collection of frames constituting a fragment) is indexed and durably persisted on the backend. Once the SDK receives the Persisted ACK, it will map the ACK timestamp to the timeline and trim the entire fragment from the buffer as those have been durably persisted and will not need to be re-transmitted even on an error/rollback. 

Apologies for a long explanation but this gives some background for other readers to follow.

lherman-cs, 2020-07-27:
Since the question has been answered, I'll mark this as resolved. Please feel free to reopen it for further assistance!

closed 2020-07-27 by lherman-cs