awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #513: given error when run from outside of the directory

rraghav476, 2020-07-17:
if i am running it from outside just like home or from localhost i got error like
this 

log4cplus:ERROR could not open file kvs_log_configuration
log4cplus:ERROR No appenders could be found for logger (com.amazonaws.kinesis.video.gstreamer).
log4cplus:ERROR Please initialize the log4cplus system properly

but when i try inside the directory where sample_app located its run properly
can u help me

disa6302, 2020-07-17:
@rraghav476 , 

The log_configuration file is located inside the samples directory. If you run the samples, it should pick up that path from the build directory itself. 
If you are running gstreamer pipeline, one thing you can do is adding this: `log-config=<path-to-kvs_log_configuration-file>` to the pipeline and it should work.

disa6302, 2020-07-19:
@rraghav476 , 
I am closing this ticket. Feel free to reach out if you face any issues.

closed 2020-07-19 by disa6302

Alex-S-H-P, 2021-06-17:
@disa6302 hi.

I have a similar issue and therefore set out to apply your solution. It does not work and wherever I place the `log-config` i get back syntax errors for the pipe (`Error : the pipe could not be build : syntax error`), or, if i place it near the end i also get `(gst-launch-1.0:15025): GStreamer-CRITICAL **: 09:49:30.606: gst_element_link_pads_filtered: assertion 'GST_IS_BIN (parent)' failed
` errors.

Can you please precise what you meant by your answer ?
Thanks