awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #517: [BUG]Failed to playback with codecdata missing exception

Subburaj47, 2020-07-22:
**Describe the bug**
When the encoder is abruptly stopped while the app pushes the frames to the SDK, we see a problem during the playback. Although, during this time the clips are pushed to the cloud. 
During playback, we tend to get a missing codec private data exception. Would like to understand why this is happening for the frames pushed. Here's the log we got from the cloud.

{"line":{"time":"2020-07-16T17:25:17.378+00:00","message":"Unable to get streamUrl from kvs for mac=XXXXXXXX streamName=XXXXXXX message=Missing codec private data in fragment for track 1. (Service: AmazonKinesisVideoArchivedMedia; Status Code: 400; Error Code: MissingCodecPrivateDataException; Request ID: 5e725a90-0eb9-4d45-849e-c1f16c3ef2e4; Proxy: null) awsReqId=5e725a90-0eb9-4d45-849e-c1f16c3ef2e4","level":"WARN"},"source":"stdout","tag":"30fc1ef10700"}

**To Reproduce**
Steps to reproduce the behavior:

Kill/stop the encoder when frames are pushed by the app to the producer SDK.

**Expected behavior**
There should be seamless playback whenever the client requests the streams.

**SDK version number**
3.0.0

**Open source building**

**Screenshots**

**Platform (please complete the following information):**
Running on a thirdparty system. Here are the details.
/mnt/ramdisk/root # cat /proc/version
Linux version 3.10.104+ (charlie@isbu-virtual-machine) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-2) ) #1 PREEMPT Mon May 18 16:01:21 CST 2020

**Additional context**
Add any other context about the problem here.

MushMal, 2020-07-22:
@Subburaj47 can you please clarify the scenario? My understanding is that the producer SDK is not being fed frames? What's the producer SDK behavior? How do you (or do you) supply the CPD? 

Producer SDK by default, can extract the CPD from the stream if and only if the NAL adaption flags are specified to adapt the CPD and frames from Annex-B -> AvCC and when your underlying elementary stream has the CPD prepended to the I-frame such as 

| Annex-B start code | SPS | Annex-B start code | PPS | Annex-B start code | I-frame |

In this case, the SDK will extract the missing CPD (if not specified) automatically and attach it.

In case the producer is re-started, and for some reason you don't have the above mentioned frame format then there won't be any CPD included in the generated MKV header.

sharadrj, 2020-07-31:
@Subburaj47 , can you dump the outgoing MKV video in a file following the steps here? It will allow us to dig deeper.
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#debug

Subburaj47, 2020-08-20:
Missed to follow up here @sharadrj. As discussed offline, the issue was due to frame drops (which had key frames also dropped) and subsequent p-frames weren't having the CPD data. The commit shared worked out.
https://github.com/awslabs/amazon-kinesis-video-streams-pic/commit/1988f99fe77a3b37e9ed33643c20190a5c2e1689
This issue can be closed.

closed 2020-08-20 by MushMal