awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #518: [QUESTION]How many frames are dropped when the buffer pressure happens

Subburaj47, 2020-07-22:
When there is a buffer pressure happening (probably because of the frames getting piled up due to network retries), how many frames would be dropped at any instance. 

** Any design considerations/constraints **
We are taking decisions at the app level based on the droppedFrameHandler callback. Would like to understand more if the entire fragment is dropped or if the fragments are dropped exactly to regulate the buffer pressure. If it is the latter, how is the fragmentation handled. Here's one scenario I could think of.

Let's assume the fragmentation duration is 15 seconds (150 video frames). I have three cases to understand here.
1) If the SDK is in the middle of a fragment upload and the droppedFrame callback is issued. If the frames are dropped, would it be dropped until the next key frame?
2) SDK has just exactly completed the pushing the previous fragment and is about to start with the next fragment's key frame. Would the entire fragment is dropped until the next key frame? If not, how would the streaming proceed without a key frame.
3) If the SDK is about to complete pushing the fragment and has few (probably 50 frames) left out to go for the next key frame. Assume to avoid the buffer pressure, we might need 100 frames to be pulled out of the store. Now how much frames would be dropped?

Apologies for being wordy here. Thanks for your responses in advance.

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

MushMal, 2020-07-22:
Great questions and I will be adding a short document soon that goes over these. Sorry, it's not been added yet. Here are the answers:

* The SDK can be in the middle of a frame itself when the pressure events happen. 
* Once the pressure happens, the default behavior (which can be controlled) is as follows: The partially sent frame is still kept in memory in the attempt to sent the final bits in order to not cause syntactically invalid MKV. The next frames are pushed out of the tail (like P-frames) until the next Key-frame. This will still keep the syntactic validity of the MKV but the trailing frames of the fragment will be missing due to the pressure.
* Yes, the default behavior is to try to keep the MKV syntactically correct so if there is a need to drop a key-frame then the entire fragment will be dropped in the default policy.
* Not sure I understood the last question... If the first 100 frames are already sent but are still in the buffer (as the SDK has not yet received the persisted ACK) then those will be dropped from the tail without triggering the dropped frame report (as they have been sent). I need to double-check the behavior of the default policy but I believe the entire fragment will be dropped in this case.

Subburaj47, 2020-07-23:
> Great questions and I will be adding a short document soon that goes over these. Sorry, it's not been added yet. Here are the answers:
>
> * The SDK can be in the middle of a frame itself when the [...]

Subburaj47, 2020-07-23:
Thanks for the response @MushMal . 
I have another question spun around your reply :) Let's assume that a clip would have 270 frames. SDK has uploaded 100 frames (which includes a key frame at the very starting) and there is a pressure happening and remaining 170 frames are dropped. I have two questions here.

1. Would SDK ack for the 100 frames pushed?
2. We are looking for a common point between the SDK and the app. So that, the persisted ack given could be mapped to the individual clip boundary which inturn would be used for the telemetry analysis like how many clips are acknowledged as persisted and how many failed. **Is there any unique data the SDK can return the app which would be fair enough to mark as clip x has been uploaded successfully**? I know the timecode is unique. But that is given only during the ack so this cannot serve the purpose.

Btw closed by mistake and reopened back.

MushMal, 2020-07-23:
To answer your question I would like to give a little background info. Due to the fact that we have a low-latency streaming, the backend can only (persisted) ACK when it "closes" the current fragment and indexes and durably persists. Now, the question is how the backend can determine the fragment boundary. The only "marker" that the backend has that will deterministically indicate the current fragment boundary is the next fragment start. The only exception here is the specialized EoFR frame and the end of streaming session but both are not applicable to your question. Now, back to your first question:

1) The SDK will drop the remaining 170 frames assuming it has already pushed the leading 100 frames. Let's also assume the SDK will start the new fragment at frame 271 which is a key frame.
2) The backend will only see the first 100 frames constituting the fragment and then the 271st frame which is the key frame will "close" the current 100 frame long fragment and open a new one. From the perspective of the backend, the fragment is only 100 frames long and the next fragment starts at 101st frame which from the perspective of the SDK (due to the dropped frames) is the 271st.

This is the default behavior but for video this behavior is the most common to ensure syntactic and somewhat semantic consistency (due to not starting a p-frame fragment). From the perspective of the SDK buffer pressure, it makes sense as it attempts to free up as much buffer in case of the dropped frame on pressure as possible with minimal semantic impact on the streaming overall.

The next question is open for interpretation. In your case, your application knows the fragment boundaries and timestamps so it can actually map the persisted ACK timestamp to the fragments. The only issue is the dropped frames - in this case, the dropped frame callback has a timestamp with it that can be mapped to the precise frame(s) that are being dropped. 

I am not sure I understand why the persisted ACK timestamp is inadequate? This, in fact, is the best indication that the entire fragment not only has been uploaded but indexed and durably persisted and the acknowledgement has been received and processed by the SDK.

Baka7, 2020-07-25:
Hello MushMal,
For your question , **I am not sure I understand why the persisted ACK timestamp is inadequate?**
I have understand your answer but i have one use case where it's not enough to handle.

Key point : Every 15secs we will set the data in CustomData obj and keep on changing that data which we need to map . (1 clip = 15 sec duration).You recommend us to map it in PERSISTED ACK , if network fails  we wont get ACKs exactly at 15secs to map until connectivity.

1) I am explaining my use case here,
Due to the network issue SDK didn't push clips to server thereby I didn't receive any ACK's for clips.
Once the network connection is available, SDK pushes the clips in content store and sending all types of ACK's for all clips at a time. I get ACK's like BUFFERING , RECEIVED , PERSISTED for all clips at a interval of 1 sec.
We can't map our data here based on the timecode because we don't know what's the timecode value earlier to map it with our data for every 15sec.  As you know , we are able to get our lost clips acknowledgements with the expected timecode but with same data  (for all the clips )  which we set more recently during the last 15sec cycle.

Is there another way to get the timecode that content store is creating and assigning to the clips other than ACK's  handler?
In our scenario, we need it to be send from SDK for every clip from any callbacks  . If that's possible, we will able to handle our use case.

Thanks in advance.

MushMal, 2020-07-27:
Sorry but I am still unclear. You have configured the SDK with frame timestamps so the SDK will use the timestamps that you specify. That means you can store the clip first frame timestamp which is what will be returned to you in the callbacks like dropped frame and in the ACKs. From the perspective of the SDK the frames are only distinguished by their timestamps and its what the content view is using to apply temporal logic to it

Baka7, 2020-07-28:
Is there any case we can get Fragment Timecode as 0 for any clip ? 
I used to get Fragment Timecode as 0 but i didn't set the 0 as timestamp while setting the key frame .
That's why I didn't do like that. Occasionally Fragment TImecode comes 0 .

MushMal, 2020-07-28:
I believe you are using absolute time

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/StreamDefinition.cpp#L20

in which case the MKV clusters will contain the timestamp of the key frame, though, time-scaled at 1ms per default time unit in MKV.

What's the situation that you are getting 0 timestamp? Is this when you receive fragment ACK callback? In case of an error, it's possible that the backend will set this to 0 if the parsing fails.

MushMal, 2020-08-03:
Resolving the issue as the main questions have been answered.

closed 2020-08-03 by MushMal