awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #521: when i am try to create my application or compile sample app it given fatal error

rraghav476, 2020-07-24:
A one liner description about the use case and what you are trying to achieve

** Any design considerations/constraints **
Explain in detail how you would like to integrate our SDK into your solution 

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

rraghav476, 2020-07-24:
kinesis_video_gstreamer_sample_app.cpp:1:21: fatal error: gst/gst.h: No such file or directory

lherman-cs, 2020-07-24:
@rraghav476 have you installed the dependency requirements?

Mac:
```
brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav
```

Linux
```
sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
```

If you have specified, `-DBUILD_DEPENDENCIES=ON`, you still need to install gstreamer and its plugins by yourself.

rraghav476, 2020-07-25:
@lherman-cs thank for reply
yes i am installed the dependency requirements
by using Linux command as you mention above
OK let me explain how i an going on
step 1 clone the repository
    git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
step 2 install dependency
    sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev   gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
step 3 make build directory
step 4 run cmake and make
and after that I am write my code and run it give fatal error about

lherman-cs, 2020-07-27:
@rraghav476 Could you check gstreamer's include paths, `pkg-config --cflags-only-I gstreamer-1.0`?

And, could you verify if those paths are searchable from gcc/clang, `gcc -xc -E -v -`? This will give a list of search paths from gcc/clang.

rraghav476, 2020-07-28:
@lherman-cs on running 
`pkg-config --cflags-only-I gstreamer-1.0`
it return gstreamer paths like
`-I/usr/include/gstreamer-1.0 -I/usr/lib/x86_64-linux-gnu/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linuxgnu/glib-2.0/include`
but these paths are not searchable by gcc/clang
by running 'gcc -xc -E -v -` it return me
[image]

MushMal, 2020-07-28:
Something seems to be wrong with the GStreamer install path. Are you able to run GStreamer commands like gst-launch-1.0 from your command prompt?

rraghav476, 2020-07-28:
i can't able to run GStreamer using `gst-launch-1.0`
it's return error 
ERROR: pipeline could not be constructed: empty pipeline not allowed

MushMal, 2020-07-28:
That’s ok. So, it does seem that young have the GStreamer core installed and the path is set. Can you find the gst/gst.h file? Perhaps, you have two separate installations?

rraghav476, 2020-07-28:
@MushMal 
i can't find gst/gst.h
sorry what do you mean by two separate installations

MushMal, 2020-07-28:
For some reason your installation of GStreamer does not include the header files. I am not sure why but if you have installed the devel package and it still doesn’t show up then you might want to re-install it. Also, GStreamer forums may be a good source of information as well

rraghav476, 2020-07-28:
thank you for information

MushMal, 2020-07-28:
I am going to resolve this issue for now. Please don't hesitate to reopen or cut a new one

closed 2020-07-28 by MushMal