awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #522: How to track stream status in C++ Producer?

thasin96, 2020-07-27:
How to track health/status of RTSP Streams in cpp Producer? 

I am able to upload stream to Kinesis Video Stream using RTSP Url. My question is, How can I track status of RTSP Streams in SDK? For Example, My stream has been disconnected due to Network issue or incorrect URL configuration. How can I track the lost connection?

I have tried the following scenario. I have started Active RTSP stream and provided the RTSP URL to RTSP file(Multistream app sample). My stream is being processed to KVS successfully. Next, I forcefully stopped RTSP Stream. Now, SDK stopped processing my stream due to failed active stream. When I again active The RTSP Stream, C++ Application is not processing my old Stream. It's in same state where it assumes link is broken. I expected that CPP code automatically processes my stream when it is back to available. 

My questions are,
1. How to check whether stream is active or not (Health Check)?
2. How to automatically reconnect to stream when it's available.?

MushMal, 2020-07-27:
There really isn’t any API call to get the current state the  core of the SDK is in. An application could provide an overwrite of the callbacks and get notified on the state transitions. Perhaps it’s a good idea to expose an API that could return a state but I am not yet sure of the value that it provides. 

Regarding your case. I assume you are using one of the samples based on the GStreamer pipeline. You might want to provide the debug logs to understand what is happening and whether it’s the SDK itself or the GStreamer rtsp element.

thasin96, 2020-07-27:
@MushMal - this is the log

[INFO ] [27-07-2020 06:49:55:746.620 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [27-07-2020 06:49:55:746.720 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [27-07-2020 06:49:55:746.741 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [27-07-2020 06:49:55:746.772 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [27-07-2020 06:49:55:746.797 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [27-07-2020 06:49:55:746.837 GMT] getSecurityTokenHandler invoked
[DEBUG] [27-07-2020 06:49:55:746.883 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1595832595746860260 Expiration: 0
[INFO ] [27-07-2020 06:49:55:746.909 GMT] New credentials expiration is 1595836195
[DEBUG] [27-07-2020 06:49:55:746.949 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [27-07-2020 06:49:55:746.972 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [27-07-2020 06:49:55:746.996 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [27-07-2020 06:49:55:747.017 GMT] clientReadyHandler invoked
[DEBUG] [27-07-2020 06:49:55:747.058 GMT] Client is ready
[INFO ] [27-07-2020 06:49:55:747.171 GMT] Creating Kinesis Video Stream blabla
[INFO ] [27-07-2020 06:49:55:747.214 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [27-07-2020 06:49:55:747.230 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [27-07-2020 06:49:55:747.244 GMT] logStreamInfo(): 	Stream name: blabla 
[DEBUG] [27-07-2020 06:49:55:747.259 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [27-07-2020 06:49:55:747.274 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [27-07-2020 06:49:55:747.289 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [27-07-2020 06:49:55:747.304 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [27-07-2020 06:49:55:747.319 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [27-07-2020 06:49:55:747.333 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [27-07-2020 06:49:55:747.348 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [27-07-2020 06:49:55:747.363 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [27-07-2020 06:49:55:747.377 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [27-07-2020 06:49:55:747.392 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [27-07-2020 06:49:55:747.407 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [27-07-2020 06:49:55:747.421 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [27-07-2020 06:49:55:747.436 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [27-07-2020 06:49:55:747.451 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [27-07-2020 06:49:55:747.466 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [27-07-2020 06:49:55:747.481 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [27-07-2020 06:49:55:747.496 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [27-07-2020 06:49:55:747.511 GMT] logStreamInfo(): Track list
[DEBUG] [27-07-2020 06:49:55:747.536 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [27-07-2020 06:49:55:747.552 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [27-07-2020 06:49:55:747.566 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [27-07-2020 06:49:55:747.580 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [27-07-2020 06:49:55:747.595 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [27-07-2020 06:49:55:747.648 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [27-07-2020 06:49:56:330.645 GMT] writeHeaderCallback(): RequestId: 9394312b-387c-4ac0-aa8e-db37b8561b54
[DEBUG] [27-07-2020 06:49:56:331.031 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.595829605025E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-south-1:709356462602:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-south-1:709356462602:stream/blabla/1595829605025","StreamName":"blabla","Version":"xJAysCMKAcZixnphOxzf"}}
[INFO ] [27-07-2020 06:49:56:335.618 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [27-07-2020 06:49:56:335.807 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [27-07-2020 06:49:56:699.589 GMT] writeHeaderCallback(): RequestId: b10ee315-30c6-4081-b64f-414e1d5829b7
[DEBUG] [27-07-2020 06:49:56:699.825 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com"}
[INFO ] [27-07-2020 06:49:56:704.476 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [27-07-2020 06:49:56:704.632 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [27-07-2020 06:49:56:704.801 GMT] getStreamingTokenHandler invoked
[... cut ...]
[DEBUG] [27-07-2020 06:49:56:705.303 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [27-07-2020 06:49:56:705.377 GMT] streamReadyHandler invoked
[DEBUG] [27-07-2020 06:49:56:705.853 GMT] Stream is ready: blabla
[DEBUG] [27-07-2020 06:49:56:721.352 GMT] appsink configured
[DEBUG] [27-07-2020 06:49:56:723.290 GMT] Pipeline playing

In system output, it stays Pipeline playing line. It's not sending any data because link is not active. I want to capture this notify.

MushMal, 2020-07-27:
This line comes from GStreamer. I am sure there is a way to get notified from GStreamer but it’s a good question for GStreamer forums. The logs don’t have anything that describes your original scenario

thasin96, 2020-07-27:
@MushMal - Thanks for the advice. I will ask more details in GStreamer forum. I have a final question. the above log I showed is for running RTSP broken link first time.This is the log for running first with active URL and forcefully I made the link down. 

[INFO ] [27-07-2020 07:12:36:936.266 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [27-07-2020 07:12:36:936.365 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [27-07-2020 07:12:36:936.382 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [27-07-2020 07:12:36:936.411 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [27-07-2020 07:12:36:936.435 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [27-07-2020 07:12:36:936.471 GMT] getSecurityTokenHandler invoked
[DEBUG] [27-07-2020 07:12:36:936.514 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1595833956936492106 Expiration: 0
[INFO ] [27-07-2020 07:12:36:936.540 GMT] New credentials expiration is 1595837556
[DEBUG] [27-07-2020 07:12:36:936.572 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [27-07-2020 07:12:36:936.592 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [27-07-2020 07:12:36:936.611 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [27-07-2020 07:12:36:936.628 GMT] clientReadyHandler invoked
[DEBUG] [27-07-2020 07:12:36:936.667 GMT] Client is ready
rtsp://127.0.0.1:7001/e3e9a385-7fe0-3ba5-5482-a86cde7faf48
admin
ThoughtClan@123
blabla
[INFO ] [27-07-2020 07:12:36:936.763 GMT] Creating Kinesis Video Stream blabla
[INFO ] [27-07-2020 07:12:36:936.811 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [27-07-2020 07:12:36:936.826 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [27-07-2020 07:12:36:936.839 GMT] logStreamInfo(): 	Stream name: blabla 
[DEBUG] [27-07-2020 07:12:36:936.851 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [27-07-2020 07:12:36:936.864 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [27-07-2020 07:12:36:936.877 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [27-07-2020 07:12:36:936.891 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [27-07-2020 07:12:36:936.904 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [27-07-2020 07:12:36:936.916 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [27-07-2020 07:12:36:936.928 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [27-07-2020 07:12:36:936.941 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [27-07-2020 07:12:36:936.954 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [27-07-2020 07:12:36:936.967 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [27-07-2020 07:12:36:936.979 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [27-07-2020 07:12:36:936.992 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [27-07-2020 07:12:36:937.005 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [27-07-2020 07:12:36:937.018 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [27-07-2020 07:12:36:937.031 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [27-07-2020 07:12:36:937.043 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [27-07-2020 07:12:36:937.056 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [27-07-2020 07:12:36:937.068 GMT] logStreamInfo(): Track list
[DEBUG] [27-07-2020 07:12:36:937.080 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [27-07-2020 07:12:36:937.092 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [27-07-2020 07:12:36:937.105 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [27-07-2020 07:12:36:937.118 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [27-07-2020 07:12:36:937.130 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [27-07-2020 07:12:36:937.179 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [27-07-2020 07:12:37:453.522 GMT] writeHeaderCallback(): RequestId: 74ac8b7e-67bf-42d2-8c63-7fb4609ea766
[DEBUG] [27-07-2020 07:12:37:453.923 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.595829605025E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-south-1:709356462602:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-south-1:709356462602:stream/blabla/1595829605025","StreamName":"blabla","Version":"xJAysCMKAcZixnphOxzf"}}
[INFO ] [27-07-2020 07:12:37:458.901 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [27-07-2020 07:12:37:459.054 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [27-07-2020 07:12:38:140.283 GMT] writeHeaderCallback(): RequestId: 3d68e63b-5f24-4589-9a15-46cbbb0149b4
[DEBUG] [27-07-2020 07:12:38:140.508 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com"}
[INFO ] [27-07-2020 07:12:38:143.614 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[... cut ...]
[WARN ] [27-07-2020 07:13:25:326.648 GMT] putStreamCurlHandler(): Stream with streamHandle 15904696 has exited without triggering end-of-stream. Service call result: 599
[INFO ] [27-07-2020 07:13:25:327.078 GMT] kinesisVideoStreamTerminated(): Stream 0xf2afb8 terminated upload handle 0 with service call result 599.
[DEBUG] [27-07-2020 07:13:25:327.471 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200

My question is Is SDK capable of handling broken RTSP link? What will happen when the link is not active? Does it look for it's status after n secs? If so, the link is active Does it try to reconnect?

MushMal, 2020-07-27:
The logs indicate that indeed the networking is down which is why we get a timeout from the networking client (libCurl). The SDK has logic built in when it won’t retry if it’s buffer is empty (no data needs to be sent) and there are no frames that are put. It moves to Ready state and will issue PutMedia as soon as there is a new frame that’s produced into it. In this case, it’s likely that the rtsp element doesn’t produce frames after the network recovery so the KVS plugin doesn’t really do anything but wait until a frame is pushed into it.

sharadrj, 2020-08-08:
Closing this issue. @thasin96, please open new issues or reopen this one if you have further questions.

closed 2020-08-08 by sharadrj