awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #523: [FEATURE] install via apt (package manager)

js1972, 2020-07-28:
**Is your feature request related to a problem? Please describe.**
Its clunky having to manually set env vars and building from source. Why cant we install via apt with the kvssink gstreamer plugin installed into the correct location.

**Describe the solution you'd like**
I'd like to be able to do :
```
sudo apt install kvssink (or kvs producer etc)
gst-inspect-1.0 kvssink
```

MushMal, 2020-07-28:
This is a valid request. Ideally, kvssink should be part of the GStreamer plugin package but we haven’t got around yet it pushing it

js1972, 2020-07-28:
Eagerly looking forward to a proper gstreamer package. With the current build progress I'm unable to build in a docker container as well - get openssl conflicts with boto3.

ishangupta3, 2020-07-29:
@js1972 it is me from the nvidia forum; I agree and looking forward to a proper gstreamer package.

hassanctech, 2020-11-12:
We're discussing this internally and hope to work on this sometime in the near future, but cannot provide an ETA at the moment.

rkacenski, 2021-05-24:
Any progress on this? Trying to do this myself currently, with little success