awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #524: libopenssl build failure

amitrjaiswal1, 2020-07-28:
I am facing this build error on Windows machine.....

CMake Error at C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/tmp/project_libopenssl-gitclone.cmake:52 
(message): Failed to update submodules in: 'C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl'ninja: build stopped: subcommand failed.

** Any design considerations/constraints **
We are trying to build SDK through MSVC (Visual Studio 2019).

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

MushMal, 2020-07-28:
Not really sure what might cause OpenSSL build failures. We have had some issues with parallel build built that usually gets past the failure after the retry. Are you able to replicate the script as in Travis.yaml file when the Travis Ci gets build for Windows?

amitrjaiswal1, 2020-07-28:
Below is output log...

 [CMake] fatal: unable to access 'https://boringssl.googlesource.com/boringssl/': SSL certificate problem: unable to get local issuer certificate
[CMake] fatal: clone of 'https://boringssl.googlesource.com/boringssl' into submodule path 'C:/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/boringssl' failed
[CMake] Failed to clone 'boringssl'. Retry scheduled

lherman-cs, 2020-07-28:
@amitrjaiswal1 it seems that the issue is within your system, it failed to get public certificates. Could you try to use `curl https://boringssl.googlesource.com/boringssl/`? It should give you the same error.

amitrjaiswal1, 2020-07-28:
@lherman-cs Got below error for 'curl https://boringssl.googlesource.com/boringssl/'
curl: (35) schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - The revocation function was unable to check revocation for the certificate.

MushMal, 2020-07-28:
@amitrjaiswal1 the build/runtime failure is not really related to KVS and we won't be able to help much. Perhaps you could ask the question in OpenSSL forums and they might help.

Alternatively, you could clean up your system and ensure you have a non-broken OpenSSL libraries..

amitrjaiswal1, 2020-07-28:
@MushMal will do. Thanks for prompt update.

closed 2020-07-28 by MushMal

fertrevino, 2022-09-04:
Did you manage to find a solution? Facing the same problem...