awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #525: Audio and video sync issue- audio track not working

my56, 2020-07-29:
I'm trying to stream video and audio in multitrack,I'm gathering audio as aac using java sound api, video as h264 using webcam api and for multitrack I'm referring to AudioVideoFrameSource, AudioVideoFileMediaSourceConfiguration and AudioVideoFileMediaSource class.Not dealing with the thread sleep between putting audio/video frames and the file related operations.

Here I'm facing audio and video sync issue,audio plays like a fast forwarding mode.

MushMal, 2020-07-29:
Its hard to understand what's going on with your media pipeline but you will likely need to "stabilize" the audio track first - perhaps by synchronizing it against the wall clock.

MushMal, 2020-08-03:
Resolving as stale

closed 2020-08-03 by MushMal