awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #526: How KVS deals Network Failure?

thasin96, 2020-07-29:
I would like to understand in detail how KVS deals with Network Failure?

I am using kvs_gstreamer_sample application to stream from RTSP. I am able to successfully send frames to KVS. I would like to know, What would happen if Network Failure occurs. 
I have tried following two scenarios. 

1. Network Failure while processing RTSP local RTSP url
2. Network Failure while processing RTSP public RTSP url

**case 1 :**
the video source is present locally and a network disruption occurs. 

**Expected behaviour** - wait for the network to come back, reconnect and start streaming data again 

**current behaviour** - waits for the network to come back, reconnects and then pushes only the data which was not pushed when the network was down and once that is done, it terminates the process rather than continuing. 

**Log:**
```
[DEBUG] [29-07-2020 05:04:51:929.204 GMT] postReadCallback(): Wrote 5134 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [29-07-2020 05:04:51:935.835 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [29-07-2020 05:04:52:513.676 GMT] postReadCallback(): Wrote 5122 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [29-07-2020 05:04:52:513.950 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[WARN ] [29-07-2020 05:05:28:553.081 GMT] curlCompleteSync(): curl perform failed for url https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [29-07-2020 05:05:28:554.061 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIA2KKHQUIFLBAKBTN4/20200729/ap-south-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=f3d94683ff1637120d7dda4a04eee06e28aa5a43f39eb
[DEBUG] [29-07-2020 05:05:28:554.183 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: mobile-stream with upload handle: 0 exited. http status: 0
[WARN ] [29-07-2020 05:05:28:554.567 GMT] putStreamCurlHandler(): Stream with streamHandle 21469512 has exited without triggering end-of-stream. Service call result: 599
[INFO ] [29-07-2020 05:05:28:554.644 GMT] kinesisVideoStreamTerminated(): Stream 0x1479948 terminated upload handle 0 with service call result 599.
[DEBUG] [29-07-2020 05:05:28:554.720 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
```

**questions:**
1. can the behaviour be changed to get the expected behaviour?
2. how long does it wait for the network to comeback before terminating? 

**case 2:**
the video source is not present locally ( For example: a public RTSP stream) and a network error occurs, here both conection to the RTSP stream and AKVS is lost. 

**expected behaviour:** wait for network to come back, reconnect and start streaming data again. 

**current behaviour 1:** waits for 30 seconds, if no network still, it terminates the process

**current behaviour 2:** network comes back within 30 seconds ( it continues to stream)

**Log:**
```
[DEBUG] [29-07-2020 05:56:43:421.952 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [29-07-2020 05:56:43:422.023 GMT] Reporting fragment ack received. Ack timecode 1596002177863
[DEBUG] [29-07-2020 05:56:43:710.328 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:44:405.261 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:44:830.588 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:45:299.960 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:45:862.597 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:46:369.853 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:46:993.354 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:47:000.593 GMT] postWriteCallback(): Curl post body write function for stream with handle: stream-2 and upload handle: 759 returned: {"EventType":"RECEIVED","FragmentTimecode":1596002177863,"FragmentNumber":"91343852333181486862043791187064884619567357599"}

[DEBUG] [29-07-2020 05:56:47:001.296 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [29-07-2020 05:56:47:001.646 GMT] Reporting fragment ack received. Ack timecode 1596002177863
[DEBUG] [29-07-2020 05:56:47:005.087 GMT] postWriteCallback(): Curl post body write function for stream with handle: stream-2 and upload handle: 759 returned: {"EventType":"BUFFERING","FragmentTimecode":1596002183363,"FragmentNumber":"91343852333181486866995551344206406684112058170"}

[DEBUG] [29-07-2020 05:56:47:005.325 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [29-07-2020 05:56:47:005.405 GMT] Reporting fragment ack received. Ack timecode 1596002183363
[DEBUG] [29-07-2020 05:56:47:128.433 GMT] postWriteCallback(): Curl post body write function for stream with handle: stream-2 and upload handle: 759 returned: {"EventType":"PERSISTED","FragmentTimecode":1596002177863,"FragmentNumber":"91343852333181486862043791187064884619567357599"}

[DEBUG] [29-07-2020 05:56:47:129.238 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [29-07-2020 05:56:47:129.939 GMT] Reporting fragment ack received. Ack timecode 1596002177863
[DEBUG] [29-07-2020 05:56:47:850.915 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:48:339.054 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[... 8 lines cut ...]
[DEBUG] [29-07-2020 05:56:52:108.993 GMT] Reporting fragment ack received. Ack timecode 1596002183363
[DEBUG] [29-07-2020 05:56:52:127.985 GMT] postWriteCallback(): Curl post body write function for stream with handle: stream-2 and upload handle: 759 returned: {"EventType":"BUFFERING","FragmentTimecode":1596002188863,"FragmentNumber":"91343852333181486871947311501347929157124287154"}

[DEBUG] [29-07-2020 05:56:52:128.083 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [29-07-2020 05:56:52:128.118 GMT] Reporting fragment ack received. Ack timecode 1596002188863
[DEBUG] [29-07-2020 05:56:52:248.893 GMT] postWriteCallback(): Curl post body write function for stream with handle: stream-2 and upload handle: 759 returned: {"EventType":"PERSISTED","FragmentTimecode":1596002183363,"FragmentNumber":"91343852333181486866995551344206406684112058170"}

[DEBUG] [29-07-2020 05:56:52:249.063 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [29-07-2020 05:56:52:249.100 GMT] Reporting fragment ack received. Ack timecode 1596002183363
[DEBUG] [29-07-2020 05:56:52:823.937 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:53:382.956 GMT] postReadCallback(): Wrote 7713 bytes to Kinesis Video. Upload stream handle: 759
[DEBUG] [29-07-2020 05:56:53:458.053 GMT] postReadCallback(): Pausing CURL read for upload handle: 759
[WARN ] [29-07-2020 05:57:29:911.517 GMT] curlCompleteSync(): curl perform failed for url https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [29-07-2020 05:57:29:912.440 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-703775db.kinesisvideo.ap-south-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIA2KKHQUIFLBAKBTN4/20200729/ap-south-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=49b9bfa27afbfa5985511ba4d920cc405cb9c03e97888
[DEBUG] [29-07-2020 05:57:29:912.556 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: stream-2 with upload handle: 759 exited. http status: 0
[WARN ] [29-07-2020 05:57:29:913.089 GMT] putStreamCurlHandler(): Stream with streamHandle 15624520 has exited without triggering end-of-stream. Service call result: 599
[INFO ] [29-07-2020 05:57:29:913.167 GMT] kinesisVideoStreamTerminated(): Stream 0xee6948 terminated upload handle 759 with service call result 599.
[DEBUG] [29-07-2020 05:57:29:913.244 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
```

**questions:**
1. can the wait time be changed?

MushMal, 2020-07-29:
@thasin96 which asset are you using? Are you using kvssink within gst-launch or your app or you are using the producer interface directly in your app like the video only sample in C producer SDK?

To give a few answers:

* KVS SDK does not control the rtsp part - its up to your rtsp implementation so we can't comment on it.
* Default C Producer behavior is to keep re-streaming. The logs you've provided do not include the state machine recovery. 
* The core of the SDK can be configured to not re-stream or recover.

In order to isolate and understand the behavior, you could try to use kvs video only sample in C producer samples and run your experiments by modifying the code.

MushMal, 2020-08-03:
@thasin96 would you please kindly resolve the issue if the response answered your questions. If not, please let me know and I will add more clarifications

closed 2020-08-03 by thasin96