awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #527: Unable to Build Project Using KVS Producer SDK as Dependency

lummish, 2020-07-29:
I'm currently trying to integrate this SDK in a separate application but am having a lot of trouble building it, even with `-DBUILD_DEPENDENCIES` on.

My project is built with CMake. I've added this repository as a submodule (recursively) and added it as a subdirectory. All of the third-party dependencies (curl, log4cplus, openssl) build correctly, but it seems as though compilation of the library itself fails due to the inclusion of system libraries that do not exist. Below you'll find the relevant section of my `CMakeLists.txt`, the output from running CMake from my project's build directory, and finally the output of `make`:

## Relevant CMake Section
```cmake
...
add_subdirectory(dependencies/amazon-kinesis-video-streams-producer-sdk-cpp)
...
```

## CMake Output
```
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   exception
-- Found AWS SDK for C++, Version: 1.7.354, Install Root:/usr/local, Platform Prefix:, Platform Dependent Libraries: pthread;crypto;ssl;z;curl
-- Components specified for AWSSDK: s3, application wll be depending on libs: aws-cpp-sdk-s3;aws-cpp-sdk-core
-- Try finding aws-cpp-sdk-core
-- Found aws-cpp-sdk-core
-- Try finding aws-cpp-sdk-s3
-- Found aws-cpp-sdk-s3
CMake Warning (dev) at dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeLists.txt:8 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Kinesis Video Cpp Producer path is /path/to/my/repo
-- dependencies install path is /path/to/my/repo/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /path/to/my/repo/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No configure step for 'libkvscproducer-download'
[ 33%] No build step for 'libkvscproducer-download'
[ 44%] No install step for 'libkvscproducer-download'
[ 55%] No test step for 'libkvscproducer-download'
[ 66%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
CMake Warning (dev) at dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:4 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Kinesis Video Producer path is /path/to/my/repo/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- openssl already built
-- curl already built
-- Finished building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /path/to/my/repo/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic
[ 11%] Performing update step for 'libkvspic-download'
[ 22%] No configure step for 'libkvspic-download'
[ 33%] No build step for 'libkvspic-download'
[ 44%] No install step for 'libkvspic-download'
[ 55%] No test step for 'libkvspic-download'
[ 66%] Completed 'libkvspic-download'
[100%] Built target libkvspic-download
CMake Warning (dev) at dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeLists.txt:4 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenSSL: /path/to/my/repo/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so (found version "1.1.1g")
-- Bitness 64 bits
-- Found OpenSSL: /path/to/my/repo/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so (found suitable version "1.1.1g", minimum required is "1")
found log4cplus.
-- adding symlinks
-- Configuring done
-- Generating done
-- Build files have been written to: /path/to/my/repo/release
```

## make
```
[ 17%] Built target kvspicUtils
[ 25%] Built target kvsCommonCurl
[ 57%] Built target kvspic
[ 67%] Built target cproducer
[ 67%] Building CXX object dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/KinesisVideoProducer.dir/src/Auth.cpp.o
In file included from /home/automotus/picard/optix/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.cpp:1:0:
/home/automotus/picard/optix/dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.h:12:10: fatal error: Logger.h: No such file or directory
 #include <Logger.h>
          ^~~~~~~~~~
compilation terminated.
dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/KinesisVideoProducer.dir/build.make:62: recipe for target 'dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/KinesisVideoProducer.dir/src/Auth.cpp.o' failed
make[2]: *** [dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/KinesisVideoProducer.dir/src/Auth.cpp.o] Error 1
CMakeFiles/Makefile2:90: recipe for target 'dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/KinesisVideoProducer.dir/all' failed
make[1]: *** [dependencies/amazon-kinesis-video-streams-producer-sdk-cpp/CMakeFiles/KinesisVideoProducer.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

It seems to have something to do with the unusual inclusion style of header files in the SDK. Why are project headers included using angle brackets? I've tried to install the library but it seems as though the install step is not configured.

Am I doing something wrong? Is there a recommended method for using this SDK as a dependency? I had no trouble building the GStreamer plugin, but it doesn't satisfy my use-case. I have to use the producer library directly but building it as a dependency has proven quite difficult.

MushMal, 2020-07-29:
You can follow the same pattern of usage as in the samples. It’s hard to debug your cmake files

MushMal, 2020-07-30:
I am just not sure why you are facing any build issues at all. Which platform are you using? Could you confirm that you are pulling a clean git clone with recursion of the latest SDK commit and attempting to build with the prerequisites? We have the Travis CI builds which are clean.

lummish, 2020-07-30:
@MushMal just to clarify, I had no problems whatsoever building the library and GStreamer plugin in isolation. The build procedure works perfectly well (as long as `BUILD_DEPENDENCIES` is enabled) on the project in isolation. This wouldn't be a problem at all if the project's CMake included an install step. 

I'm building on armv8.

Problems only arise when I pull this repository in as a submodule in some other project and attempt to build it that way. As you probably saw in [my PR](#528), this is because certain CMake variables refer to the parent CMake context's source directory, and not this project's source directory during the build process. This causes files to be referenced using incorrect paths and causes the build to fail.

MushMal, 2020-07-30:
Oh, thanks for the clarification. That makes sense. Let's try to shepherd this change in.

closed 2020-07-30 by MushMal

andreikukenov, 2023-02-22:
@lummish Hello. I faced the fame issue. Have you managed to include kvs sdk as subdirectory?

lummish, 2023-02-22:
> @lummish Hello. I faced the fame issue. Have you managed to include kvs sdk as subdirectory? 

@andreikukenov unfortunately it's been a few years since this issue was raised and I can't remember the details.

I believe the PR referenced above resolved the issue for me.

andreikukenov, 2023-02-23:
@lummish Thank you!

PR #531 (2020-07-30): Replace CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR
Replace use of CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR to enable
building the SDK in an add_subdirectory() instruction.

Resolves #527

*Issue #, if available:*

*Description of changes:*

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit 5f1aede0 (2020-07-30): Replace CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR (#531)

Replace use of CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR to enable
building the SDK in an add_subdirectory() instruction.

Resolves #527
PR #528 (2020-07-30): Replace CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR
*Issue #, if available:*
Resolves #527

*Description of changes:*
Replace use of CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR to enable
building the SDK in an add_subdirectory() instruction.

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit 0ba30166 (2020-07-30): Replace CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR (#528)

Replace use of CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR to enable
building the SDK in an add_subdirectory() instruction.

Resolves #527