awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #529: [QUESTION] Command for streaming file source

ishangupta3, 2020-07-30:
I am trying to stream from a local file source to aws Kinesis video stream. I believe I am using the wrong command, can someone share the appropriate command to stream from a local file source (.mp4).

gst-launch-1.0 filesrc location=/deepstream-5.0/samples/streams/sample_1080p_h264.mp4 ! videoconvert ! video/x-raw,width=1920,height=1080 ! x264enc bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="MyKVStream" storage-size=512 access-key="AKIAZ7MYGKRQHUAELXEZ" secret-key="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" aws-region="us-west-2

disa6302, 2020-07-30:
Hello @ishangupta3, 

Could you try this pipeline? 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/macos.md#running-the-gst-launch-10-command-to-upload-mp4-file-that-contains-both-audio-and-video-in-mac-os . This is for macos, but you could try it for the system you are running in. 

One more thing, I have edited your question to remove the access key and secret key you added. Make sure you do not disclose those details in the future. Thanks!

ishangupta3, 2020-07-30:
Thank you for doing that. I will try that command, I am using Ubuntu 18.04.

ishangupta3, 2020-07-30:
@disa6302 I ran the command for the mp4, while it did say it was writing packets to AWS Kinesis Video Stream - sorry for the long error message. But it seems to send packets successfully then it just stops with an error:

> Current transfer rate (bps): 31876704 (31129 Kbps)
2020-07-29 19:34:25 [547630256624] INFO - stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 00000055853ba5a8.
2020-07-29 19:34:25 [547621863920] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:25 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:25 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:25 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:25 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:25 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:26 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:26 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:26 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:26 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:26 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:27 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:27 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:27 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:27 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:27 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:27 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:27 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:28 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:29 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:30 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:30 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:30 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:30 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:30 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-07-29 19:34:31 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[... cut ...]
2020-07-29 19:34:41 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 2
2020-07-29 19:34:41 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 2
2020-07-29 19:34:41 [547524432368] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 2
2020-07-29 19:34:41 [547524432368] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: MyKVStream and upload handle: 2 returned: {"EventType":"ERROR","ErrorCode":"INVALID_MKV_DATA","ErrorId":4006}

MushMal, 2020-07-30:
You are getting STATUS_ACK_ERR_FRAGMENT_DURATION_REACHED which indicates thats your pipeline is not producing I-frames which will drive the fragmentation. You might want to re-encode your file to have frequent fragments (less than 10 seconds long).

ishangupta3, 2020-07-30:
@MushMal thanks for the response, I will do that. Also, do you have any resources where I can better understand the different errors for this plugin?

MushMal, 2020-07-30:
The documentation covers most of them: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html

However, the documentation could go stale due to fast paced development so if there is an error you don’t see there you can check the public include files in the native repository

ishangupta3, 2020-07-30:
yessir thank you @MushMal and @disa6302

MushMal, 2020-07-30:
I am going to resolve this issue but I encourage re-opening issues or cutting new ones to get help directly from us or from community members.

closed 2020-07-30 by MushMal