awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #532: Integrate SDK with DeepStream

pmuvva, 2020-08-01:
Hi, 

I am working on NVIDIA DeepStream SDK 5.0. I want to integrate kiness-video-Producer SDK with Deepstream. 
I am new to amazon-kinesis-video-streams-producer-sdk-cpp. Please guide me how to integrate it.

MushMal, 2020-08-02:
@pmuvva are you looking for help with DeepStream or the KVS? I would recommend asking questions regarding DeepStream in the forums related to it to get more information on how to configure, extract frames and the metadata. Once you have that, you can use the samples in C Producer SDK to push those frames into KVS.

Alternatively, if they have a GStreamer integration then you could use KVS GStreamer plugin.

Please let us know how we can help.

muvva-sensable, 2020-08-13:
Thanks for reply. I want to control the frame rate, when I ran below command without frame rate change it works:
gst-launch-1.0 -v filesrc location="/home/path/test8.mp4" ! qtdemux name=demux ! queue ! h264parse ! kvssink name=sink stream-name="MyKVStream" access-key="xxxxxxx" secret-key="xcccxx"

When I am trying to control frame rate below command:
 gst-launch-1.0 -v filesrc location="/home/path/test8.mp4" ! qtdemux name=demux ! queue !videoconvert !video/x-raw,framerate=30/1 !nvv4l2h264enc ! h264parse ! kvssink name=sink stream-name="MyKVStream" access-key="xxxxxxx" secret-key="xcccxx" I am getting error: 

Look like I am doing some fundamental mistake. Please help me where am I making mistake?

muvva-sensable, 2020-08-14:
Hi, I am implementing Kinesis producer in Gstreamer. My aim is to upload the .mp4 from 30fps to 25fps. To achieve this operation, do I need to include "converter-caps-encoder" in the pipeline. Please give me some example to convert 30fps to 25fps of input file and upload to kinesis. Please help me to clarify this issue.

disa6302, 2020-08-21:
> Thanks for reply. I want to control the frame rate, when I ran below command without frame rate change it works:
> gst-launch-1.0 -v filesrc location="/home/path/test8.mp4" ! qtdemux name=demux ! [...]

What is the error?

hassanctech, 2020-08-24:
@muvva-sensable @pmuvva Generally speaking if you wish to change the framerate of video, you must decode it, change the framerate and then re-encode it.  There is a [videorate](https://gstreamer.freedesktop.org/documentation/videorate/index.html?gi-language=c) GStreamer plugin but it doesn't work directly on h264.

Starting from an mp4 (which I assume is h264 encoded) you will need to introduce a decode element which gives you video/x-raw, then after that have the videorate element to change the framerate from 30fps to 25fps.  Finally you will need to re-encode the video.  Please let me know if this does not help, we can dive deeper into this.

For reference, Here is a sample to help upload an mp4 file to KVS:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_file_uploader_sample.cpp

muvva-sensable, 2020-08-27:
Thank you for explaining detail description. It works.

hassanctech, 2020-08-27:
You're welcome, I'm glad it's working!  I'm going to close this issue, please don't hesitate to reach out again if you hit any other issues!

closed 2020-08-27 by hassanctech