awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #533: How to reuse cmake build file for new machine?

thasin96, 2020-08-03:
I have installed all the dependencies and library files are required to run C++ Producer SDK application in my machine (ubuntu 16.04). I want to run the application on new docker machine (ubuntu 16.04).
Is it possible to transfer my machine entire directory into new docker image so that I can minimise running time of installation process of  all the dependencies and library files which are required to  run C++ Producer SDK application? 

Currently, I am installing all the dependencies on docker as well. This takes huge amount of time. I want to minimise this process. What is the best approach to handle this situation?

lherman-cs, 2020-08-03:
@thasin96 I would not recommend to reuse the built files as they might have absolute paths from your old machine. But, if you really want to do that, you could try to copy the "open-source" folder, this will give you the shared dependencies and include files. But again, it's very likely that you need to change some paths.

Since you're using docker, probably it's worth to write your Dockerfile and make sure that the compilation step happens at the beginning so that if something changes after the compilation step, you don't have to recompile everything since it's going to be cached.

thasin96, 2020-08-04:
@lherman-cs - Okay, you suggest me to build libs in docker as well right?

lherman-cs, 2020-08-04:
@thasin96 Yes, that's my recommended path. Since your question has been answered, please feel free to reopen it if you need further assistance!

closed 2020-08-04 by lherman-cs

thasin96, 2020-08-06:
@lherman-cs - I have one more question. What are all the required folders/libs, If I just want to run the already created application? In this case I don't want to compile the code in docker environment. I would compile the code in some other machine. I just want to run the application in docker environment.

lherman-cs, 2020-08-07:
I'd recommend to build it in docker environment because it can ensure its compatibility and repeatability. I'm not sure why compiling in the docker environment an issue. If you're worried about the image size, you can easily do this with multi stage build, https://docs.docker.com/develop/develop-images/multistage-build/. 

Also, I think you'll get a much better answer from the docker community.