awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #535: [QUESTION] How to troubleshoot "Missing codec private data error"

suggestedfixes, 2020-08-06:
We are using gstreamer kvssink, and for one of our devices. We kept getting
"Missing codec private data error", once we restart the app, the error disappears.
Missing codec private data errorRefresh
The fragment did not contain any codec private data. Ensure that the producer is generating valid codec private data.

Details
Status code: 400
Request ID: 39952d7e-80e3-4868-badb-4af9cef35f9f
Region: us-east-2
Time: August 5, 2020 8:57:28 PM

MushMal, 2020-08-06:
This is how the kvs sink applies the CPD for video: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L933

Basically, it checks if it has already applied the CPD and if not, attempts to get it from the gstreamer upstream encoder element. It then applies the CPD using start(cpd) method.

You might want to debug this on your platform and understand why the upstream doesn't provide the CPD.

chehefen, 2020-08-07:
How did you get the error in the first place? I'm asking because if restarting the app fix the issue, it should also work when you start the app for the first time. Can you give more details on how you get the error?

suggestedfixes, 2020-08-08:
We have two mechanisms for restarting
- restarting gstreamer
- restarting the app

One thing we are suspecting is that CPD got lost during gstreamer restart.

suggestedfixes, 2020-08-08:
We are still trying to capture the logs.

MushMal, 2020-08-08:
I am not sure I understand your application's architecture. Assume your application is using kvssink directly. How do you re-start the application? It will also restart the gstreamer pipeline as the process teardown will teardown the gstreamer user pieces.

Currently, the SDK stores the CPD at the beginning of the stream object creation and reuses it for consecutive sessions - it doesn't re-request the CPD. If you re-create the KVS stream object, it will need to be supplied a new CPD. In case of the kvs sink, it will have to extract the CPD from the gstreamer upstream element. 

You need to make sure when the new stream is created, the gstreamer pipeline supplies the CPD. If you have a custom solution, however, then you need to perhaps modify the code and supply your CPD that you store directly by calling stream->strart(cpd) as I pointed earlier.

disa6302, 2020-08-21:
@suggestedfixes , 

Any updates?

suggestedfixes, 2020-09-01:
@disa6302 This got queued up along with other issues
I found the code snippet the person is working on this on_new_sample():
I'm suspecting that sometimes, when the first on_new_sample() is evoked, it might not have the cpd ready.
```
 if (!data->stream_started) {
        data->stream_started = true;
        GstCaps* gstcaps  = (GstCaps*) gst_sample_get_caps(sample);
        GstStructure * gststructforcaps = gst_caps_get_structure(gstcaps, 0);
        const GValue *gstStreamFormat = gst_structure_get_value(gststructforcaps, "codec_data");
        gchar *cpd = gst_value_serialize(gstStreamFormat);
        data->kinesis_video_stream->start(std::string(cpd));
        g_free(cpd);
    }
```

hassanctech, 2020-10-05:
To verify the CPD is not null, right after this line:

`gchar *cpd = gst_value_serialize(gstStreamFormat);`

can you try:
```
if (cpd == NULL) {
g_printerr("CPD is NULL!\n");
}
```

This way we can narrow down what the root cause is, if it is in fact null we might need to skip frames until encountering a key-frame which would have the CPD.

MushMal, 2020-10-10:
@suggestedfixes could you please summarize the issue? I am still unsure whether this is actionable from the KVS side.

hassanctech, 2020-10-30:
I'm closing this due to 30 days of inactivity.  @suggestedfixes you can open a new issue if you still need help.

closed 2020-10-30 by hassanctech