awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #536: [BUG]Build failed Windows

amitrjaiswal1, 2020-08-08:
Most of the part looks downloaded successfully but gives below error after around 30 mins..
Please help.
**************************************************************************************************************

-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPkgConfig.cmake:45 (find_package_handle_standard_args)
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:109 (find_package)

MushMal, 2020-08-10:
Seems you are missing pkgconfig per error output

amitrjaiswal1, 2020-08-10:
Where to get this ?

> Seems you are missing pkgconfig per error output

MushMal, 2020-08-10:
Here are some answers on how to install: https://stackoverflow.com/questions/1710922/how-to-install-pkg-config-in-windows

hardikchimedia, 2020-08-11:
Where to install pkgconfig for MSVC ?

MushMal, 2020-08-11:
There is a project called pkg-config-lite https://sourceforge.net/projects/pkgconfiglite/ you might try.

GStreamer is packaged with the executable: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/CMakeLists.txt#L120

amitrjaiswal1, 2020-08-13:
**Thanks Mush. Looks like there is some progress with this. After trying further step it is not progressing....can you please help us to move ahead.
Below is build result...**

[image]

MushMal, 2020-08-13:
Looks like you are not building dependencies such as gtest. Are you specifying `-DBUILD_TEST=TRUE -- Build unit/integration tests, may be useful for confirm support for your device. ./tst/webrtc_client_test` per readme?

amitrjaiswal1, 2020-08-16:
Below command from bat file was executed..

cmake -G "NMake Makefiles" -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE ..

disa6302, 2020-09-21:
@amitrjaiswal1 , 

Based on your comment, it looks like the build succeeded. Closing this.

closed 2020-09-21 by disa6302