awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #538: [BUG] Use of `using namespace std;` and `using namespace std::chrono;` should be forbidden and breaks this code's ability to be used in many cases

emersonknapp, 2020-08-11:
**Describe the bug**
Headers in this library use `using namespace std;` and `using namespace std::chrono;` (see example https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/StreamDefinition.h#L18) 

This means that, for example, when I link against a library that uses boost's `time_point`, my code cannot compile because of ambiguous definitions. It is C++ best practice to never `using namespace` in a header because it can inadvertently change the meaning of downstream code, and its effects cannot be undone by the consuming code.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**SDK version number**
Include the SDK version you are running. If it is a specific commit on master branch, include that

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: [e.g. Linux, Windows]
 - Processor: [e.g. ARMv5]
 - Version [e.g. 18.04]

**Additional context**
Add any other context about the problem here.

MushMal, 2020-08-11:
+1 on this. @emersonknapp do you want to to take a stab at it? We would love to get public contributions to the codebase

chehefen, 2020-08-17:
closing this issue as it has been fixed in #544

closed 2020-08-17 by chehefen