awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #542: How to change authentication mechanism in Producer client?

thasin96, 2020-08-13:
The current SDK uses secret key/ access key pair to create client credential. I would like to use x.509 certificate to create client credential. How to change from secret key authentication mechanism ti X.509 certificate authentication mechanism? If you need further details, please let me know.

MushMal, 2020-08-13:
C++ Producer layer provides for a generic auth integration model where you could implement your own mechanisms for fetching the AWS session tokens. One implementation is based on IoT integration and it's provider is in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/credential-providers/IotCertCredentialProvider.h 

You will need to create IoT certificates and install those on the device.

thasin96, 2020-08-14:
@MushMal  - could you tell me the docs related to how to generate session token using `IotCertCredentialProvider.h`? I have got certificate and I have no clue how to proceed further

MushMal, 2020-08-14:
Here is an article describing how to obtain the certificate, the private key and the endpoint which you feed to create the IoT provider. Once you have the IoT provider you pass it to the producer

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html

thasin96, 2020-08-14:
@MushMal - Following the suggested link I am able to get required token and other parameters. Using the parameters I am able to successfully process streaming. Is the same approach available for WebRTC (signalling) in c SDK. Because, I couldn't find the same flow in amazon-kinesis-video-streams-webrtc-sdk-c. Is passing session token possible in amazon-kinesis-video-streams-webrtc-sdk-c?

MushMal, 2020-08-14:
It’s exactly the same. You need to create an IoT credential provider and supply it to the signaling channel. There are some pointer if you search the issues.

MushMal, 2020-08-14:
As the primary question has been resolved, I am closing this issue. Please cut a new one against the appropriate repository if you have further issues with it

closed 2020-08-14 by MushMal