awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #547: C++ custom producer using cproducer - Can not view live stream

malibu, 2020-08-17:
Trying to create a producer using cproducer SDK api (following the sample KvsVideoOnlyStreamingSample.c from docker windows image, since we ran into build issues with the latest bits on windows as reported by other issues).

For the most part we're able to produce fragments using putFrame. However, we are running into issues viewing the live video in the console as well as using the javascript HLS viewer sample. It looks like the clip can be downloaded for specific duration from the console without issues.

Here's the setup.
Our backed vision application generates video frames after object detection with ROI embedded in the frames. These frames are coming from  a back end application as opencv mat frames (640 * 480). We encode the frames to h.264 using encoder (https://github.com/cisco/openh264). We feed the frames to a Kinesis video streams as encoded by the library. 

1. Create a stream with key-frame-fragmentation to true which as per answers here will use fragmentation based on I frames. This appears to be working fine. 
 NAL adaptation flags : NAL_ADAPTATION_ANNEXB_CPD_NALS | NAL_ADAPTATION_ANNEXB_NALS
 content_type :"video/h264",

2. The CPD info from encoder is extracted and used to start the stream (The console shows the correct codec. avc1.42c029 (H.264 Baseline@4.1))
3. Frames from the encoder are fed in using putFrame with FRAME_FLAG_KEY_FRAME set for generated I FRAMES.

Among other errors, we had issues with fragment time vs actual frame time that caused some frames to be dropped. However, those errors are fixed now since we adjusted the frame time (t) to be consistent from the frame t-1+ frame duration, putFrame submits without errors.

The current fragment time reported by the console is around 1.5 sec which is correct based on frame production.

Currently, we have not implemented any ACK or back pressure logic on the app side as we don't see any network or buffering issues for a this test configuration. Is this a must? What else are we missing?

MushMal, 2020-08-17:
@malibu the default SDK pressure policy attempts to do recover from the pressures or errors so you don't necessarily need to handle the ACKs if you rely on the default high-level logic/behavior. 
It's hard to tell what exactly is going on but seems that the console playback does indeed playback a few seconds and then can't find the consequent fragments.

Could you please include debug logs from the producer side so we can see whether you have any streaming ingestion issues in the first place? 

P.S. The black screen with no spinner indicates an issue with your encoder.

malibu, 2020-08-17:
Thanks for the quick response.
Here's the relevant sequences from the log (See attached). Hopefully, this shows enough.
It's always the spinning wheel now. If we go to the start of the stream, it displays about 2-3 seconds of good video then back to the spinning [wheel.](https://s3.amazonaws.com/novelcoronavirustracker.com/log1.txt)

MushMal, 2020-08-17:
The logs are for 7 seconds streaming to us-east-1 uploading 5 fragments. There are no problems with the producer itself during this time. What's your retention period? Could you select producer time playback in the console and try to playback from the beginning of the selected time?

It's hard to tell what's going on if you actually see a playback progressing but a black screen is displayed. That's a separate debug issue. But, for now, let's focus on the missing fragments issue first. If your log continues on with no problems with the upload then I am puzzled with why the backend can't find your fragments given the server ts or producer ts. There are some intricacies with stitching the playback timelines so I am not sure whether that's the issue but doesn't seem to be

malibu, 2020-08-18:
Ok thanks!. We will reduce it to the spinning wheel issue with "searching for media". We can skip to any time and see two seconds of video after that the spinning starts again. If you select "Download clip" from the console and try to download any valid time in the past live segments, it will download fine without issues. As such. the issue is continuous live replay which runs into the wait cursor.

MushMal, 2020-08-18:
I see. Yes indeed, the player doesn’t support the sparse playback timelines. You can choose the HLS or DASH playback which your player might or might not support. An alternative is to use server timestamp if the producer is continuous even with presentation timestamps that are sparse. If that doesn’t work then you will likely need a custom player that understands the sparse playback of your application.

I am not sure whether the original question has been answered and if so could I ask you to close this issue?

malibu, 2020-08-18:
> I see. Yes indeed, the player doesn’t support the sparse playback timelines. You can choose the HLS or DASH playback which your player might or might not support. An alternative is to use server [...]

I think maybe there is a misunderstanding from my previous post and a small snapshot of the log I attached. What I described was the way the console player is behaving, it's not a desired outcome. We have a continuous publisher and would like the player to do continuous playback just like any other ingestion. But currently, the console doesn't playback but shows the first two seconds from a seek time which you described as sparse playback.

It's puzzling because the fragments are ingested fine as established by the clip download from the console. It looks like there is some setting/limitation in the ingestion or the inability to parse certain codecs by kinesis video. I noticed that our encoder puts out two consecutive IDR frames which potentially could be the source of trouble.

The HLS player sample also shows the same behavior We can create a custom player to circumvent the issues. However, it would be good to know what trips the console player and if there is any setting that can be changed during ingestion to make the continuous play work which would be helpful to make the HLS player work too.

MushMal, 2020-08-18:
This is quite odd. Let me confirm a few points...

1) Your playback fails with HLS test page and the console in both Producer and Service timestamps - it plays back a few fragments and stops with a spinner.
2) Your producer does get Persisted ACKs for all of your fragments
3) ListFragments API does not return the Persisted fragments with various selectors.

The e2e should work OK even if occasionally you have a single frame fragments.

Action items:
In order to get this debugger properly, could I ask you to send an email to kinesis-video-support@amazon.com with the following info

1) Producer side debug logs - from start (to capture the settings you are using) and for a few minutes
2) Include your account ID and stream name (though, the stream name can be extracted from the logs)
3) Which assets you are using and what version - e.g. Producer C++ SDK with the commit hash: 123xxx
4) Include the ListFragments API parameters you are using.

disa6302, 2020-09-28:
What is the latest on this ticket?

MushMal, 2020-10-10:
Resolving due to inactivity

closed 2020-10-10 by MushMal