awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #548: KVS Producer Multi Stream with Session Token Authentication stream is being stopped abruptly.

thasin96, 2020-08-18:
**Describe the bug**
I am using amazon-kinesis-video-streams-producer-sdk-cpp/ kvs_gstreamer_multistream_sample to perform streaming AWS KVS. I have modified default flow of authentication. Initially, I passed My account Access/ Secret Key to run the application. Now, instead of using that I resisted my device on IOT hub and getting temporary Access/ Secret Key and Session token. After taking the new credentials, I am keeping these into Env Variable, like below 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**SDK version number**
Include the SDK version you are running. If it is a specific commit on master branch, include that

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: [e.g. Linux, Windows]
 - Processor: [e.g. ARMv5]
 - Version [e.g. 18.04]

**Additional context**
Add any other context about the problem here.

MushMal, 2020-08-18:
From the log it seems that you are hitting fragment duration error:
```

[DEBUG] [18-08-2020 06:07:24:321.538 GMT] postWriteCallback(): Curl post body write function for stream with handle: durability_check and upload handle: 2 returned: {"EventType":"ERROR","FragmentTimecode":1597730810769,"FragmentNumber":"91343852333181476993185798004477358642159968820","ErrorCode":"MAX_FRAGMENT_DURATION_REACHED","ErrorId":4002}

[DEBUG] [18-08-2020 06:07:24:321.806 GMT] streamErrorHandler invoked
[ERROR] [18-08-2020 06:07:24:321.920 GMT] Reporting stream error. Errored timecode: 1174840000 Status: 1375731806
[WARN ] [18-08-2020 06:07:24:322.011 GMT] continuousRetryStreamErrorReportHandler(): Reporting stream error. Errored timecode: 1174840000 Status: 0x5200005e
```

The SDK logic then skips the "bad" fragment and continues with the next fragment as it makes no sense to retry the "bad" fragment. 

What happens later is that your media pipeline stops producing frames which causes the networking thread to time out:
`
[WARN ] [18-08-2020 06:08:16:393.397 GMT] curlCompleteSync(): curl perform failed for url https://s-670061c1.kinesisvideo.ap-southeast-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
`

This is the end of the log and I am not sure what's going on later.

Recommend to check your encoder/media pipeline settings and the way you integrate with the SDK. The max fragment duration is defined as 10 seconds so if you have configured the SDK with key-frame-fragmentation then your media pipeline should produce an I-frame every less than 10 seconds.

MushMal, 2020-08-18:
Removed "bug" tag as this is not a bug but an actual designed behavior

thasin96, 2020-08-18:
@MushMal  - Okay. I will try with some other RTSP stream. Meanwhile, Could you tell me What happens to stream when session is expired?

MushMal, 2020-08-18:
If you specify the expiration correctly, the SDK will refresh the token (ask the credential provider to refresh) and will use the fresh token to create and stream a new session. The max enforced duration is 45 mins for the SDK - random jitter up-to 3 minutes

MushMal, 2020-08-18:
Can you please resolve the issue if your main question is answered

thasin96, 2020-08-18:
okay sure. I have started, I will update you soon.

thasin96, 2020-08-18:
@MushMal - You are right. After changing Stream source, I am able to send stream  without any disturbance. I need some help in Session token. Shall I continue the refresh token mechanism doubt here? or do you want me to close this thread and create a new thread for the discussion?

MushMal, 2020-08-18:
We prefer the issues being on topic as they will be easier to search and reference for others who might have similar issues,

Please go ahead and resolve this issue. Cut a new issue with detailed information of the issue you are facing or if you have a question then include the details to avoid back-and-fro

closed 2020-08-18 by thasin96