awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #549: How Session token refresh works in AWS KVS?

thasin96, 2020-08-18:
I used to pass my Access/Secret Key to create stream for each edge device to start streaming. Somehow, I do not want Access/Secret Key to be stored on edge device because it's not safe. So, I have registered each edge device into AWS IOT. From there I got Certs. Using these Certificates, I am generating Temporary  Access/Secret Key and Session Token. The expiration of theses keys is 1 hour. To automate this process, I have written a python script that will communicate to credential server using my certificates and calls Producer application by setting environment variables. 

I am calling Producer service like below,
```
export AWS_ACCESS_KEY_ID={{ aws_access }}
export AWS_SECRET_ACCESS_KEY={{ aws_secret }}
export AWS_DEFAULT_REGION={{ region }}
export AWS_SESSION_TOKEN={{ aws_token }}
./amazon-kinesis-video-streams-producer-sdk-cpp/build/kvs_gstreamer_multistream_sample {{ rtsp_file }}
```

My question is,

 Assume I have started a stream using session token. The stream is being processed without any issue. 

1. What happens to the application once the session token is expired (The application runs more than 1 hour)? 
2. Does SDK require new token to continue the running application? If so, how can I pass session token to already running service? 

Note: For every time a  stream is started I am fetching a new session token. I am dealing this case externally. My only concern is about running streams whose session token is expired. 

Please let me know, If you require any additional details.

MushMal, 2020-08-18:
If you are integrating with IoT then you don't need any of this complexity. If you are using C++ SDK then check out https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/credential-providers/IotCertCredentialProvider.h which is an IoT specific credential provider. Simply supply this provider to the Producer object and it will do the rest. 

Similarly, if you are using C Producer repository then there is the core IoT credential provider there.

thasin96, 2020-08-18:
@MushMal - yeah I am integrating with IoT. could you tell me how to  supply this provider to the Producer object? If I get documentation how to use IoT credential provider, it would be easy for me to understand and integrate.

MushMal, 2020-08-18:
The documentation is scarce but the header files themselves are self-explanatory. You can create the credential provider object - in this case IoT credential provider using it's public constructor. Then, when you create your producer object, you supply it: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/KinesisVideoProducer.h#L80

You can look at the sample how the producer object is created: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp#L520

thasin96, 2020-08-18:
@MushMal - based on your suggestion, I tried to modify `kvs_gstreamer_multistream_sample`.

In this what should be assigned to `stream_name`?
Because kvs_gstreamer_sample.cpp is calling `kinesis_video_init` function for single stream. 
In my case it's multiple. What should be the value assigned for `data->stream_name`?

thasin96, 2020-08-18:
@MushMal - After updating stream_name as thing name it starts working.

thasin96, 2020-08-18:
@MushMal - As per your suggestion, I have integrated IOT in cpp. I want to do the same thing for WebRTC c SDK as well.I coudn't find iot related files in c SDK. could you help me?

MushMal, 2020-08-18:
https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c/search?q=IOT&type=Issues

Multiple answered questions

disa6302, 2020-08-23:
Closing this. Feel free to open a new issue with any questions you might have. Thanks!

closed 2020-08-23 by disa6302