awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #550: [QUESTION] 'kvssink' plugin not built correctly

hwanders97, 2020-08-20:
I followed the instructions to build the kvssink plugin on my Mac:
1. Install all of the dependency libraries (including gstreamer)
2. Run `cmake .. -DBUILD_GSTREAMER_PLUGIN=TRUE` in the build directory
3. Run `make` in the build directory
- `libgstkvssink.so` is in the build directory
4. I set GST_PLUGIN_PATH correctly: `/Users/hwanders/workplace/KVSDemo/amazon-kinesis-video-streams-producer-sdk-cpp/build`
5. I set LD_LIBRARY_PATH correctly: `/Users/hwanders/workplace/KVSDemo/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib`

Running `gst-inspect-1.0 kvssink` or any other gstreamer command with `kvssink` yields: `No such element or plugin 'kvssink'`

Can you let me know how I can fix this? Thank you!

MushMal, 2020-08-20:
This is rather strange. Seems like you are using the proper instructions to set the paths defined here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp#loading-element

Don you happen to have an older GStreamer installed on the system somewhere?

hwanders97, 2020-08-20:
Running `gst-launch-1.0 --gst-version` results in `GStreamer Core Library version 1.16.2`

I had previously tried running the install-script and min-install-script of the older version of this library. Could that have impacted my environment?

MushMal, 2020-08-20:
It might have issues if you are running gst-launch from one version but linked the library to another. As I am out of ideas assuming you double checked the paths, I recommend wiping the old gstreamer out and starting from scratch

hwanders97, 2020-08-20:
I ran `brew uninstall gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly gst-libav`
Then reinstalled `gstreamer` and `gst-plugins-*`.
After cmake and make again I still see `No such element or plugin 'kvssink'`.

Can you tell me what version of the gstreamer I need? Or how I can check where the library is linked?

MushMal, 2020-08-20:
otool -L <lib> will dump the dependent library linkage.

I am not sure why you are having such an issue. Just to double check a couple of things.. could you confirm

* You actually have the library in your build directory under file name `libgstkvssink.so`
* You have run the following command from the ROOT directory of your repository:
```
export GST_PLUGIN_PATH=`pwd`/build
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
```

Also, could you run gst-inspect-1.0 kvssink and include the output?

The version of the GStreamer that's used in the one in brew. Here is the line that installs the prerequisites from the Readme

`$ brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav`

hwanders97, 2020-08-21:
I got things working after following the advice in this issue: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/550

Since I am using MacOS Catalina, I needed to turn off System Integrity Protection (SIP). I restarted my computer and followed the steps here: https://help.mulesoft.com/s/article/Variables-LD-LIBRARY-PATH-DYLD-LIBRARY-PATH-are-ignored-on-MAC-OS-if-System-Integrity-Protect-SIP-is-enable

I reinstalled all dependencies and rebuilt the sdk. 

`git-inspect-1.0 kvssink` works along with the macos examples for streaming your Mac webcam to a KVS stream.

Thank you for your help!

closed 2020-08-21 by hwanders97

himadrii27, 2024-06-05:
on running the command cmake .. -DBUILD_GSTREAMER_PLUGIN=ON
its giving me errors like 'this source directory doesnt appear to contain CMakeLists.txt ' can you help me with this?

niyatim23, 2024-06-05:
I think you're not running cmake from the `build` folder. Please follow the steps on the readme or the CI:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/49fa6cbb905958726fe54d98b837c3edf895025d/.github/workflows/ci.yml#L243-L278

himadrii27, 2024-06-06:
Now i am getting this error. Do you have any idea how to fix this?
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:74 (build_dependency)