awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #551: How to store AWS Kinesis video Stream to S3 bucket?

bhavikapanara, 2020-08-21:
I am getting the stream from my IP camera successfully using Gstreamer. How can I store it to S3 bucket?

Thanks,
Bhavika

MushMal, 2020-08-21:
You can retrieve the fragments using ListFragment API following with GetMediaForFragment or using GetMedia realtime API. Check out the public API documentation: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_Operations_Amazon_Kinesis_Video_Streams_Archived_Media.html

You can also view the parser library and some of the samples in https://github.com/aws/amazon-kinesis-video-streams-parser-library

bhavikapanara, 2020-08-21:
@MushMal Thanks for your quick reply.

Actually I want to get stream using Python. Here I have written the code to retrieve frames from Kinesis.

```
import boto3

kinesis_client = boto3.client('kinesisvideo',region_name='us-east-2')

response = kinesis_client.get_data_endpoint(StreamARN='arn:aws:kinesisvideo:us-east-2:291596880574:stream/MilkmanStream/...' , APIName='GET_MEDIA')

endpoint = response['DataEndpoint']

video_client = boto3.client('kinesis-video-media',endpoint_url=endpoint,region_name='us-east-2')

stream = video_client.get_media(StreamARN='arn:aws:kinesisvideo:us-east-2:291596880574:stream/MilkmanStream/...',StartSelector={'StartSelectorType': 'NOW'})

with open('video_rtsp_01.webm', 'wb+') as f:
    chunk = stream['Payload'].read(1024*8)
    while chunk:
        f.write(chunk)
        chunk = stream['Payload'].read(1024*8)
```

But, this python script not able to store payload data even I can able to get something in the payload.

Please help to resolve this issue. What is wrong with this?

disa6302, 2020-08-23:
@bhavikapanara , 

Closing this issue since the question is answered. Feel free to reopen or create a new issue if you have any questions. Thank you!

closed 2020-08-23 by disa6302

BodapatlaKiranReddy, 2021-01-22:
@MushMal @disa6302 
Hi,

I am able to push live video to Kinesis video stream but i want that video to be stored in S3 bucket using lambda function

i don't have idea how to do it in java, any help appreciated

MushMal, 2021-01-22:
Parser library has some examples of doing so: https://github.com/aws/amazon-kinesis-video-streams-parser-library

Some information about the samples is available in the readme. You might take a look at OutputSegmentMerger to create a file from fragments retrieved from the call to GetMedia and then you can use AWS SDK to store those files in S3 if you want.

shahid-zain, 2021-08-13:
@MushMal @bhavikapanara 
Hi guys,
I am able to push live video to Kinesis video stream but now I want to detect text from the video using aws rekognition service,
How do I do that.