awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #552: [QUESTION] Why PutMedia stop automatically every 2/3 days

FlorianRuen, 2020-08-24:
Hi there,

I'm using the KVS stream producer running on Raspberry PI 4 to send stream from RTSP camera.
Every works fine, my stream is available on KVS Dashboard and I can ask for HLS to get the stream on my web application.
But when my application run for 2 or 3 days, my application doesn't stop (still sending something, because there is some network up/down traffic), but nothing on KVS. So I need to restart my app to get something works again.

** Any design considerations/constraints **
My application ask a database to get some informations about camera, RTSP url or something else. Create the command (gst-launch ...) and run the command using Subprocess.Popen in new thread (one camera = one thread) and run it in a while (because we check some working hours to run or stop the pipeline)

As you can see, this is the chart on KVS: https://ibb.co/m8n1vQk

I opened a new case on the KVS Parser Library and it seems there is a limitation on KVS Session, maybe there is something similar on producer, and I need to regenerate credentials or something else ?

I can't find anything in your documentation on this subject, and there is nothing on my application side that can cause stop sending stream (app is still in the loop when the stream stop).

Thanks for your help, hope we can find a solution;

Kind regards,
Florian.

MushMal, 2020-08-24:
@FlorianRuen you can think of the producer and the consumer as separate entities. The issue regarding the consumer have been answered - unfortunately, currently, there is no automatic session refreshing done as the case is with the producer.

In case of your producer, you need to do a little more debugging to understand what's going on. The information provided is too vague to see whether it's an issue with the producer or the RTSP or the elementary stream or the consumer API that you are using. 

* Try to capture the producer logs when the issue happens
* Try to check the streaming using AWS console - see if there is playback
* Check your CW and see whether the fragments are still being produced
* Check your RTSP source

FlorianRuen, 2020-08-24:
@MushMal 

I will try to capture the producer logs into file during 3 days and I will check if it's stop.

My playback duration is 1 hour so if the pipeline stop at 1 or 2 AM, I can't see the playback at the morning (it generally happend during the night if I check the KVS charts)

My RTSP source works all the time, becaue I can see the stream from the Camera web interface directly without any problems, so we can remove this issue.

The dashboard of KVS say "No data" when I check, so I don't know:

- If the credentials are invalid, so my application send the fragment but nothing is treat by KVS
- There is a network issue: my application send the fragment but something block the fragments

MushMal, 2020-08-24:
Not sure I understood what you're stating... So, when this issue happens and you can't see the playback using HLS - do you see the playback on the console media view page? Does it give you "no fragments found"? If so, it's a tale-tale sign that the producer is not uploading and something happened at the producer side. We really need the logs to understand what's going on.

If the credentials are invalid, the SDK will fail to communicate with the backend. The default continuous retry policy handler at higher layer will retry but will fail over and over.

FlorianRuen, 2020-08-24:
Yes it say not fragments found and the charts say "No Data" so it seems there is not fragments.
I just launch the pipeline and write all the logs in a file, I will wait the pipeline stop and I will check the logs to get the error message.

But what it's very strange, it's I'm using subprocess from Python and the subprocess catch the error message from command, in this case it doesn't stop, so no error message from Gstreamer

MushMal, 2020-08-24:
You could try to redirect the debug console log output from the kvs sink (I assume you are using the KVS sink) into a file that you could later see.

FlorianRuen, 2020-08-24:
Yes it's what I'm doing, now I'm waiting the pipeline to stop, to see if there is an error code or something else that can explain with there is no more video stream on KVS

MushMal, 2020-08-24:
Thanks. Please make sure you have the debug spew enabled to capture the details

FlorianRuen, 2020-08-25:
@MushMal Yestarday, I launched the pipeline to see what happend, and today, the stream was stop, but when I try to launch it again, I can't because I get an error (maybe the both is related to the same issue).

Here is the logs (I just remove the line in red to hide my RTSP source credentials):
`Définition du pipeline à PAUSED...
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2020-08-25 16:03:32 [3069599760] INFO - heapInitialize(): Initializing native heap with limit size 268435456, spill ratio 0% and flags 0x00000001
2020-08-25 16:03:32 [3069599760] INFO - heapInitialize(): Creating AIV heap.
2020-08-25 16:03:32 [3069599760] INFO - heapInitialize(): Heap is initialized OK
2020-08-25 16:03:32 [3069599760] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-08-25 16:03:32 [3069599760] DEBUG - getSecurityTokenHandler invoked
2020-08-25 16:03:32 [3069599760] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1598364212354415714 Expiration: 0
2020-08-25 16:03:32 [3069599760] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2020-08-25 16:03:32 [3069599760] INFO - createDeviceResultEvent(): Create device result event.
2020-08-25 16:03:32 [3069599760] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-08-25 16:03:32 [3069599760] DEBUG - clientReadyHandler invoked
2020-08-25 16:03:32 [3069599760] INFO - try creating stream
2020-08-25 16:03:32 [3069599760] INFO - Creating Kinesis Video Stream 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy
2020-08-25 16:03:32 [3069599760] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Stream name: 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy 
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Content type: video/h264 
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Framerate: 25
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): Track list
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Track id: 1
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2020-08-25 16:03:32 [3069599760] DEBUG - logStreamInfo(): 	Track cpd: NULL
2020-08-25 16:03:32 [3069599760] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-08-25 16:03:32 [3069599760] INFO - writeHeaderCallback(): RequestId: 5c46896c-3ff7-4378-860a-4d2a945e7ace
2020-08-25 16:03:32 [2765481056] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.597654777819E9,"DataRetentionInHours":1,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:eu-central-1:150832573572:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:eu-central-1:150832573572:stream/21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy/1597654777819","StreamName":"21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy","Version":"zTHeDHdPTFMjl
2020-08-25 16:03:32 [2765481056] INFO - describeStreamResultEvent(): Describe stream result event.
2020-08-25 16:03:32 [2765481056] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2020-08-25 16:03:32 [2765481056] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2020-08-25 16:03:32 [2765481056] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2020-08-25 16:03:32 [2765481056] INFO - writeHeaderCallback(): RequestId: e937e266-0313-4fd1-9910-03b5d6d583a1
[... cut ...]
Execution ended after 0:00:20.054500507
Définition du pipeline à PAUSED...
Définition du pipeline à READY (prêt)…
Définition du pipeline à NULL…
Libération du pipeline…
INFO - Freeing Kinesis Video Stream 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy
2020-08-25 16:03:59 [3069599760] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2020-08-25 16:03:59 [3069599760] DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
2020-08-25 16:03:59 [3069599760] `

Any idea ?

MushMal, 2020-08-25:
If my French is good enough then I can guess that "Le pipeline est actif et n’a pas besoin de phase PREROLL…" means the pipeline failed to preroll.

This is unrelated to KVS. You can run your solution with GST_DEBUG=5 to get more information about the GStreamer error and see which component fails to preroll.

Another thing that I would recommend is to run the pipeline without KVS all together - just run in with a fakesink for example.

If it's something not very obvious then I would recommend cutting an issue in GStreamer forums to get some help.

FlorianRuen, 2020-08-25:
Your french is pretty good ! At one detail

It means that the pipeline is already active so the PREROLL isn't necessary. I don't think this is an issue with Gstreamer.

My pipeline is currently running when I write this message using GST DEBUG level 6 to get the maximun logs we can capture

MushMal, 2020-08-25:
From the log, it seems that the plugin is just starting - this is why we have the initialization in the kvssink. Is there anything that forces the kvssink to terminate?

FlorianRuen, 2020-08-25:
The only thing that can terminate the kvssink pipeline is a working hours check. Every minute, I check if current time is in working hours (between two datetimes). If not, I force the pipeline to stop and add a cron job to run again at specific time.

This is the only thing that can stop the kvssink, and in my case, the working hours are not set, so by default, the pipeline need to run for unlimited time

MushMal, 2020-08-25:
I really have no clue. You might want to debug into this issue but from KVS perspective it's unrelated.

As suggested, you could try to run your pipeline without kvssink and with fakesink to eliminate KVS all together.

FlorianRuen, 2020-08-25:
I'm waiting for the pipeline to stop (if it stop), and I'm coming back with the full logs to be sure. If we still have no clues, I will try without kvssink.

FlorianRuen, 2020-08-26:
@MushMal I'm back on this subject, because the pipeline was stop this night and I got the full logs.

Here is the last lines before the pipeline stop. I have the full logs since the pipeline was start, I can send the full files (18mb) to you with private message or e-mail if you want (or if necessary).

I don't really understand the last lines of logs, because it seems, one of pipeline element send an EOS to the RTSP source, which explain why everything stop. But I don't know why EOS is sent here.

 **I've translate all French lines to English one.**

```
2020-08-26 01:24:00 [2644481120] DEBUG - fragmentAckReceivedHandler invoked
2020-08-26 01:24:00 [2644481120] DEBUG - postReadCallback(): Wrote 942 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:00 [2644481120] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 56
2020-08-26 01:24:00 [2644481120] DEBUG - postReadCallback(): Wrote 2119 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:01 [2644481120] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 56
2020-08-26 01:24:01 [2644481120] DEBUG - postReadCallback(): Wrote 3131 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:02 [2644481120] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 56
2020-08-26 01:24:02 [2644481120] DEBUG - postReadCallback(): Wrote 3596 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:03 [2644481120] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 56
2020-08-26 01:24:03 [2644481120] DEBUG - postReadCallback(): Wrote 4531 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:04 [2644481120] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 56
2020-08-26 01:24:04 [2644481120] DEBUG - postReadCallback(): Wrote 2773 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:05 [2644481120] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 56
2020-08-26 01:24:05 [2644481120] INFO - stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 0000000001de48d8.
2020-08-26 01:24:33 [2678051936] INFO - getStreamData(): Handle 56 waiting for last persisted ack with ts 15983978410020000
2020-08-26 01:24:34 [2644481120] DEBUG - postReadCallback(): Wrote 84 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:34 [2644481120] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 56 waiting for last ack.
2020-08-26 01:24:34 [2644481120] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy and upload handle: 56 returned: {"EventType":"RECEIVED","FragmentTimecode":1598397841002,"FragmentNumber":"91343852333185131377326487975803346104476006664"}

2020-08-26 01:24:34 [2644481120] DEBUG - fragmentAckReceivedHandler invoked
2020-08-26 01:24:34 [2644481120] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy and upload handle: 56 returned: {"EventType":"PERSISTED","FragmentTimecode":1598397841002,"FragmentNumber":"91343852333185131377326487975803346104476006664"}

2020-08-26 01:24:34 [2644481120] DEBUG - fragmentAckReceivedHandler invoked
2020-08-26 01:24:34 [2644481120] INFO - getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 56
2020-08-26 01:24:34 [2644481120] DEBUG - streamClosedHandler invoked for upload handle: 56
2020-08-26 01:24:34 [2644481120] DEBUG - Reported streamClosed callback for stream handle 31344856. Upload handle 56
2020-08-26 01:24:34 [2644481120] INFO - postReadCallback(): Reported end-of-stream for stream 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy. Upload handle: 56
2020-08-26 01:24:34 [2644481120] DEBUG - postReadCallback(): Wrote 0 bytes to Kinesis Video. Upload stream handle: 56
2020-08-26 01:24:34 [2644481120] INFO - Sending eos
2020-08-26 01:24:34 [2678051936] EOS received from element "pipeline0".
Execution ended after 9:12:01.484842895
Definition of the pipeline at PAUSED ...
Set the pipeline to READY (ready)…
Setting the pipeline to NULL ...
Release of the pipeline ...
INFO - Freeing Kinesis Video Stream 21hc1y8v2w70dgljebrec5e80blzu5smwrwqx44vl83gpvdjzy
2020-08-26 01:24:34 [3069444112] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2020-08-26 01:24:34 [3069444112] DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
2020-08-26 01:24:34 [3069444112] 
```

MushMal, 2020-08-26:
The snippet of log you have attached indicates that indeed something in your pipeline or the device stops the pipeline. The KVS plugin gracefully terminated and exits.

As I mentioned a couple of times before, you should debug your pipeline without the KVS plugin and with fake sink to entirely eliminate KVS. GStreamer forums are a good source of information for these questions.

FlorianRuen, 2020-08-26:
I will try to launch the pipeline using

`GST_DEBUG=*:6 gst-launch-1.0 rtspsrc location=XXXX short-header=TRUE ! rtph264depay ! h264parse ! matroskamux ! filesink location=test.mkv`

And see if the pipeline stop or not

disa6302, 2020-09-17:
@FlorianRuen ,

Closinng this since it seems more related to your pipeline. Feel free to open a new issue if you face any problems

closed 2020-09-17 by disa6302